UnifiedEngineering/T-962-improvements #142: Not working after thermocouple mod

jrozner, 2018-05-11:
I recently built and installed the better thermocouple mod. I populated all 4 ICs but only have the standard two thermocouples connected. They're connected to the 0 and 1 ICs (after messing this up the first time) and the oven recognizes them and detects the temperature. However, when I try to reflow the curve is completely missed and it looks like the lamp never really turns on or warms the oven up. Will having no probes connected to the extra two positions with the ICs populated cause the oven to not heat correctly?

dtmr-e, 2018-05-14:
> Will having no probes connected to the extra two positions with the ICs populated cause the oven to not heat correctly?

Do the temperatures of the TC board (of inputs connected to oven's sensors) show up as left and right, not as X1 and X2? Then it should work. X1 or X2 can be left open. I have installed a TC socket on the backside of my oven for a PCB probe and the TC at that socket (X1) is not normally connected. Reflow still works (X1 temperature disappears). I've installed a piece of normal wire at the unused 4th TC input. That input (X2) thus will the measure the cold junction temperature.

jrozner, 2018-05-14:
Thanks for the suggestion. I was dumb and plugged the thermocouples in backwards which resulted in bad readings I assume. I swapped them around and everything seems to be fine now. I successfully reflowed two boards last night with no issue.

closed 2018-05-14 by jrozner