UnifiedEngineering/T-962-improvements #144: T962A reflow oven

signolo, 2018-05-18:
Good evening,

I found its firmware for the T962 reflow oven.
I own a T962A, the hardware is identical, but has 4 infrared lamps.

Can your firmware be used for the T962A oven?

Is there a preheating in the latest firmware version?

Thanks in advance and I hope in your answer.

A greeting

wulfmans, 2018-05-18:
I have an A version works fine.

signolo, 2018-05-19:
Is there a preheating in the latest firmware version?

> I have an A version works fine.

radensb, 2018-06-07:
> Is there a preheating in the latest firmware version?

There is a bake option, if that's what you mean. It allows you to get the ovens internal temperature to a specified temp before you run the profile, which usually includes 20-30 seconds of "preheat" time before the ramp-up to the soak stage, but that 20-30 seconds will be insufficient to heat the ovens internals to the proper temperature. 

Typically, I use the bake mode to get the oven to a stable 50C-60C, then run a dry run of my desired profile with a thermocouple on a spare PCB. If all looked good with the temp curves, I set the loaded PCB in and run the profile again. Doing this, I get excellent results every time.

PiquantJason, 2018-08-23:
Hello All,

I have this unit and its blowing fuses.  I checked the leads feeding down to the lamps and it's basically a dead short.

Any thoughts?

What should it read? 

thanks!
Jason