UnifiedEngineering/T-962-improvements #150: Physical dimensions for T-962 controller PCB

maskau, 2018-08-25:
I am considering developing a replacement controller PCB for the T-962 reflow oven - mostly for the fun of it. Did anybody measure the physical dimensions of the controller PCB - or is this information available from the manufacturer?

maskau, 2018-08-25:
This post partially answers my question: https://github.com/UnifiedEngineering/T-962-improvements/issues/123. I might as well take the oven apart and measure out the circuit board with a caliper. I am especially interested in documentation on the cutouts, if anybody happens to have that.

closed 2018-08-25 by maskau

wulfmans, 2018-08-31:
anybody had a go with this ?

https://www.ebay.com/itm/T962C-Infrared-Heater-Reflow-Oven-Soldering-Machine-400x600mm-Warranty-Function/263866310545?epid=27019533886&hash=item3d6fa85f91:g:~coAAOSwyCNbasMm:sc:FedExHomeDelivery!85937!US!-1