UnifiedEngineering/T-962-improvements #151: Wide thermocouple temperature variation.

gomond, 2018-10-16:
Hi I have just completed all the mods on the straight out t-962 small oven. And I am finding the left thermocouple reading twice what the right one is end even with minimum offset it reads 35 degrees C while the right reads 24deg.
I used my dual thermocouple meter and placed the 2 thermocouples close to the inbuilt ones and they both read close to the 24deg one.

so the oven won't follow the reflow curve correctly.

Sorry I am very new to SMD Reflowing.

GitLang, 2018-10-16:
There's quite often a discrepancy between the two, particularly when the temperature is changing rapidly. What do you see if you put the oven in bake mode and leave it for a while at, say, 50 degC ?

gomond, 2018-10-17:
I did so just before and here are the results. I have tried tuning. I am going to strip down again and check the thermocouple terminations. It is strange that everyone has the left side of the oven cooler as the heating elements go to the edges on both sides.

[image]
[image]

radensb, 2018-10-17:
Do you have the rear fan running at all for circulation? Thermal mass plays a big part in temperature reading stability of those hanging sensors. The thermal mass of the sensors themselves are very small. A lopsided air current could be moving across one thermocouple differently than the other. Try adding some thermal mass to the sensor. For example, you could use some Kapton tape to tape a small PCB to each sensor (pcb's should be the same so they have the same thermal mass) or maybe something like a penny clipped to each sensor if no spare pcb's are available. Run you test again and see what the temp differences are.

I have always thought it poor form to rely on the hanging probes to gauge the PCB surface temp and control the heating elements. Too much fluctuation and variation which negatively affects the thermal control. I have gotten infinity better results switching the thermal control to be guided by sensors that are actually attached to the PCB (or a representative PCB) that is being reflowed.

zenseidk, 2018-11-08:
Hi,
I also have this oven and have performed the upgrades/changes suggested here (firmware/fan/ds18b20).
My oven also suffers from absurdly varying TC readings.
There's no apparent linearity to it, so I don't see how it would be possible to fix it via a linear gain plus an offset.
I tried putting a third TC from a Fluke meter taped to the drawer so that the TC end is between the two built in TCs and do some characterization. 
I also get 20-30 degrees difference between the two sides and a ~20-40 degrees lower actual temp, compared to the external Fluke meter.
Of course, if you measure on a PCB surface, you will get a lower temperature than if you measure in air, due to the mass of the PCB.

I have settled (for now) with a gain of the left side of 0.86 and 1.28 for the right side.
Then they measure around 20 degrees off at 25 C, but get better at high temperatures.

I also noticed that the Fluke meter was much more lively and showed higher temperatures much faster, but still settled around the same temperature as the "actual" temperature +15 C.
I suppose that could be also be the firmware that is ding a moving average or other filtering?

johnboiles, 2019-10-20:
I just got a T-962 yesterday and I've been testing it out. On manual mode I'm also seeing significant temperature differentials in the two thermocouples.

Photo during cooldown after ramping up to 245C (67C difference between the thermocouples)
[image]

Photo during heating to 100C (31.4C difference between the thermocouples)
[image]