UnifiedEngineering/T-962-improvements #157: One of four MAX31850 thermocouple amplifiers will not work

Bibbbi, 2019-02-06:
I have a custom T962A-Upgrade-Board https://github.com/rharrison/T962A-UpgradeBoard.
One of four MAX31850 thermocouple amplifiers is not working properly.
When i connect the board to an arduino all four thermocouple amplifiers are detected and provide temperature values. 

```
******************************************
ROM = 28 61 64 11 B3 9E 7F AD
  Chip = DS18B20
  Data = 1 81 1 FF FF 7F FF FF FF 27  CRC=27
  Address = 0xF
  Temperature = 24.06 Celsius, 75.31 Fahrenheit
******************************************
ROM = 3B 9C FB 84 9 C8 C 80
  Chip = MAX31850
  Data = 1 40 2 C0 18 F3 FF FF FF 71  CRC=71
  Address = 0x3
  Temperature = 36.00 Celsius, 96.80 Fahrenheit
******************************************
ROM = 3B 52 FB 84 9 C8 C C
  Chip = MAX31850
  Data = 1 5C 2 70 19 F0 FF FF FF 58  CRC=58
  Address = 0x0
  Temperature = 37.75 Celsius, 99.95 Fahrenheit
******************************************
ROM = 3B C1 C8 84 9 C8 6C 4B
  Chip = MAX31850
  Data = 1 4C 2 40 19 F1 FF FF FF 6E  CRC=6E
  Address = 0x1
  Temperature = 36.75 Celsius, 98.15 Fahrenheit
******************************************
ROM = 3B 67 FB 84 9 C8 C A
  Chip = MAX31850
  Data = 1 48 2 60 19 F2 FF FF FF 45  CRC=45
  Address = 0x2
  Temperature = 36.50 Celsius, 97.70 Fahrenheit
******************************************

No more addresses.
```
When i connect the board to an T962 all four thermocouple amplifiers are detected. Only one returns no values. Temp3 always indicates 0°C.

```
T-962-controller open source firmware (v0.5.2)
See https://github.com/UnifiedEngineering/T-962-improvement for more details.
Initializing improved reflow oven...
Reset reason(s): [POR][BODR]
Running on an LPC2134(/01) rev F
Waiting for keys to be released... Done!
Buzzer_Init 
ADC_Init called
OneWire_Init called
Scanning 1-wire bus...
 Found ad7f9eb311646128 [DS18B20 Temperature sensor]
 Found 800cc80984fb9c3b [Thermocouple interface, ID f]
 Found 0c0cc80984fb523b [Thermocouple interface, ID 0]
 Found 4b6cc80984c8c13b [Thermocouple interface, ID 1]
 Found 0a0cc80984fb673b [Thermocouple interface, ID 2]
SPI_TC_Init called
SC18IS602B_Init - No chip found
 bake setpoint values: 0, c8, 200
SystemFan_Init
# Time,  Temp0, Temp1, Temp2, Temp3,  Set,Actual, Heat, Fan,  ColdJ, Mode
 bake setpoint values: 0, c8, 200

Starting reflow with profile: RAMP SPEED TEST
Starting reflow with profile: RAMP SPEED TEST
 bake setpoint values: 0, c8, 200
# Time,  Temp0, Temp1, Temp2, Temp3,  Set,Actual, Heat, Fan,  ColdJ, Mode
   0.0,   40.5,  38.8,  39.0,   0.0,   50,  39.6,    0, 255,   25.3, REFLOW
   0.5,   40.5,  38.8,  39.0,   0.0,   50,  39.6,  176,  10,   25.3, REFLOW
   1.0,   40.2,  38.8,  39.2,   0.0,   50,  39.5,  241,  10,   25.3, REFLOW
   1.5,   40.5,  38.8,  39.0,   0.0,   50,  39.6,  207,  10,   25.3, REFLOW
   2.0,   40.2,  38.5,  39.0,   0.0,   50,  39.4,  244,  10,   25.2, REFLOW
   2.5,   40.5,  38.8,  39.0,   0.0,   50,  39.6,  207,  10,   25.3, REFLOW
Stopping bake/reflow
```

[image]

[image]

[image]

aaronw2, 2019-02-06:
I'm having the same problem with my board. Sensor 3 does not work (address 2). I've been struggling because I always thought I had a bad solder joint or something.
[https://github.com/aaronw2/T-962-Temp-Fan](url)

Bibbbi, 2019-02-07:
Yes, it works now. I cut the DQ pin on the DS18B20.

```
T-962-controller open source firmware (v0.5.2)
See https://github.com/UnifiedEngineering/T-962-improvement for more details.
Initializing improved reflow oven...
Reset reason(s): [POR][BODR]
Running on an LPC2134(/01) rev F
Waiting for keys to be released... Done!
Buzzer_Init 
ADC_Init called
OneWire_Init called
Scanning 1-wire bus...
 Found 800cc80984fb9c3b [Thermocouple interface, ID 3]
 Found 0c0cc80984fb523b [Thermocouple interface, ID 0]
 Found 4b6cc80984c8c13b [Thermocouple interface, ID 1]
 Found 0a0cc80984fb673b [Thermocouple interface, ID 2]
SPI_TC_Init called
SC18IS602B_Init - No chip found
```

Bibbbi, 2019-02-07:
The ReflowController software displays the temperature too high (factor 10). 
What source code changes are necessary?

Source Code: [https://github.com/duncanellison/Reflow-Controller-T-962-](url)

[image]

zian31, 2021-03-07:
What happens on F3 page of the T-962 LCD with 4 TC connected with 1-Wire ? (and no DS18B20 connected for cold junction)
Can someone post a picture of this F3 page ? Thx a lot. And maybe the F2 page also : is there only 2 TC which can be adjusted with Ax+B : the AIR ones or the TRAY ones ?
All the four : AIR LEFT & RIGHT and TRAY LEFT & RIGHT temperatures are shown ?
What is the average shown ? And with this 4 TC, which temperature average is used for the PID ? There is a formula ?
Thanks a lot for your answer, I'm asking it before making this modification.

xnk, 2021-03-08:
This screen shows an oven that _does_ have the separate cold junction sensor, but it really doesn’t affect this screen (the 1-wire sensors also presents a cold junction temperature, it’s just that the separate cold junction temperature is priority 1 to determine system board temperature for system fan control).

F2 screen TC settings are only used for the internal ADCs, no adjustments there affects 1-wire sensors.

This is 30 seconds after power-on without any heating action (no boards to reflow today)

zian31, 2021-03-08:
Thanks for these informations, but I can't see the pictures. Do you have 4 TC or 2 TC connected with 1-wire ?
Do you know if only Temp0 from TC0 (address 0000) and Temp1 from TC1 (address 0001) are used to create "Actual" (average of TC0 and TC1) ? If I look at the TXT data, if find this : Actual = (Temp0 + Temp1) /2
So Temp2 and Temp3 are just for information ?
So is it possible to connect Temp0 to Tray(L) and Temp1 to Tray(R) to be more precise ? Thanks !

xnk, 2021-03-08:
Apparently it seems the image was stripped when replying by email.
[image]

Correct, normally the code will only use the average of ID1 and ID2 as input to the regulation. One way to use additional sensors is the code switched in by this define:
https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/sensor.c#L13-L21
You could easily rework that to average your third and fourth sensor instead.

zian31, 2021-03-08:
Thanks a lot ! Your picture shows the TRAY X1 and X2 Temp, it's perfect.
I want to keep the same software, and just invert the connections of AIR thermocouple and TRAY thermocouple, so that TRAY thermocouple will be used as input to the regulation (So TRAY will be showned as L and R on F3 screen, and AIR as X1 and X2).
Is my solution correct ?

xnk, 2021-03-08:
I can’t see why that wouldn’t work. It’s probably almost as fast to implement as just changing the firmware. Give it a go!

zian31, 2021-03-08:
Thx, I will try it because TRAY temp seems more realistic as AIR ones. I think I will try with only 2 TC 1-Wire connected to the TRAY TC because I can't find a 4 TC 1-Wire board with 4x MAX31850K on any stock anywhere.

I've another question concerning the ReflowController software used to show the printscreen here (with displays the temperature too high, factor 10 - Source Code: https://github.com/duncanellison/Reflow-Controller-T-962- ) :
Where can I find a compiled version for windows ? And is there anonther one also for Linux ? I don't find it in the source code link.