UnifiedEngineering/T-962-improvements #160: Idea: reverse main fan and add carbon filter

stan23, 2019-02-20:
I was wondering whether the main fan can be reversed (i.e. suck air from the bottom and blow it out at the back), and put a carbon filter behind the fan.
That should reduce the smell from the flux.

Has anyone tried this before?

GitLang, 2019-02-20:
You will very quickly overheat the fan. It would be sucking oven heat through it instead of room temperature ambient air.

stan23, 2019-03-01:
Yeah that's right.
So what if I put the filter under the oven, maybe using higher feet?

GitLang, 2019-03-01:
Air comes out of every gap and orrifice in the thing, you would have quite a job.

stan23, 2019-03-01:
Well, opening the window is not sufficient here in the basement. I'm not sure how to improve this...

m-kozlowski, 2019-03-01:
I'd just put external fume absorber near the oven and don't try to reinvent the wheel.

mikeanton, 2019-03-01:
I built a fume hood of sorts in my basement.  I hung a plastic curtain from the ceiling around a high bench in the corner of the room, and installed a reasonably high flow exhaust fan (200 cfm), that vents to the outside.  If the oven is run in that chamber, there is no smell in the rest of the house.  This has been handy for other smelly things I use as well, and it keeps the wife happy.

stan23, 2019-03-31:
I'm not using the oven that often, so my replies are quite late.

I tried the external fume absorber, and even taped 3 sides of the gap oven-ground, but hat doesn't help.

Next I'll try some kind of hood with a carbon filter...

beyon, 2019-04-12:
My oven has been sitting unused much due to fume issues and lack of time(mostly lack of time).

I did try building a small box that I put under the oven with only an opening for the tube to my solder fume extractor (Weller Zero Smog something).

It kind of worked with the fumes but the problem was that the oven wouldn't get up to high enough temperature with the fume extractor on, however turning it on later during the soldering/heating process did provide a way to get the temperature to fall quicker if you need that for your soldering profiles :) (but with fumes before starting the extractor.

Guess I should try another closed box around the oven+box+fume tube with an air-inlet that only opens once the fume extractor has started.

rafii6312, 2021-01-16:
@stan23 Hello, this might be no longer a thing (given the 2 years),
but I have designed a fan duct with a built in carbon filter.
If you (or anyone else) are interested, just PN me and I'll send you the STL files.
I'll also upload my project to thingiverse the next days.

[image]

stan23, 2021-01-16:
I'm not using the oven lately.
How does that work for the high temperature of the exhaust gas?

rafii6312, 2021-01-16:
Ive noticed that the exhaust temps are not that high because of the high airflow.
i have also bought some aluminium-tape to tape the inside of the duct, in case it
gets too hot. for now i printed my parts with PLA which has a pretty low melting point,
but i doubt that the exhaust temps exceed 140°C over time (the table that is blowing on was about 60°C).
In case i get some temp problems there, I'll print again with high temp filaments (230-330°C).
this is still a prototype tho