UnifiedEngineering/T-962-improvements #161: Oven not turning on

thefloe1, 2019-02-28:
I made the upgrades (including the 4 termocouple readout) and used the oven now and then. It was not in use for some time now and when I wanted to use it again now the oven doesn't turn on any more. The display turns on, the fans turn on but then it somehow resets I would say and starts again. 

With the controller PCB power from an external laboratory supply everything works fine and the controller is drawing around 120mA at 9V. So I expected the transformer to be broken. After replacing it with a 9V, 6VA model I still have the same issue. As it is running fine when the controller is power externally and I guess the current draw is in the normal range, I don't expect a component on the controller board to be broken.

Has anyone else had such problem and figured out a solution?

Thanks

GitLang, 2019-02-28:
Have you measured the DC voltage and current when powered by your 6VA transformer?. You may want to try a heavier duty transformer at a higher voltage as per the discussion we had here in the last week or two. I'm interested to know how low your DC rail is. Also check the amount of ripple - you may have some of the rectification diodes dead (Q2,3,4,5). The main smoothing cap C21 could be dead too. Does it look ok or is the lid popped or bulging?. 

Have you checked around the reset line and the ISP socket for shorts, opens, bad joints etc?. I have just recalled that the software has a watchdog timer. Could it be resetting every watchdog period?

thefloe1, 2019-02-28:
C21 looks good. The bridge rectifier I tested by reversing the DC power supply and everything worked. So the diodes should be fine.

The ripple I didn't measure but in both cases the 3.3V rail was at 3.3V. I will check with some scope when I'm back.

thefloe1, 2019-03-12:
Hi,

Just came back and gave the oven another test.

I only had a simple scope at hand but I can see that the voltage rails are not stable. 
The input voltage is around 11V and dropping by 5V every 170ms.

I guess I will habe to take the whole PCB out to do further tests and focus on the input section.

thefloe1, 2019-03-13:
Update:

Here is the output from the serial interface:

`
T-962-controller open source firmware (v0.5.1)

See https://github.com/UnifiedEngineering/T-962-improvement for more details.

Initializing improved reflow oven...
Reset reason(s): [POR][BODR]
Running on an LPC2134(/01) rev F
Waiting for keys to be released... Done!
Buzzer_Init 
ADC_Init called
OneWire_Init called
Scanning 1-wire bus...
 Found 9f2cfc0958b8d93b [Thermocouple interface, ID 1]
 Found 0f2cfc0958b8e53b [Thermocouple interface, ID 2]
 Found 0b2cfc0958b8c33b [Thermocouple interface, ID 0]
 Found eb0cfc0958cc533b [Thermocouple interface, ID 3]
SPI_TC_Init called
SC18IS602B_Init - No chip found
SystemFan_Init
# Time,  Temp0, Temp1, Temp2ü
T-962-controller open source firmware (v0.5.1)

See https://github.com/UnifiedEngineering/T-962-improvement for more details.

Initializing improved reflow oven...
Reset reason(s): [BODR]
`

So it's not the watchdog bu the brown-out detector. So I will take a closer look at the power supply section.