UnifiedEngineering/T-962-improvements #162: Cant get OUT of bootloader, or its bricked

jdeglavina, 2019-03-22:
I've been trying to load the new firmware almost all day not, but I've been unable to get past connecting to the controller. I get the autobaud error when trying to connect. 

Right now, with nothing connected to the serial interface pins, when powered on, the oven and fan go to full and the screen in blank. I've tried resetting via the reset pin with no change.

Infear that I crossed a power wire at some point and fried something. Anything I can check here to figure out what the problem is?

GitLang, 2019-03-23:
Sounds like the oven is still in Programming mode. Try removing the program mode wire and rebooting the oven. You should get the normal menu.

To program it, you should actrivate the program pin, pulse reset, program the chip and verify it. Turn off. remove program mode link and power on.

jdeglavina, 2019-03-23:
Thanks for replying. Yeah, that's the problem. With no serial interface connected to the oven at all, the heater and fan comes on, and the screen is blank. Power cycling does nothing. Pulling ISP and Reset low to high does nothing. Pulling the reset pin low from an on-board ground causes the screen to light up then turn off completely.

The only other thing I can think of at the moment is maybe I need some better grounding. I don't have anything connected to the top lid. I just took the paint off of the connection for earth ground.

jdeglavina, 2019-04-01:
So, I replaced the board and now it works fine. Original board was definitely not right.

closed 2019-04-01 by jdeglavina