UnifiedEngineering/T-962-improvements #163: Unable to flash the T-962A oven

BLF-Lexel, 2019-03-30:
I have followed the instructions but I dont get with flashmagic or LPC2000 a connection to the oven

I measured the connections and the prtogramming pins are connected
USB CH340 Serial board detected and driver installed

Manual bootloader mode turns on fan and heater, I diconnected outputs after a while, but I did sometimes get blank screen mostly the menu

Starting the oven and hold F1 does not turn on the bootloader it will be stuck in the T962 IR heater screen

GitLang, 2019-03-31:
Did you use a 3.3 volt USB <-> serial converter?. 5V won't work and may well cause damage.

BLF-Lexel, 2019-03-31:
Problem was that someone meant it was a good idea to define that you can not flash with 1:1 pin connection, you need to connect 

    TXD --> RXD
    RXD --> TXD
    GND --> GND

I dont know which idiot of programming expert defined the serial connection like this but thats totally weird in my eyes

closed 2019-03-31 by BLF-Lexel

GitLang, 2019-03-31:
That is completely normal and has been since long before programming experts existed. Think about it - TX is an output, RX is an input. Output goes to Input.

thesoftwarejedi, 2019-05-18:
@BLF-Lexel Think of serial communication as 2 telephones.  A telephone has a RX (earpiece) and a TX (microphone).  To wire them up, you wouldn't connect the earpieces and microphones to each other.

This wasn't thought up by some "idiot of programming expert", it's simply logical.