UnifiedEngineering/T-962-improvements #164: Just another compatible board - How can I donate you?

Pnoxi, 2019-05-01:
I made another upgrade for T962. It will be available on tindie [tindie](https://www.tindie.com/products/15262/).

[](https://hackaday.io/project/164001-t962t962a-controller-upgrade)

[T962/T962A Controller Upgrade (Part 2) - Installing the controller to T962
](https://youtu.be/GOKtl5OaKDw)

**I can support you from every sold piece. But as? I prefer PayPal.**

[image]

[image]

aaronw2, 2019-05-01:
Hi,

Just send me a kit and I'll be happy. In my case I am using all 4 
thermocouples so I have two on the board. I like what you did with it. A 
mention in the documentation would also be appreciated. Are you going to have 
two versions, one with the optional thermocouple support?

-Aaronb

On Wednesday, May 1, 2019 3:45:05 AM PDT Pnoxi wrote:
> I made another upgrade for T962. It will be available on tindie
> [tindie](https://www.tindie.com/products/15262/).
> 
> [](https://hackaday.io/project/164001-t962t962a-controller-upgrade)
> 
> [T962/T962A Controller Upgrade (Part 2) - Installing the controller to T962
> ](https://youtu.be/GOKtl5OaKDw)
> 
> **I can support you from every sold piece. But as? I prefer PayPal.**
> 
> ![2](https://user-images.githubusercontent.com/44557164/57014051-906e4b00-6c
> 0e-11e9-8330-b67e7f4e5017.jpg)
> 
> ![5](https://user-images.githubusercontent.com/44557164/57014065-a714a200-6c
> 0e-11e9-9e2c-17406f720d32.jpg)

aaronw2, 2019-09-12:
*poke* Any update on this?

Pnoxi, 2019-09-15:
> On Thursday, September 12, 2019 7:16:14 AM PDT GitLang wrote: What exactly is it that you are waiting for?. You can buy it now straight off -- https://www.tindie.com/products/pnoxi/t-962t-962a-reflow-oven-controller-up grade/ There is stock of the 2 thrmocouple but 4 thermocouple is sold out.
> You offered to send me a percentage since you used my design and I said that instead just send > -Aaron

closed 2019-09-15 by Pnoxi

Pnoxi, 2019-09-15:
> You offered to send me a percentage since you used my design and I said that instead just send me a couple of boards.
> -Aaron

Hi, this donate question belonged to the original author of UnifiedEngineering. Not for you aaronw2.
I didn't use your design to develop my board. The development of my board began in 2017.
So I don't see why I should send you one piece for free... :astonished: 
[image]