UnifiedEngineering/T-962-improvements #167: Stoopid Noobie Question

KarlFrick, 2019-06-21:
Is there anywhere I can download a compiled .hex or .bin file of the most recent release of this project? I can't seem to find one anywhere!

radensb, 2019-06-22:
https://github.com/UnifiedEngineering/T-962-improvements/releases

KarlFrick, 2019-06-24:
DOH! I was missing the "Assets" drop-down, I was too drawn to the links on the left, it just kept sending me  in circles. All's well! 

But it seems I am not being successful in loading the hex file into the controller via the PNOXI interface with the USB port on the side of the oven. The RX LED on the PNOXI interface shows activity, but the display never seems to reflect a successful update, as "NO VERSION INFO" is displayed on the splash screen.

I have tried several different baud rates, but so far, no luck. There must be a document I haven't yet found that tells me the proper serial framing. Anyone have any suggestions?
-karl

MofoMiq, 2019-07-17:
[image]
Karl,

When I did a long press of the tac switch on the Pnoxi board, the red LED on the Controller board did not go all the way out, it just got dimmer.  Also, the screen did not go blank.  I had a menu showing on the display while it was programming the firmware.  

Attached is a pic of the settings in the latest version of Flashmagic 12.25 5558Beta.  Did you see the clock setting in now buried in the VIEW --> PREFERENCES --> HARDWARE tab now.