UnifiedEngineering/T-962-improvements #171: Cannot read or write MCU

Mark-81, 2019-08-31:
I'm trying to update the firmware. I checked the MCU is LPC2134F8064/01, XTAL is 11,0592 MHz and I connected a 3V3V UART/USB converter to my pc that runs Win7. 
Using FlashMagic I cannot do anything whatever baudrate I select. For example reading the signature leads to "LPC2134: Device does not support operation". Reading a sector leads to "Operation failed". Writing the firmware leads to "Failed to send data to the device".

Erasing is the only thing that seems to work...

Had anyone ever encountered this behavior?

GitLang, 2019-08-31:
I guess you realise you should hgave read and saved the contents before erasing them.... :)

Have a read of an earlier post and see if you are following it correctly

https://github.com/UnifiedEngineering/T-962-improvements/issues/169

Mark-81, 2019-09-01:
Don't worry, I knew that - but the oven became pretty unusable (didn't melt the solder cream completely anymore) so no harm if I would brick it :-)

I've already read the issue you linked. Anyway I found the problem. There was an inconsistency with the decimal point in the crystal settings in FlashMagic. Fixed that I was able to flash the firmware.

closed 2019-09-01 by Mark-81