UnifiedEngineering/T-962-improvements #172: How to tune TC?

Mark-81, 2019-08-31:
After leaving the oven off for some hours I turn it on and I read:

> Cold junction: 32 °C (about the room temperature)
> Left: 57 °C
> Right: 37 °C
> Act: 47 °C

In the setup menu I see I can set gain and offset for both TC but honestly I cannot find any documentation about the recommended procedure to do that. The same applies for the test profiles.

Where could I find such a documentation?

Mark-81, 2019-09-01:
In the meanwhile I read almost all issues that talk about this topic. 
Some further clarification (I apologize if I should RTF, but as said I cannot find it, neither on the UE website).

**The actual point that should follow the temperature profile is the PCB.** So if I tape a temperature sensor (i.e. the common Fluke probe: https://uk.rs-online.com/web/p/thermocouples/4723319/) to a dummy board I should know the _real_ effect of the heating system. I'm aware the reading is lower (actually, slower) due to the thermal mass, but giving enough time all the parts in the oven should reach the same temperature- possibly disturbed by the fan. Applying some mass to the internal TCs might reduce a little bit this difference.

Someone suggest to run a dummy cycle to raise the internal temperature: why should it not be enough to manually preheat, say @ 70 °C for 10 minutes? I mean, after the dummy cycle the temperature comes down again.

In my case, anyway, at 250 °C I read:

> Act: 249 °C
> Left: 261 °C
> Right: 238 °C
> Cold junction: 38 °C
> Actual PCB temperature: 210 °C

Of course it cannot melt it! I also replaced the lamps (https://www.aliexpress.com/item/32547243899.html) without any change.

What might lead to such a behavior?

GitLang, 2019-09-01:
> Act: 249 °C
> Left: 261 °C
> Right: 238 °C
> Cold junction: 38 °C
> Actual PCB temperature: 210 °C

Hard to make sense of without defining what "actual" means. I would say that no one reading of those is any more relev ant than the others. If the thermocouples were calibrated at the factory then those readings should be fine (within a degree or so). These ovens are notorious for left-right unbalance and you will read plenty of opinons of why on here. Turbulance from the fan is one contributor,

As for everything settling to a stable temperature, my opinion is that a controlled temperature air flow would achieve (and does achieve) exactly that. In this oven the heat source is radiant, so the temperature distribution is complicated. Somebody did a good analysis of the subject back on here somewhere.

Most people have found that using the digital thermocouple addon pcb helps, it cedrtainly removes any calibration or drift from the chain and gives another couple of channels. But still people see quite a variation on left-right temperatures. Two things that people have done - firstly, make a fan airflow lieariser which is basically a fan sized aperture with a stack of small copper pipes soldered in, so a lot of turbulence is removed. Second, somebody else air added a small fan in top of the oven blowing downward which went a long way toward evening things out. I think it had a fabricated metal blade rotor and shaft assembly inside the oven, shaft through the roof, and fan motor in the enclosure above.

Don't try to raise the temperature too quickly in the profile. Start with one degree C per second, and make sure the hold periods are long enough for the slow oven to catch up. After the improvements mentioned, the one thing that really need some work is the PISD algos and coefficients, they just don't work properly.

If you want to make an assesment of temperatures, I would say -

1. Use the bake function and wait long enough for things to really settle. You can then judge it at different steady state temperatures.

2. Don't add a thermocouple to a PCB or any other thermal mass, it is just about meaningless. Soldering results are the only thing that really matter.

3. If you doubt the fitted thermocouples then put one of your own in close contact and make sure the temperature has settled. Also remove any thermal mass that you can from the thermocouple, like insulation.

4. If you really want to achieve a stable temperature for measurement purposes, you could remove the ovens control system from the loop and drive the lamps diretly off a well rated variac.

Good luck.

Mark-81, 2019-09-01:
Thanks a lot for the time spent to answer.
"Actual" means I taped an external probe to the PCB. I got you hints but there is one major point I didn't get:

>Don't add a thermocouple to a PCB or any other thermal mass, it is just about meaningless. Soldering results are the only thing that really matter.

I thought that to have good soldering results, the _PCB temperature_ should follow the chosen profile - no matter what temperatures are read elsewhere. Is this assumption wrong?

I decided to try these improvements because after a couple of years of very good results, now the oven doesn't melt completely the solder cream, _even the profile is followed correctly_ (with both original and modified firmware). For this reason I guessed the TCs are not reading the temperature properly and my measurement confirmed the temperature at the PCB surface is a lot lower than expected.

GitLang, 2019-09-01:
It's a completely valid assumption in theory, but it doesn't work in practice. Where on the PCB do you measure?. The only place that matters is the paste temperature, with the proviso that components don't exceed their soldering temperature spec. The solder paste temperature, and hence "the soldering" that I refererred to will be different all over the board during the last rise slope, but even then it will be to the end of the high plateau before all solder has reached liquidus and flowed fully. The temperature at any one place or one TC will not tell you when that point is reached.

It does sound like you have a fault. I would recommend fitting the digital TC board and new thermocouples and take it from there. You obviosly need to fault find the solid state relays/triacs and the tubes.

You say the solder isn't melting "even if you follow the profile". That just can't be!. If the solder follows the profile and it crosses liquidus for long enough, then it must fully flow. Some of the information has to be wrong. If it follows the profiles the profile ok, how can the thermocouples be dodgy?. I can only assume you mean the *reported* temperature follows the profile but the *actual* temperature is wrong, possibly due to a fault somewhere. If so, read para 2 again :). BTW, a lot of people who followed the areticle to build the first digital TC interface board had to buy 10 bare PCB's from the pcb maker. So there should be somebody somewhere who can give you one.

Mark-81, 2019-09-01:
Yep, of course I meant the reported temperature follows the profile - i.e. the two curves on the screen match. I'll investigate further. Thank you.

In any case, coming back to the OP: is not there any documentation about all the features of the new firmware?

GitLang, 2019-09-01:
None that I know of, but you have the full source code and a schematic of the PCB.

closed 2019-09-03 by Mark-81