UnifiedEngineering/T-962-improvements #174: Reflow Oven Vent

Raasha, 2019-09-30:
Hello.
I found an interesting solution to the problem of smell when soldering in a reflow oven.
https://www.briandorey.com/post/t962-infrared-reflow-oven-vent-hack
Who did the ventilation for their reflow ovens?

Raasha, 2020-06-17:
My improvements for ventilation:
[image]
[image]

and additional heating:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

result:
[image]
[image]

mikeanton, 2020-06-17:
Do you have a temperature log of a thermocouple on the PCB?  The air temperature alone is less important.

Raasha, 2020-06-18:
The log looked, but did not record. I have installed on a permanent basis 2 additional thermocouples attached to the breadboard. These boards are installed on the sides and at a height of 10 mm. Made a modification for the thermocouple interface on the MAX31855. The main thermocouples are lowered as far as possible. I raise the soldering boards by about 15 mm. The fan speed is set = 35. Two fans are installed. One is a factory one that blows on an additional heater at the rear. The second is the exhaust installed in front of the ventilation duct in the room, see the second photo. The viewing window was closed with several PCB plates to minimize the loss of the template. Also closed all possible gaps from below. The goal was the lack of smell when soldering, without compromising the quality of soldering.
According to the log, the temperature on the boards installed with additional thermocouples lags the temperature of the main thermocouples by 5-15 degrees depending on the mode. Soldering is performed quality. While filling 60% of the stove.