UnifiedEngineering/T-962-improvements #175: can't find wiki page for adding DS18B20

pmozbert, 2019-10-19:
I see lot's of mentions of the cold junction mod with DS18B20 temperature sensor, but can't find the wiki page about how to do it. The closes I see is the one for the Dallas chip, but nothing for the 1-wire maxim chip.

radensb, 2019-10-19:
https://github.com/UnifiedEngineering/T-962-improvements/wiki/Dallas-18B20-for-cold-junction-mod

johnboiles, 2019-10-19:
That page references the "the cold-junction mod as described in the Wiki". Sounds like there should be another page for the Maxim 18B20. I ran into the same thing today.

m-kozlowski, 2019-10-19:
There is no dedicated page, as far as i know. The best documentation for 1wire mod is the "Cold junction compensation" paragraph of README.md along with [this image](https://raw.githubusercontent.com/wiki/UnifiedEngineering/T-962-improvements/images/tempsensor-mod.jpg) from the wiki start page

pmozbert, 2019-10-19:
Thanks, I was hoping to find information about which pads to use on the board, though i can sort of make it out from the picture. And also the specifications for the resistor that must be added.  Also, there are several variations of the Maxim 18B20.  Some use parasitic power, others not.  So it would be good to know exactly which one is being referred to. And finally, in the picture it shows 2 pins soldered to ground(?).   But for the parasitic chip, one of the pins is NC.  So I'm wondering why it's being also attached to ground.

m-kozlowski, 2019-10-20:
all of the 18B20s I've used required pin3 connection to GND to work in parasite mode.
If yours doesn't need that and have pin 3 unconnected, then soldering both pins to ground might still be good idea due to increased stability.

Resistor value (4k7) is mentioned in README.md, same as 18B20 documentation suggests. My guess is anything between 2 and 5K should work.

gcormier, 2019-11-08:
Found a good video as well that covers it - indeed the Wiki is missing some basic info.

https://www.youtube.com/watch?v=HjYqSfLzBKk

pmozbert, 2019-11-08:
I was able to complete the mods.   Here's a couple notes.
1.  I used the [Maxim DS18B20-PAR+T&R](https://www.mouser.com/ProductDetail/Maxim-Integrated/DS18B20-PAR%2bTR?qs=sGAEpiMZZMucenltShoSnqz2Nq7rAtSUc2%252Bf2sR2S48%3D).  I soldered pins 1 and 3 to ground, even though pin 3 is marked NC. 
2. there was no exposed ground pad at the place where you're supposed to solder the 18B20.  I scraped away the mask layer to expose the ground pour so I could solder to it.  I found that bending a small foot on the pins made it easier to solder.  As others have mentioned, I was very careful bending the pins to avoid damage to the internals.
3.  I followed the [flashing instructions](https://github.com/UnifiedEngineering/T-962-improvements/wiki/Flashing-the-LPC21xx-controller) in the wiki, but the part about entering the bootloader by grounding nIsp and then nReset didn't work for me.  I had to do it by grounding nIsp and then powering on the oven.
4. I guess it should be obvious but the flashing steps don't state it.  You have to power on the oven before you can flash it.
5. I used [this USB to TTL serial converter ](https://www.amazon.com/gp/product/B072K3Z3TL/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1)with CP2102 chip to connect from my pc to the oven for flashing.  It worked fine.  As mentioned in the flashing wiki, I only needed to connect GND, RX, and TX. 
6.  I found that there was no ground connectivity between the top and bottom case.  To fix this, I scraped away the paint around the screw holes that bolt the top to the bottom, so now the screws make the connection. 
7.  I used 3/4" wide kapton tape.  This width went just fine, and seemed to be best choice to balance width and cost.

karrots, 2019-11-13:
There are two part numbers for the DS18B20. 1) DS18B20 and 2) DS18B20-PAR. The -PAR model only supports parasitic power hence the NC pin. The non-PAR model supports powered mode or parasitic mode. When used in the parasitic mode you connect VCC to GND.

It's helpful to read the section titled **Powering the DS18B20** in the datasheet to ensure proper power. They also don't suggest running in the parasitic mode above 100C.

https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf
https://datasheets.maximintegrated.com/en/ds/DS18B20-PAR.pdf

jwestmoreland, 2020-04-05:
Hello All and just a comment:

It should be possible to power the PCB via the USB-Serial adapter.  I did that when I took out the PCB - I'm using one of the cables from FTDI but I'm getting one of these C232HD-DDHSP-0 (URL:  https://www.mouser.com/ProductDetail/FTDI/C232HD-DDHSP-0?qs=%2Fha2pyFaduibYjSO6Dz1ybsSG%252B9jH52tzwTckDaG7GOSJXKPayvYHw%3D%3D ) since you can use FLASHMAGIC to control the n_ISP and Resetn via RTS and DTR.

Like what's mentioned on this site - the 3.3V rail is close to the ISP connector - on the RHS there's a 4.72K resistor that a pin can be soldered to.  A pic on this site shows a connector and I believe that's the point where the soldering was done.  I soldered to that with a standard 0.1" header pin and powered the PCB from that.  Note, I had the board out of the chassis - I haven't tried this yet with the board in the chassis.  I do want to look at the serial data when the unit is running, so I'm planning on adding the  C232HD-DDHSP-0.

HTH,
John W.