UnifiedEngineering/T-962-improvements #181: Thanks - Great Job with this!

jwestmoreland, 2020-04-05:
Hello UE,

Thanks for the great work - I followed the instructions and it's working as advertised.

The PID Control Test is dead-nuts on:

[image]

I got this unit back in 2013 - and just did the mod.

Thanks Again!
John W.

GitLang, 2020-04-05:
Your message seems to be isolated so I can't see what instructions you followed exactly. If it was setting up the PID, I'd like to know how, as I find the PID just nuts rather than dead-nuts on :)

jwestmoreland, 2020-04-07:
Hello @GitLang,

Ha!  OK - the UE folks don't appear to post their e-mail addresses so this post was in lieu of that.

My remark regarding the PID is shown in the pic - if you click on that you'll see - the PID is tracking the design requirements closely.

Regards,
John W.

GitLang, 2020-04-07:
I agree, the PID looks much better than I have ever seen this oven do. That is why I asked you what instructions you used to get it so good, please.

jwestmoreland, 2020-04-23:
Hello @GitLang ,

In case it's helpful - I've posted the logfile from the PID run here:

[PID-TEST-LOG.txt](https://github.com/UnifiedEngineering/T-962-improvements/files/4520736/PID-TEST-LOG.txt)

Regards,
John W.

Shaische, 2020-06-05:
@jwestmoreland As @GitLang has stated, what did you do? Do you have instructions you followed?

GitLang, 2020-06-06:
Yes, it really would be great if we could see just how this improved PID setup is done. I don't think anybody else has suceeded in getting anywhere near to a good PID response from this oven yet.

Shaische, 2020-06-12:
My oven starts to overshoot at around 160C. It showed 200C on multimeter when the oven showed around 160C. I immediately shut it off fearing a thermal runaway event. How do you tune the thermocouple using offset/gains on the setup page? I don't see any instructions. @GitLang do you know?

ez2X8pk61DHkdztgFT6O, 2021-01-14:
Treating this as the ‘thank you’ thread: thank you! 

My 07-2020 T-962 is nicely working with its new firmware and a ds18b20 T-sensor. Note that I hooked up DTR and RTS to the reset/program mode pins as grounding the program mode pin and booting did put the controller in programming mode (from empty screen, spinning fan, and hot heater) but did not allow for programming (auto baudrate failure). In my system the thermocouple readout showed a very large offset for one themocouple of >25oC. Should not be possible given the specs of the amplifier. Anyway, I run into the +-25 degree maximum of the offset correction. Looked in the code and saw that I could either go for 0.5 degree steps or add another byte of persistent storage to increase the range.