UnifiedEngineering/T-962-improvements #184: T962A has stopped working

Raasha, 2020-05-16:
Hello.

Help understand where to look for a malfunction.
I tested the oven for an hour. Then he waited until it cools down. Pulled out a thermocouple to test. He turned on the stove, it started, pressed F3, after which the stove hung. Turned off, waited 10 seconds, turned on. The screen saver appears, the LEDs of the fan and the ten are lit, and then nothing happens. After 2 seconds, the hike that the controller restarts and the situation repeats. Attached a picture.
What could happen and what to check?

Thermocouples connected via MAX31855K

Raasha, 2020-05-16:
[image]

Raasha, 2020-05-17:
Found a malfunction.  The input for the F3 keyboard is always at zero, so the watchdog timer is triggered.  The pull-up resistor is okay.  Supply voltage ok.  There is no short circuit.  Can someone recompile the firmware to any free output of the controller?

GitLang, 2020-05-17:
I't more than likely to be the switch gone short circuit?. I't definitely the first thing to check.

Raasha, 2020-06-17:
Maybe someone will come in handy. I changed the controller, programmed it and it worked. As expected, the port pin was always at a logical zero with the controller turned on. There was no short circuit in the pin. Also added resistors 1k to limit the current between the controller and the keyboard.
[image]