UnifiedEngineering/T-962-improvements #185: How to do PID tuning?

Shaische, 2020-06-05:
Hi, I've seen users post logs here of temperatures. How do you get these? Did you install one of the additional add-on boards some people offer to plug via USB?

Also, how do you guys do PID tuning to tune the temperature more precisely? I don't see any instructions.

katodo, 2021-01-20:
I'same problem, I'm using T-962 + https://www.tindie.com/products/pnoxi/t-962t-962a-reflow-oven-controller-upgrade/ HW
[image]

fah, 2021-01-29:
I think if the heating elements are too weak, the PID controller won't help either.
At least thats what I deduced from the diagram in https://github.com/UnifiedEngineering/T-962-improvements/issues/185#issuecomment-763944439 .

It seems to work well for cooling and I suspect that there is a PID algorithm working here or the graph is very coarse (?).