UnifiedEngineering/T-962-improvements #186: Backlight on but (mostly) no text on LCD

ecurtz, 2020-06-16:
Just flashed my T-962 with the 0.5.2 release. It seemed to be working fine but I had nothing on the LCD after reassembling (backlight is on). I took it back apart and have occasionally gotten the LCD display after wiggling the wires and stuff but only once or twice. It will either be working when powered on or not, I haven't gotten it to come back by wiggling the cable while running. Is there ANY chance this is a software issue or do I just need to scrape off all the hot glue and replace the ribbon cable to the LCD?

MaikBoe, 2020-06-16:
I have the same issue. When I turn it on after some hours without power, sometimes the display is blank and the backlight on.
After turning it off, wait for about 5-10 seconds and turn it on again. This is my solution. Then mine LCD is working normally.

ecurtz, 2020-06-16:
Thank you! That does in fact appear work on my machine as well, which makes it sound like a bug with connecting to the LCD? Maybe there needs to be a few seconds delay at startup for it to charge caps or whatever is happening?

plains203, 2021-02-24:
The resistor mod fixed it for me as well. Thanks @gojimmypi