UnifiedEngineering/T-962-improvements #187: Unable to flash with ground connected

Vitalic66, 2020-06-24:
Sorry, need to ask here again,

I tried 3 different usb ftdi devices always with the same result. No communication.

Tried it with win7 win10 and ubuntu 18.04 . Tried all possible baudrates.

Connected Tx Rx and ground but i only get the garbage output if i remove the ground.

Tx shows 0.9V Rx 3.22V .

I already read about the 3.3v level trick from the resistor next to the isp connector, but which signal do i have to connect? 3.3v to the Tx or/and the Rx pin of the isp connector?

Bootloader mode should be active as the screen stays blank and the heater/fan turn on.

wulfmans, 2020-06-24:
its been years but if i remember right
you need tx,rx ground AND 3.3v power
the USB side does not power things

MaikBoe, 2020-06-24:
I did it a few weeks ago.
RX => TX
TX => RX
GND => GND

I looked at google and found „T962A Reflow Ofen Modifikation“ from Timber.

Vitalic66, 2020-06-24:
That is what i did with 3 different ftdi dongles but all with the same result.

Need to remove ground to see something happening in lcp21isp -termonly terminal under linux.

In windows with flashmagic i always get failed to autobaud when.

I've read something abou the 3.3v problem, but which line has to be connected? Rx Tx or both. Connecting both sure won't work.

MaikBoe, 2020-06-24:
Flash Magic on the first try.

The FTDI Adapter must be set on 3,3V with Jumper.

The ISP Pins from left if you can read the „ISP“

1. n_ISP (connected to GPIO0.14, ground during reset to enter ISP mode)

2. n_RESET (active low reset input)

3. TXD0 (UART0 output to host, LVTTL 3.3V)

4. RXD0 (UART0 input to MCU, LVTTL 3.3V)

5. Ground (this ground is floating with respect to earth ground in the mains connector and chassis)

Wire setup:
I had wired Pin 1 and 2 over two seperate switches to Ground.
Pin 3 (TXD0) to RX on the FTDI.
Pin 4(RXD0) to TX on the FTDI.
Pin 5 GND to GND on the FTDI.

To enter Bootloader:

1. Ground n_ISP (Pin 1)
2. Ground and release n_RESET (Pin 2)
3. Release n_ISP (Pin 1)

To flash with Flash Magic select the correct COM Port
Baudrate 57600
Oscillator 11.0592

You don‘t need 3,3 V supply.

Vitalic66, 2020-06-24:
Sure, checked, doublechecked, tripplechecked... no communication at all.

Also tried Rx and Tx loop to see ftdi working, no problem on echos in puttie.

Somehow i guess that something is wrong with the pcb or the lpc2134.

As i already mentioned, i see garbage output in console when removing the ground connector.

MaikBoe, 2020-06-24:
I bought this one

DSD TECH USB zu TTL Seriell Adapter Konverter mit FTDI FT232RL Chip
https://www.amazon.de/dp/B07BBPX8B8/ref=cm_sw_r_cp_api_i_9q68EbB58MG0S

Perhaps adapters not compatible?
I doubt that the Controller or the LPC2134 got problems or damaged.

Vitalic66, 2020-06-24:
Bought this one

https://www.amazon.de/gp/aw/d/B01N9RZK6I?ref=ppx_pt2_mob_b_prod_image

Also tried my adafruit ft232 same results.

wulfmans, 2020-06-24:
you need 3.3v your connection above does not have that

MaikBoe, 2020-06-24:
Seems to be the same chip.
I don‘t know about the quality from AZ Delivery. The DS18B20 you should not consider to buy from this company. They won‘t work.
I bought my T962A from Mophorn over „Amazon.de“ last month.
Latest drivers and Flash Magic from the internet.
In a few days I will get 3 PCB‘s and a stencil from Oshpark for the MAX31850 Thermocouple and USB to TTL Mod.

Vitalic66, 2020-06-25:
So I checked all traces from isp to ic, no problem.

Desoldered xtal, 11.059, no problem.

Somehow I don't know what to do anymore. Can't establish com with ground connected.

Only getting something done when ground is not connected.  I can start the flash after ignoring the message the device detection failed, but at the end i get an error "hex prologue error", flash failed.

Any new suggestions?

How do you exactly start bootloader? Main off, push button n_isp and button n_reset, main on, release buttons.

Or

Main off, push n_isp, main on, push n_reset, release buttons.

Or

Main on, push n_isp, push n_reset, release n_reset, release n_isp?

Is main on or off before bootloader?

MaikBoe, 2020-06-26:
- Ground n_ISP (Pin 1)
 - Power on
 - Ground n_RESET (Pin 2)
 - Release n_RESET (Pin 2)
 - Release n_ISP (Pin 1)

Vitalic66, 2020-06-27:
No success at all, tried suggested usb ttl converter from amazon and organized a laptop with real rs232 plus rs232 to 3.3v ttl converter.

Last way is to desolder lpc2134 and put a new one on the pcb. Just ordered. Update will come...

Vitalic66, 2020-07-01:
So i received a new LPC2134, soldered it and guess what, success.

So at least i got another LPC2134 with oem firmware...

closed 2020-07-01 by Vitalic66