UnifiedEngineering/T-962-improvements #191: BT139-600E triac

Blisk, 2020-11-10:
Did anyone find a better replacement for that triac BT-139 600E maybe some 25A triac which will take less heat?
I am also planning to put a digital thermo regulator for that small fan so it will turn regarding a heat of heatsink.

DimaSM338, 2021-07-04:
> Did anyone find a better replacement for that triac BT-139 600E maybe some 25A triac which will take less heat?
> Maybe BTA25-800 or BTA24-600?
> I am also planning to put a digital thermo regulator for that small fan so it will turn regarding a heat of heatsink.

I used - instead of BT139-600E.127 (16A) I installed BTA140-800 (26A).
More details here

https://github.com/UnifiedEngineering/T-962-improvements/issues/210

George-2005, 2021-08-06:
> I used - instead of BT139-600E.127 (16A) I installed BTA140-800 (26A).

I'm also thinking of replacing the triac and install BTA140-800 in my new T-962. But as I've read in [https://github.com/taliesin/T-962-improvements/wiki/Basic-electrical-changes](url), it should be also needed to replace the optotriacs. AFAIK BTA140-800 has the higher Igt than BT139-600E so it seems that the original optotriac couldn't drive it safely. And the new optotriacs MOC3083 are zero-crossing.

 
Unfortunately the "Power Electronics" (mains switching etc.) is not my cup of tea so I'm not sure if the following thoughts of mine are correct:
1. Replace BT139-600E by BTA140-800 for heating lamps (retain original heatsink) - I don't plan to add any other lamp, just to replace triac by more powerful one to reduce its heating
2. Retain BT139-600E for fan
3. Replace both MOC3021 by MOC3083 to improve the BTA140-800 & BT139-600E driving

 
So could someone be willing the review the proposed changes mentioned above and also answer the following questions? 
- Is it so beneficial to use optotriacs with zero-crossing?
- Are the other changes (mentioned by [https://github.com/taliesin](url)) needed? - i.e. remove (short) R16 (2.4 K) and  replace R6 & R8 (1K by 680 ohms)? 
- Why is needed to pullup Pin 1 and 3 of 74hc04 to 3.3V as they aren't floating pins, actually but they are connected to MCU pins 13 & 14? Do such MCU outputs behave as open collector? Or is it something like voltage level shifting?