UnifiedEngineering/T-962-improvements #192: System fan without additional FET - use 4 wire PWM controlled fan

woytekm, 2020-11-11:
This may be interesting to people wanting to do system fan modification: instead of using additional transistor to drive existing fan, you can use 4x4 fan with PWM speed control (4 wire). I have just installed Dell 0W7GXP W7GXP PT4MM 4x4 12V brushless fan from Dell R210II that i had laying around. It barely fits because of it's depth, but does not touch anything. Also, to use this one, you'll need longer M3 screws. 

Red wire - 12V, black wire - GND, blue wire to AD0 test point. Variable speed control works without any additional mods to firmware. 

I think it may be worth mentioning in the wiki.

bgrigoriu, 2021-04-24:
Hi, 
Where do you get the 12V ?

woytekm, 2021-04-25:
Hi, you use existing fan power cable to get GND and 12V. Blue wire from PWM controlled fan should be connected to AD0 test point, which is on oven main board.

bgrigoriu, 2021-04-25:
Ok  But where is à 12V on the PCB ?
Do you know how much currents needs the PWM control of the fan? 
Do you have some other compatible references ?

woytekm, 2021-04-25:
Original fan is connected to oven main board via two wire cable, you get 12V and GND from this cable/board connector. I don't remember connector ID on the board, but you can easily check it yourself. As to current for PWM control, i think that most PWM controlled fans conform to Intel standard, which says that PWM control signal for a fan should not exceed 5 miliamps of sourced current (https://www.glkinst.com/cables/cable_pics/4_Wire_PWM_Spec.pdf).

bgrigoriu, 2021-04-25:
Ok
That's a lot ore clear.
Thanks for taking the time for pupping a newbie.

GitLang, 2021-04-26:
Note, that the oven does not source the pwm current, it sinks it.  PWM drive to fans is an open collector transistor to ground on every one I have ever seen. backed up on the Intel spec which says  "Current sink capability: 5mA required,  8mA recommended "

woytekm, 2021-04-26:
Thanks GitLang, i stand corrected.