UnifiedEngineering/T-962-improvements #194: T-937M Support - Enhancement

FL140, 2021-02-06:
I just got a puhui T-937M reflow oven which is the bigger brother of the T-962 and try to find out if it is possible to support this model by this custom firmware.

Is there any interest in supporting this model?

The advantage of this model are the 3300 W for the heating.
It is also advertised as lead-free soldering oven.

mikeanton, 2021-02-06:
I have interest in this as well, since I also bought one about a year ago.  With rewriting the firmware in mind, I purchased a complete spare set of electronics for the oven, so that I have a test bed, without wrecking the oven.  Given how different this is from the T962 series, I don't think it makes sense to adhere too much to this project, so it would be better to just replicate what the current firmware does, but implement it in a better way so that the PID control of the oven actually works properly.

FL140, 2021-02-06:
Well I haven't checked out the hardware of the T-937M vs. the T-962 jet since the T-962 was no option because it is not suitable for lead free soldering. Have you already done any reserch on this?

Regarding the firmware it would be good if we could keep one code base for all the different models this is the reason why I opened this request here in the first place. 

Having had a look at the hardware would you recommend to go with the current PCB and enhance it or build something new from scratch and just use the sensors/display?

Another interesting topic would be a open source control software, which runs on all major platforms not just Windows, available for the T-937M. Probably using WiFi instead of the Serial to USB connection? I don't think there is a control software for the T-962 in the original firmware, correct?

GitLang, 2021-02-06:
A good start would be tracing the schematic fully and getting a dump of the firmware to disassemble.

mikeanton, 2021-02-07:
@GitLang That's assuming that the firmware isn't protected.  Yes, at some point I'll trace the hardware, alas, I haven't had time as I've been too busy assembling thousands of PCBs.  I wouldn't worry too much about trying to reverse engineer the software once the hardware has been figured out.  It isn't implemented correctly anyhow, so why worry about what it does.

I regularly see temperature swings of +-10C or more during the soak period, which is unacceptable.  It will drive itself into oscillations, by overshooting on temperature, then turning on the cooling fan, which causes an undershoot, rinse and repeat.  It also turns off the convection fan during cooling, which I can't figure out why they would have done that.  This probably creates all sorts of hot and cold spots, and even with the convection fan, the temperature isn't very even in the oven.  Sure, it's much better than the T-962, and it can do lead free, but that isn't saying much.

One thing to note, is that you don't want to connect the exhaust port to another fan that sucks on the oven.  It needs to be vented into free air essentially, or it will cool down the oven and wreck the heating ability it has.  I made the mistake of initially coupling it to an exhaust fan.  Now I loosely couple it by leaving a 3" gap between the two hoses, and running the exhaust fan at a lower speed.

The hardware used on the T-937M is totally different than the T-962 series.  For starters, it uses the STM32 series of microcontroller, rather than the LPC21xx series that the T-962 uses.  This alone would mean that the firmware is going to be significantly different.  I'd probably attempt to use the hardware that's there, at least initially, and only replace it if it doesn't look like it can be made to work.  The one thing that I might do is use a proper thermocouple interface like the mods that we've done to the T-962, as that certainly is a better way to do it.

It does appear like it might have cold junction compensation at least, as it looks like there is a thermistor in there, but of course it is nowhere near the actual cold junction.

What I uses on my T-937M for controlling it is an ethernet to serial bridge (Wi-Fi ones are also available).  This still looks like a COM port on the PC, so it allows me to control the oven without stringing a long cable.  I use this one: [Waveshare Etherenet to RS232/RS485](https://www.waveshare.com/product/iot-communication/wired-comm-converter/ethernet-to-rs232-rs485/rs232-485-to-eth.htm).  In order to get it working, a mod was required to force the DTR line active, as this is what is used to power the optocouplers on the oven.

I personally wouldn't worry about having a common version between the two ovens.  The T-937M profile setting is done in a completely different way than the T-962, and it is an improvement, though still not ideal.  Other commercial ovens just use a few control parameters to set a profile, rather than drawing a profile, the latter seems like a poor to do it to me.  For instance, there might be a slope setting to reach soak temperature, and then a temperature at the start of the soak, and a temperature at the end, with a time period.  The oven then doesn't start timing the soak, until the initial temperature is reached.  The same is done for reflow, where you have a slope value, a peak temperature, and a time.  Reflow timer doesn't start until the temperature has been reached.  None of the Chinese ovens seem to do it this way, so instead, you program it based on time and temperature, but if the oven can't track the programmed shape, then the times are all wrong.  This makes profiling the oven far more difficult than it has to be.

You are correct that there wasn't any control software on the T-962.  They didn't expose an interface for this.  I think someone wrote one for the new firmware though, but I haven't tried it.

FL140, 2021-02-07:
@mikeanton First, thanks for the comprehensive information on what is your experience with the T-937M. 

So the next step would be to reverse engineer the PCB for the schematic. I just got the oven and I am also very busy right now with a project I need to do and therefor getting the K210 running with aws/FreeRTOS. So this is not the highest priority since the oven at least seems to work enough for working with it. 

That said, I will have a look into the oven in about a week when setting it up, probably it is not too complicated to acquire the needed information.
 
 @mikeanton have you already been able to get a dump of the serial communication between the oven and the host PC? 
 Are you running Windows? And do you have information on an existing open source, multi platform (incl. Linux) host software solution where we can integrate this model after analyzing the serial communication protocol? (The protocol is not public by the way, is it?).

FL140, 2021-02-07:
@GitLang do you think getting a firmware dump and disassemble it would reveal any useful information?

FL140, 2021-02-07:
So I took a chance and wrote puhui the manufacturer if they can provide us with the serial protocol and the schematics for the T-937M. Since everyone would benefit if we could make a really good reflow oven out of this together, we get good solution and they can sell more units, win/win. Let's see if this works. I am just not sure if the answer will be fast as Chinese new year is coming up and there is not a lot going on business wise during this time of the year.

GitLang, 2021-02-07:
> @GitLang do you think getting a firmware dump and disassemble it would reveal any useful information?

Yes, definitely. Having full knowledge of where you are coming from is a good foundation for improvement.

GitLang, 2021-02-07:
If you do consider starting again on the hardware, have a good look at the Teensy 4.0 board. It easily usable as a pcb component, and has USB and full Arduino compatibility. A  600MHz ARM Cortex-M7 with float and double hardware floating point unit can take all the worry out of processing power, with several potential programming methods. It is effectively a 1.4" x 0.7" 33 pin chip. with a heck of a lot of i/o, both digital and analogue, PWM outputs, SPI, etc.. All pins are configurable as Interrupts. Anything you want to tag on to it, like displays and keypads and likely to be 3 wire SPI type interface and have several ready written and proved driver modules. A very good starting point.

FL140, 2021-02-07:
@GitLang Well, we will see how easy this will be. If I can get you the dump would you disassemble it and have a look at the code? 

As for the a custom hardware board I will go either with a simple ESP32 variant or if we need the power with a RISC-V K210 module ([Sipeed M1w](https://www.seeedstudio.com/Sipeed-MAIX-I-module-WiFi-version-1st-RISC-V-64-AI-Module-K210-inside-p-3206.html)) as this one was the reason to get the oven in the first place. ;-)

GitLang, 2021-02-07:
I've never attempted programming dual core and AI hardware for real time embedded work - good luck with that :). Actually, I had no idea RISC-V was still going. I assume GCC supports it?.

I'll have a look at a dump for you.

mikeanton, 2021-02-07:
@FL140 I reached out to Puhui when I first got the oven to request a schematic and a copy of the firmware.  They wouldn't supply either.  In fact when they shipped me the spare electronics, the LCD display didn't even have a cable on it, and when I requested a connection diagram for that, so I didn't have to reverse engineer the cable in the oven, they wouldn't even supply that.  So, I suspect you won't get very far in your request.  Chinese companies are pretty guarded, like most companies.  Frankly I was surprised they would even sell me a set of electronics, and they went above and beyond in what they shipped, as they even included parts that I didn't think to ask for like the front panel label, and the serial interface.  I purchased a spare set of heating elements too, as having the oven down if these fail would be very inconvenient.

I can dump the serial traffic easy enough, since this is handled by the ethernet to RS232 software, but I don't think that will help much, without knowing what was done to trigger it at the time.  It is a bit complicated, and the protocol is binary.  For instance, when running a profile from the PC, the oven seems to report the temperature and other status info about every second.  Without knowing the values displayed in the software, it would be difficult to interpret the results.  I do plan to reverse engineer the protocol at some point (I doubt it is public, or that Puhui would publish it), as I would just prefer to use their software initially, even if it isn't the best.  Then at least I have something that can talk to the new version of firmware, and maintain some semblance of reverse compatibility.  I'd probably not make the oven modal as they have done, so that it could be controlled, and monitored even in standalone mode.  Having the oven connect and disconnect depending on the mode it is in, is awkward.

One thing to note is that depending on where you are located, the oven might not be considered electrically safe.  They treat the 240V input as single phase, rather than accounting for the split phase system we have in North America.  So they only fused one of the hots, where both should have a fuse on them, and the oven has to be disassembled to replace the fuse.  I can't remember if the switch interrupts both hots or not, since they don't consider them to both be hots.  I'm planning to replace the switch on my oven with a double pole circuit breaker, and to increase the cable diameter, which also seems to be a bit on the small side for the expected current.

aurelihein, 2021-04-15:
Hey,
I just bought a T962A+ and got a board with a STM32F103
Did you get anything new @FL140
Thanks for your feedback

xnk, 2021-04-15:
Interesting, I wonder what additional changes the + brings? Is the layout of the controller board still similar or is this a complete redesign?

/wj

aurelihein, 2021-04-16:
This a different design since it uses a STM32F103 instead of the LPCXXX

xnk, 2021-07-08:
My curiosity got the better of me last week and I decided to get one of the T-962A+ units: https://twitter.com/meckodon/status/1413260680787075072

The T-937M is most likely using the exact same board, with an additional TRIAC controlling the internal circulating fan.

Very early work in progress, but a fairly trivial board to support I’d say.

mikeanton, 2021-07-08:
Here are scans of the top and bottom of the T-937M MPU board in case they help in comparing to the T-962A+ boards.  If someone wants images with more detail, I'll have to take them with a camera, as these are just from a flatbed scanner, so they are a bit out of focus.

[image]
[image]

xnk, 2021-07-09:
Top and bottom of the T-962A+ board (marked v1.0.2_20170605): The only hardware mod I've done so far is soldering P8, a jumper from 3V3 to BOOT0 pin allowing the MCU to boot into serial download mode over the serial port found in the USB port. This would be enough if you simply want to overwrite the existing firmware.

[image]
[image]

The un-populated TRIAC on the bottom is enabled at the same time as a un-populated middle LED on the front panel marked FAN, which is what led me to believe it would be used for a recirculating fan. The T-937M board is quite a different formfactor though, even though the components used matches up very well (including the old-style ARM 20-pin debug connector, soldered in and everything). The MCU was flash read protected this time around though, so I suspect it's the same on the T-937M. You have to be somewhat careful when dumping the flash (not that there's much in there (looks like a total of 35448 bytes with a sha256 equalling 3863E154974FC04A600692EB4779B1DF7B0FF3275C26739C6483EE010524C36F)

Also lifted the MCU to verify what pins were actually connected:
[image]
[image]

xnk, 2021-07-09:
> Here are scans of the top and bottom of the T-937M MPU board in case they help in comparing to the T-962A+ boards. If someone wants images with more detail, I'll have to take them with a camera, as these are just from a flatbed scanner, so they are a bit out of focus.

They clearly reuse most of the designators, U7 is the MCU (which is the STM32F103RB, right?), U6 should be the EEPROM (is that a 24C64N?).

What I can't really see is what they are using three sets of TRIACS for, in addition to the fourth channel for main heat (connected to the external SSR)?

xnk, 2021-07-10:
I'm fairly sure that it'll be fairly simple to reach the same functionality as I've gotten to on the T-962A+ : https://github.com/UnifiedEngineering/T-962-improvements/issues/209 but I'm still puzzled by the amount of TRIAC outputs here. Also on the T-962A+ somebody must have messed up the layout as the "Heat" LED is actually enabled by turning the "FAN" output on, which isn't supposed to be used on the T-962A+, and the heat SSR is controlled by a different pin. That means that heat request has to be mirrored on two pins to actually provide visible feedback.

mikeanton, 2021-07-10:
> They clearly reuse most of the designators, U7 is the MCU (which is the STM32F103RB, right?), U6 should be the EEPROM (is that a 24C64N?).
>
> What I can't really see is what they are using three sets [...]

I can confirm that the MCU is an STM32F103RBT6, and the EEPROM is a 24C64N as you predicted.

Without taking the oven apart, I can confirm that all output channels on the TRIACs are used, but I don't know what they connect to.  There is also a large external SSR.  There are 6 lamps (elements) in this oven, it's rated for 3300W, as 220V, and it has a convection fan, in addition to the cooling fan.  It could be that a pair of lamps are driven by one of the TRIACs.  Someday, I'll get around to taking the oven apart again, so that I can document how the LCD is wired, and how everything else is connected.  Then I'll be able to fire up my spare boards on the test bench.  Alas, I'm way to busy assembling boards at the moment to do any of that.

xnk, 2021-07-10:
Very interesting! If at some point you have a spare board that you don’t care about the firmware in you will be able to test the alternative firmware on that one . Then we can confirm LCD pin mapping and keypad. Figuring out the puzzling number of triacs is probably the part where these boards differ. Most likely it won’t be possible to know which oven/board we run on (unless this is written in eeprom by own firmware), but this could be an easy config flag otherwise.

Whenever you get the time, no rush. I’ll have some cleanup to do before pushing the stm32 stuff to GitHub anyway.

VasilijLipensman, 2021-08-15:
Dear Werner Johansson! 

Hello. I ask for your help. 
He became the owner of the T-962 reflow oven. 
Is it possible to adjust the maximum operating time 
in the firmware from 8 minutes to 2 hours and more?
This feature will make the oven more versatile!
 
Help please! Thanks and have a nice day ;)

Best regards, Vasily.
St. Petersburg.

PricelessToolkit, 2022-03-19:
Hi, Bought the same oven T962A+  and decided to change the controller, but then I saw this thread and decided to ask if there is already a working firmware?

mikeanton, 2023-10-03:
Check the windings on the low voltage transformer (or look at the output voltage when powered).  I don't think a 110V version is going to be happy at 240V for long.

mikeanton, 2023-10-03:
Also, if that transformer was run at 240V, it's output voltage would be double the norm, so it could have taken out a bunch of stuff in the downstream DC path.

climbamountain, 2023-10-04:
> Check the windings on the low voltage transformer (or look at the output voltage when powered). I don't think a 110V version is going to be happy at 240V for long.

Thanks. I checked the transformer. The 110v input side is open now. Hope that replacing the transformer itself will fix it.

tlf30, 2025-09-02:
I purchased one of these T-937, if there is any testing I can do or info I can provide, please let me know. 

I am going to get started on reverse engineering the current serial protocol for my own use. IDK how hard it would be to grab the firmware from the board, but if you have guidance on it I can try.

mikeanton, 2025-09-03:
Someone already reverse engineered the protocol for the T-937 here: [https://git.s-ol.nu/t937-serial/-/](https://git.s-ol.nu/t937-serial/-/).  There is even a nice user web interface for talking to the oven, that I haven't fully explored, but it looks to have some nice functionality already.

So perhaps you can use this for a starting point of what you want to accomplish.

s-ol, 2025-09-08:
> Someone already reverse engineered the protocol for the T-937 here

that'd be me 😉 I've had notifications on for this thread for a while.

It's unfortunate that the serial protocol doesn't allow turning on or off the internal convection fan, otherwise it'd be a viable alternative to control the oven using an external PID controller (e.g. as a JS website via WebSerial).

@xnk do you or anyone else by chance have a copy of the reverse engineered T-96A+ schematic? The link on this wiki page is dead:

https://github.com/UnifiedEngineering/T-962-improvements/wiki/T962A-Controller-Schematics

I still haven't opened my T-937M up but would love to get this working, but if there's no way to dump the original firmware image, I'm a little worried about bricking my my device. I wonder if a compiled firmware blob is too much of an ask for Puhui - as mentioned earlier in this thread, they've not been very cooperative IME either.

mikeanton, 2025-09-08:
@s-ol When I bought my oven, I ended up getting a bunch of spare parts for it, including a control board.  At the time Puhui sold these for $70 USD plus shipping, so that might be an option for you if you want a test bed for development.

You can try to dump the firmware image, but if they programmed the security bits to protect the firmware, then you will just read garbage.  I don't know if they did this though, so there is a slim chance you could maybe read the firmware.

One of the things I've never understood about how this oven operates, is that when the cooling fan is turned on, the convection fan is turned off.  I'd have thought that it would make more sense to just keep it running all of the time.   After all, why would it be bad to have more air movement in the oven?

tlf30, 2025-09-09:
I ordered a Bus Pirate. I’m going to attempt to extract the existing firmware image.
If you have ideas on a template project I can flash, I can start to figure out which pins do what.

mikeanton, 2025-09-09:
I've not worked with these ST micros much either, so I'm afraid I won't be much help there.  I would tend to look at the PCB and figure out what does what from there.

s-ol, 2025-09-26:
@sur5r oh, I was under the impression that the memory was somehow copy protected! If the firmware can be dumped (and therefore restored) that makes hacking on the firmware a lot less "risky"

tlf30, 2025-09-26:
@sur5r are you able to send me a copy of the T-937 firmware? If so, I'll start decompiling it. 
I'm also curious on the process you used to extract it. I've not opened mine up yet, but I did get a Bus Pirate 6. Once I get back home in a couple months I want to start playing with the firmware on mine. 

My email is tlfalaska@hotmail.com, feel free to send me the firmware if you want and I'll get started digging though it. 

Thanks,
Trevor

sur5r, 2025-09-26:
Extraction was straightforward. The 20-pin connector is almost ARM-JTAG, except it's only wired for SWD. IIRC TDO is missing and some TX pin is routed there, no idea why. I used a Flipper Zero to connect to it.

I looked at it a bit in Ghidra and it looks complete enough. It's based on uC-OS/II which I found surprising. It helps making sense of things, though.

tlf30, 2025-09-26:
@sur5r  good to know, thank you!

If you get time to send me the binary, I'll take a look, otherwise I'll grab an image in late November when I'm back home.