UnifiedEngineering/T-962-improvements #196: T962A - Display problem after firmware upgrade

jensgulow, 2021-02-14:
See #115 and #186 .... 
The solution described in #186 doesn't work.

Must be a timing problem .....
From time to time I have correct signs on the right half of the display, next time blank screen, then signs on the left side ....
But all this is randomly.

For me it seems like a timing problem with the display.

PS: No problems with the original firmware - that meens the display works properly.

jensgulow, 2021-02-14:
I've tried other versions down to 0.4.0 ...... same issues.
Sometimes now normal displaying for half a second ... then overlay of the boot image in the upper half of the display ......

jensgulow, 2021-02-17:
Maybe i know the answer ..... there is a new kind of display in my oven 12864-7 V2.0
The code is written for ovens with the 12864D1. This display has another pinout. also the CS is active low. On the 12864-7 CS is active high....

But i am a noob in c-code. so no chance to change the source - code ;-(

xnk, 2021-02-18:
I wonder if they made some slight change in the original firmware to support this new display or if it in fact compatible (but a bit slower). If it is the same original firmware maybe the display timing just have to be slowed down a bit. But it sounds like the chip selects have changed?

jensgulow, 2021-02-18:
How can i help to clear this up?

jensgulow, 2021-02-18:
... and how can i slow down the display timing?
I would like to understand the source files (a little bit ;-))

jensgulow, 2021-02-18:
I've found this manual for this kind of display .... maybe it uses another chip (Control IC: AX 6107/AX6108)....

https://datasheet4u.com/datasheet-pdf/ETC/TS12864-7/pdf.php?id=614229

[image]

xnk, 2021-02-18:
Looks like a “compatible” chip (KS0108 has been around “forever”), but maybe this new one is a bit slower.

The corresponding part is this: https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/lcd.c#L201-L349 <https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/lcd.c#L201-L349>

The WriteCmd and WriteData functions is what talks to the controller. Try extending the “delays” (formed by reading FIO1PIN repeatedly, so simply adding more FIO1PIN reads will extend the delay). The fact that it “almost” works tends to indicate that the timing margin is slightly negative.

jensgulow, 2021-02-19:
I've tried this.... no change... :-(Von meinem Huawei-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: Werner Johansson <notifications@github.com>Datum: Do., 18. Feb. 2021, 22:48An: UnifiedEngineering/T-962-improvements <T-962-improvements@noreply.github.com>Cc: jensgulow <jensgulow@aol.com>, Author <author@noreply.github.com>Betreff: Re: [UnifiedEngineering/T-962-improvements] T962A - Display problem after firmware upgrade (#196)
Looks like a “compatible” chip (KS0108 has been around “forever”), but maybe this new one is a bit slower.

The corresponding part is this: https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/lcd.c#L201-L349 <https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/lcd.c#L201-L349>

The WriteCmd and WriteData functions is what talks to the controller. Try extending the “delays” (formed by reading FIO1PIN repeatedly, so simply adding more FIO1PIN reads will extend the delay). The fact that it “almost” works tends to indicate that the timing margin is slightly negative.

xnk, 2021-02-19:
Actually now that I look at the data sheet the only thing that really is different is the Address Set-up time (tASU) which is longer in the AX6108 controller. If you try putting a few (like five-six) FIO1PIN between line 207 and 208 (between RW pulled high and E pulled high) and line 276 and 277 (same thing), does that help at all? It looks like the code might actually violate even the KS0108 timing there.

Everything else looks identical (as in carbon copy) in that datasheet.

jensgulow, 2021-02-19:
Hello Werner,

i’ve tried this too. But no change.
I’m not sure if there’s everything okay with the Hardware …… I’ve double checked all solderings (they’re Looking good) ……

Let me discribe my experiences:
When i start the oven, then ramdomly happens the following: blank screen (80%) Text only on the left side (CS1??? 9%) only Right side (CS2??? 9%) both sides (max 2%)

When there is text on the screen it is always like this:  first  i see the main screen (Right side, left  side or both sides) …. Then the welcome bitmap appears – but only in the upper half…. Just 1 second later the bmp disappears and a „duplication of the lower part“ appears in the upper half. But there are „empty“ lines (nr. 3, 11, 19) in the upper part…. And from there is a slight „flickering“ in the upper part …..

I have no clue if this is usefull for you, but i hope for help.

Thanks alot for now!

Jens

Von: Werner Johansson
Gesendet: Freitag, 19. Februar 2021 18:39
An: UnifiedEngineering/T-962-improvements
Cc: jensgulow; Author
Betreff: Re: [UnifiedEngineering/T-962-improvements] T962A - Display problem after firmware upgrade (#196)

Actually now that I look at the data sheet the only thing that really is different is the Address Set-up time (tASU) which is longer in the AX6108 controller. If you try putting a few (like five-six) FIO1PIN between line 207 and 208 (between RW pulled high and E pulled high) and line 276 and 277 (same thing), does that help at all? It looks like the code might actually violate even the KS0108 timing there.

Everything else looks identical (as in carbon copy) in that datasheet. 

> 
> 
> I've tried this.... no change... :-(Von meinem Huawei-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: Werner Johansson <notifications@github.com>Datum: Do., 18. Feb. 2021, 22:48An: UnifiedEngineering/T-962-improvements <T-962-improvements@noreply.github.com>Cc: jensgulow <jensgulow@aol.com>, Author <author@noreply.github.com>Betreff: Re: [UnifiedEngineering/T-962-improvements] T962A - Display problem after firmware upgrade (#196) 
> Looks like a “compatible” chip (KS0108 has been around “forever”), but maybe this new one is a bit slower. 
> 
> The corresponding part is this: https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/lcd.c#L201-L349 <https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/lcd.c#L201-L349> 
> 
> The WriteCmd and WriteData functions is what talks to the controller. Try extending the “delays” (formed by reading FIO1PIN repeatedly, so simply adding more FIO1PIN reads will extend the delay). The fact that it “almost” works tends to indicate that the timing margin is slightly negative. 
> 
> > 
> > 
> > I've found this manual for this kind of display .... maybe it uses another chip (Control IC: AX 6107/AX6108).... 
> > 
> > https://datasheet4u.com/datasheet-pdf/ETC/TS12864-7/pdf.php?id=614229 <https://datasheet4u.com/datasheet-pdf/ETC/TS12864-7/pdf.php?id=614229> 
> > <https://user-images.githubusercontent.com/5845040/108418099-0fdc6c80-7231-11eb-91f2-ebf90d5321db.jpg> 

—

Diese E-Mail wurde von AVG auf Viren geprüft.
http://www.avg.com

xnk, 2021-02-19:
That’s really, really odd! Does the display work if you restore the original firmware? I wonder where they have connected the reset pin on the new display, as there’s no obvious reset output pin from the MCU? Could it be that the issue is the status read? That could be replaced by a long delay, at least for test. I’m not sure how smart the original firmware is, maybe it just waits a long time to make sure the controller is ready.

Also, can you share a video of how the display looks like when you’re actually getting the flickering described?

jensgulow, 2021-02-19:
Hi,

the display works with the original firmware.

Here is a video of the display-problem (after about 20-30 blank screens on start).

Maybe it's about this obvious Reset Pin???

https://user-images.githubusercontent.com/5845040/108562770-a9bf1a80-7300-11eb-81f2-273d9f006c17.mp4

jensgulow, 2021-02-19:
Here is a schematic of the T962A with the connection of the RST-Pin ...
[schematic-T962.pdf](https://github.com/UnifiedEngineering/T-962-improvements/files/6012964/schematic-T962.pdf)

jensgulow, 2021-02-20:
Just for myself..... the resistor connected to the reset pin is an pullup, and the capacitor charges on power up....right?
So that means for a short time on startup the reset pin is low, after this back high ..... 

May be the code needs a extended delay on startup? But how to do that?

xnk, 2021-02-20:
I agree that the display looks odd indeed. A lot of things looks alright though so I wonder if it’s either readback of busy flag or simply a reset happening for too long as you were thinking. Looks like that it would be a (small) RC-delay on the reset pin. As a quick test, try to add the LCD_WriteCmd(LCD_RESET_STARTLINE); and LCD_WriteCmd(LCD_ON); commands at the top of LCD_FB_Update, there’s no other initialisation done really. If that doesn’t make a difference the next step would be to take the read part out from the two write functions and assume the controller is ready after a long enough delay (a for loop over a FIO1PIN read would work, say some 10000 iterations would make a crazy long delay in this context, milliseconds I would guesstimate).

jensgulow, 2021-02-20:
Okay, with adding these 2 lines i get a slightly better outcome.
void LCD_FB_Update() {
	LCD_WriteCmd(LCD_RESET_STARTLINE); //***
	LCD_WriteCmd(LCD_ON); //***
	for (uint32_t page = 0; page < (FB_HEIGHT >> 3); page++) {
		LCD_WriteCmd(LCD_RESET_X + page);
		LCD_WriteCmd(LCD_RESET_Y);

		for(uint32_t i = 0; i < 64; i++) {
			LCD_WriteData(FB[page][i], 0);
			LCD_WriteData(FB[page][i + 64], 1);
		}
	}
}

See video ..... no persistent blank screen in >90% ..... but all in all it looks still very odd ;-(

https://user-images.githubusercontent.com/5845040/108602776-235c1480-73a4-11eb-94a3-80bf2418c834.mp4

GitLang, 2021-02-20:
I don't know if it will help, but there is an Arduino driver library for this display here :+1: 

https://playground.arduino.cc/Code/LCD12864/

It does menmtion the ST7920 chip though, so it may or may not be compatible. There are certainly plenty of hits if you search with 
TS12864 Arduino

There are several code snippets on Pudn, including a mention of the -7 display, but I can't find my pudn login atm.

xnk, 2021-02-20:
The ST7920 is a different kind of controller, in that case you only have one chip select as that single chip can handle wider displays on its own. I assume nothing has really changed in the original firmware, but maybe they did change something.

xnk, 2021-02-20:
OK progress (of sorts)! The odd thing is that it seems to mix up top/bottom despite the LCD_RESET_X and _Y commands being issues (set_address and set_page really).

How does the screen change if you only scan 32 out of the 64 iterations in the loop, or only 4 out of the 8 pages, or commenting out the left or right write data (cs0 or 1)? The fact that BOTH sides gets confused makes it look like command issues (as data is issued separately to the corresponding chip selects but commands are sent to both simultaneously).

jensgulow, 2021-02-20:
left or right write data: ..... works like it should ....(only text on the left or right side)....but same problem with the upper part of the screen ....;
Only 32 iterations: ..... same situation like discribed, only in the left half of both sides;
[image]

4 out the 8 pages: only readable text in the upper half (with the knowing patterns like discribed before: first main screen, then same screen with empty lines 3, 11,19,27
[image]

xnk, 2021-02-21:
The more I look at those picture the more I’m convinced that there’s something wrong with the D2 data pin (third data bit). That would explain why top/bottom is messed up and why every eighth pixel misbehaves (each “page” is a 8 pixel high part of the screen). Can you attach another wire in parallel with D2 to make sure your flat cable is still intact?

jensgulow, 2021-02-21:
Okay .... problem solved :-)

First of all many thanks to you, Mister Werner Johannson for your help.
Your last post pointed me in the right direction. The flat cable is okay - double checked all....
Then i took a closer look at the solderings of the MCU ....... and ther was a tiny solder bridge between PIN 8 and 9. After removing the solder bridge everything went fine.

The new firmware workes like a charme.

Ones again many thanks !!!!

PS: here some pictures of "the problem" and how it could be solved ....

[image]
the solder bridge
[image]

solder bridge removed ..... problem solved ;-)
[image]

xnk, 2021-02-21:
Oh wow, that’s one annoying issue! As the original firmware most likely wasn’t using that pin as an output it would work perfectly there... Great that you got it sorted, and thanks for reporting back!

So does the display work every time you power on now without all the changes to the lcd code, or do we need to relax the timing for the new displays you think?

jensgulow, 2021-02-21:
Yes, the display workes fine without any changes in the code... Von meinem Huawei-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: Werner Johansson <notifications@github.com>Datum: So., 21. Feb. 2021, 23:29An: UnifiedEngineering/T-962-improvements <T-962-improvements@noreply.github.com>Cc: jensgulow <jensgulow@aol.com>, Author <author@noreply.github.com>Betreff: Re: [UnifiedEngineering/T-962-improvements] T962A - Display problem after firmware upgrade (#196)
Oh wow, that’s one annoying issue! As the original firmware most likely wasn’t using that pin as an output it would work perfectly there... Great that you got it sorted, and thanks for reporting back!

So does the display work every time you power on now without all the changes to the lcd code, or do we need to relax the timing for the new displays you think?

closed 2021-02-22 by jensgulow