UnifiedEngineering/T-962-improvements #197: T962

Blisk, 2021-02-16:
I have upgraded my T962 with pinoxi controller and uploaded UnifiedEngeneering firmware, but it doesn't work.
Owen doesn't heat and fan light blinks.
Any idea what can be wrong.
Here are pictures and video.
https://nextcloud.povej.net/index.php/s/SB7MTnJzfBQXqoM

xnk, 2021-02-18:
That’s odd. It’s normal for fan light to flash (trying to run the fan real slow). Is the “lamp” indicator on at any time? Are you getting good temperature readings from the thermocouples?

Blisk, 2021-02-18:
Both fan and heat light flashes all the time except at the end when fan runs to cool down.
I found out where problem was, bad connection on heaters, I fixed that. Connector was bad on 1 pin.