UnifiedEngineering/T-962-improvements #199: UE firmware upgrade is well worth it!!!

markingle, 2021-02-22:
I finished the upgrade this weekend on my T-962 and I am very pleased with the result.  Here are the modifications:

- Installed baffles between infrared elements and oven tray
- Dallas One-wire sensor
- Upgraded latest firmware from UE

Below are before and after measurements.  Green plots are after the upgrade in the left, center, and right of oven tray.  I will try to get a step-by-step created in the next two weeks.

 
[image]

xnk, 2021-02-22:
That's great!!
The baffles does seem to improve things from what I've read. I haven't gotten that far yet as I've had very good results with the T-962A only modified with onewire thermocouple interfaces.

markingle, 2021-03-09:
Here's a link to the google doc showing my steps for the upgrade.  Its about 90% done so I thought I would share it now to get some feedback.  I am trying to show the step-by-step with as much detail as I can.

https://docs.google.com/document/d/1SC2ZqJGnKhm--BCxHBpJHlhu_njhTnSND5RYPwzEYL4/edit?usp=sharing

GitLang, 2021-03-09:
@markingle - FYI the attachment on your message of 23rd February shows waveforms as expected here, online to Github, but the subscribed email of this messaged I received by email has a photo of a UPS receipt/shipping label!!. It has a name and address on so maybe worth looking into, though it's probably a Github error. "Like Yours" mean anything to you?.

Looks like your development is going very well, nice work. What prompted choice of the ESP32, cost?. I'm very taken with the Teensy 4.1 at the moment, very powerful but a fair bit more expensive and I need 64bit Doubles in hardware. The Arduino front end finally drove me crazy and I found PlatformIO too. Wow, very very impressed with it.

markingle, 2021-03-09:
Hi!  Thanks for letting me know!  Can you forward the email to me?  I
cannot find what you are referring to....I probably grabbed the wrong
screenshot.

I am doing the ESP32 mainly because thats what works for my customers in
prototyping their projects.   Mainly I have positioned myself well to
rapidly get a prototype in hand to confirm the product idea.  Beyond that
the T962 helps me do a small batch (25 -30) for beta testing in the field.
The Teensy is a powerful and useful platform as well but I have not had
time to work it into my portfolio yet.  Upwork keeps me busy.

Thanks again for letting me know about the picture with PII on it!

Mark