UnifiedEngineering/T-962-improvements #200: incorrect temperature measurement of left element

tiborrr, 2021-03-04:
I've just finished installing the temperature junction and that actually works. but now my left temperature sensor record 800 degrees. Do you have any idea where this could have gone wrong?

What i have tried so far:
Disconnect coldjunction
[image]
[image]

Thanks in advance!

GitLang, 2021-03-04:
Maybe you disturbed the left thermocouple at the terminal block?. If the wiring looks ok, I would try disconnecting the left thermocouple wires at the terminal block and connecting a standard K type thermocouple there. At least it will rule in or out the thermocouple. If you don't have a spare K type thermocouple, then try swapping the left and the right thermocouples and see if the fault remains on the left channel or swaps to the right.

tiborrr, 2021-03-04:
Just tried swapping out the left and right thermocouples but the issue stays on the left side.

GitLang, 2021-03-04:
OK, so the problem has to be with the board or firmware. Is it the cold junction modification you did?. Get a small magnifying lens like a loupe and really carefully examine the areas you worked on, and the tracks and solder joints associated with the thermocouple circuitry. Also possible that you may have dropped a tiny piece of solder between two pins of a chip or left a single strand of wire shorting pins or chip legs out. Make sure you are in really good light, preferably daylight, or get someone to hold a torch on the board. I'm sure you know parts of the circuitry are joined to main electricity and you should be certain you turned it off and pulled the plug out.

I've just looked at your photo. The resistor you soldered between those two pads is (excuse me for saying so) but really badly solder. When solder pulls into a shiny ball like that it means it hasn't flowed onto the component. Actually it looks like it hasn't flowed onto the component or the pad. This is usually down to a lack of flux and "applying molten solder" to the work, rather than heating the work and letting the work melt the solder. Clean your iron tip and immediately tin it. Hold the component in tweezers, and with a little solder on the iron, touch the iron between the resistor end and the pad. Count to 2 or 3 and you should see evidence of solder starting to flow near the pad and component. That is the time to apply a little solder slowly with your third hand... Repeat for the other resistor end. You should see the solder form a concave meniscus between the pad and the component. A convex blob like in your picture means the solder hasn't flowed.

Big apologies if you already knew all this stuff, it help somebody though. Soldering is a fundamental skill which everybody who dabble in this stuff should be proficient at before going anywhere near firmware changes, modifications, etc.

GitLang, 2021-03-04:
One last thing, find up an old toothbrush and get a can of :

Electrolube 400 ml Aerosol Flux Remover for PCBs aka "Fluxclene". Part number is FLU400DB.

Spray on, quick scrub with the brush, spray some more with the board vertical and let it drain and dry. If you look at areas like between the green terminal block and that 8 pin chip, it is so dirty and covered in old flux that it is hard to tell if there are shorts or not, but I bet whatever you like there are small spheres of solder in it. You need to examine solder joints and you have to clean the flux off before you can do that properly. It's not to leave it on long term sometimes anyway.

GitLang, 2021-03-05:
What does "eyes emoji" mean?

tiborrr, 2021-03-05:
@GitLang thanks for taking the time and sending such a detailed explanation. Always appreciated!

Second I will order the flux remover and some solder wick to undo my mistakes to see if the situation improves. You are right in the sense that I'm not very skilled yet in soldering. While I do know what I should be doing this didn't go according to plan. 

The eyes mean that I'm reading your insights carefully.

GitLang, 2021-03-05:
I'm still getting used to 'emoticon' rather than 'smilie'. Now it's changed again! I am slowly progressing to Grandpa Simpson :)

I couldn't see your cold junction sensor. It should be in good contact with the terminal block body. I used a drop of superglue.

tiborrr, 2021-03-05:
Reverting all my steps didn't result in any improvements and also ruined one of the pads where the risistor was attached. 😑 
[image]

My guess is that indeed there might be a splat of solder or flux somewhere that causes these weird readings on the temperature circuit. 

Do you have tips on how I should check with my multimeter where I should start my search?

tiborrr, 2021-03-05:
> I'm still getting used to 'emoticon' rather than 'smilie'. Now it's changed again! I am slowly progressing to Grandpa Simpson :)
> 
> I couldn't see your cold junction sensor. It should be in good contact with the terminal block body. I used a drop of superglue.

Knowledgeable grandpa Simpson it is then🧓

GitLang, 2021-03-06:
Trying to find that kind of fault on a microcontroller board with a multimeter is tricky. There are so many combinations and I'm not sure of the dangers of a multimeter putting out more volts that the 3.3V the cpu is designed for. If you do go that way, I wouldn't randomly check point to point, I would put the COM or ground lead of the multimeter on the boards 0V and check round with the red lead of the meter. In particular I would check every node on the thermocouple circuits, checking left circuit then right circuit on each node looking for significant differences in reading. They should be more or less the same. If you can't find anything that way, clean the board all over thoroughly with the electrolube and brush. From memory, I don't think there are any relays or switches on the board, so the cleaner shouldn't do any harm. Then get your light and loupe and settle in for a long hard look at it. If you find anything or not, fit the board back in and test it, the cleaning may have dislodged something you didn't spot. Then I would put the meter on DC volts range, black lead on circuit 0V, and with the schematic in front of me, have a look round the thermocouple areas for any anomalous voltages. DONT stray onto the Live AC Mains side of the optocouplers. I'm not sure what order it should come in the fault finding, but at some point you should double check the firmware you have used. Myabe download a fresh copy of it and check the SHA or CRC or whatever if available. Good luck.

GitLang, 2021-03-09:
Well done man. When you say no cold junction, is it because you feel you can't implement it now because of that missing pad?. There are ways round that :). Easiest may be to follow the track back from the ex-pad until you reas a decent sized via or another component pad. Solder the resistor to the pad and tack a small wire on the other end of the resistor and take it to wherever it originally went. Or you can do the reverse of soldering one end of the resistor to it's original place then tacking a wire to the resistor and to wherever you can on the track from the lost pad. You can even tack it to a track if you carefully scrape the green solder resist off and then tin the track first. Should then put a drop of hot melt glue on all of the areas you've worked on - it really does protect delicate wires and resistor joints.

tiborrr, 2021-03-12:
Well the trace came off together with the pad as you can see in my image. However, you are totally right. It just requires a pad with 3.3 volts. I will need to dive into the circuit design in order to find a pad that will work. For now I'm happy the oven works. I already successfully baked 6 PCBs. The score is a 4 out of 6 success rate, which for now is good enough. Especially since I have some very small QFN sensors are hard to get right the first time around.