UnifiedEngineering/T-962-improvements #201: Executable file for Reflow Controller software

zian31, 2021-03-09:
I've one question concerning the "Reflow Controller software" used here by some users :
Source Code : https://github.com/duncanellison/Reflow-Controller-T-962-
Where can I find a executable version for windows ?
And is there anonther one also for Linux ?
I don't find executable version in the source code link.
Thx

GitLang, 2021-03-09:
I don't know the provenance of the attached but I thought I got it from here somewhere. To build theat source you need a licensed library.

[Reflow Controller.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/6109088/Reflow.Controller.zip)

zian31, 2021-03-09:
> I don't know the provenance of the attached but I thought I got it from here somewhere. To build that source you need a licensed library.
> 
> [Reflow Controller.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/6109088/Reflow.Controller.zip)

Thanks a lot ! It's perfect.

zian31, 2021-03-25:
> > I don't know the provenance of the attached but I thought I got it from here somewhere. To build that source you need a licensed library.
> > [Reflow Controller.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/6109088/Reflow.Controller.zip)
> 
> Thanks a lot ! It's perfect.

zian31, 2021-03-25:
Hi,
I've just tried the Reflow Controller.exe today, but only the top right value shows the target temp in REFLOW mode ( and there is no factor 10 in my case), all the others values are at 0 in REFLOW mode : Air(L) and Air(R) & Fan & Heat & TC1 and TC2 (I don't connect Tray TC3 and TC4 in 1-Wire). I can only see the values in the case on bottom right :  250.2,  222.2, 229.8,   0.0,   0.0,   50, 226.0,    0, 255,   24.2, REFLOW ...
And the graph shows nothing : no curves.

Maybe it's because I don't connect Tray TC3 and TC4 in 1-Wire ?

zian31, 2021-03-26:
Here are one picture of the "Reflow Controler.exe" tou sent me, and another one find in this forum,
Can someone send the other .exe file ? Thanks in advance.

[image]

[image]

GitLang, 2021-03-26:
What version is this exe?. It should show a splash screen  with the version number and a bloke in a suit pressing a button or something.

zian31, 2021-03-26:
> What version is this exe?. It should show a splash screen with the version number and a bloke in a suit pressing a button or something.

The first picture shows the version you send me (thanks for it), the curve is not displayed, and temps are all at 0 during reflow.
The second picture shows the version used by others members, here rafensb'version (radensb : perfect version, Bibbbi : version with just a pb of Tx10) on github (with legend : Time, Air...).

DimaSM338, 2021-06-27:
A similar problem, charts are not entered. I am looking for a corrected version of the program.

Did you manage to solve the problem?

zian31, 2021-06-27:
No, so maybe I will program a new software to do it : from UART data to a graph, with also maybe possibility to upload new custom profile to the oven firmware which has to be modified for that too...

DimaSM338, 2021-06-27:
I recompiled from the source code which is here the program. Got a new exe file. The interface has changed in the updated program. A legend has appeared. But the schedule is never drawn. Tomorrow I will try to update the firmware in the T-962

bgrigoriu, 2021-06-27:
Hi,
I would also need the exe if it works for you. can you publish a link ?

DimaSM338, 2021-06-28:
[https://drive.google.com/file/d/13X3Nt3af_pHkiYeQftpG3hy_G_ukFU7v/view?usp=sharing](url)

Here is a link to the file. Try it. It will be very interesting for you to work correctly. Write an answer please.

bgrigoriu, 2021-06-28:
Hi,
I downloaded your build.
I compared with the exe fiel that is available on GitHub and with a build I generated today using Visual studio and the sources also available on GitHub.
They all work the same.
You can select a profile and start the reflow but the printing of the temperatures are not well organized ang the temperatures are not  correctly displayed.
Obviously some work is needed on the coding.

Sincerely,

Bogdan

GitLang, 2021-06-29:
Dima - I cant access that page, it doesn't seem to be a valid URL. I get the error -
This page can’t be displayed

•Make sure the web address lsp://1 is correct.
•Look for the page with your search engine.
•Refresh the page in a few minutes.

wulfmans, 2021-06-29:
link is ok here

zian31, 2021-07-01:
I've tried today with your new .exe : I got the same results : legend apperas but no graph with oven working and profile 1 for example.

I think there is something bad about end of lines received with UART from oven ;
there are only LF (0A) instead of CR LF (0D 0A) sent at each end of line by the oven firware => maybe that's why datas showned are not align in this software and also bad parsing of datas => no graph ?

You can test this simple change :
frmMain.vb
Dim Terminator() As Char = {vbCr, ","} -> {vbLf, ","}

DimaSM338, 2021-07-02:
It was very interesting for me to understand how the program works. But my knowledge is not enough. Need help from more experienced ones.

I tried what you said:
    ' Dim Terminator() As Char = {vbCrLf, ","}
    ' Dim Terminator() As Char = {vbLf, ","}
    Dim Terminator() As Char = {vbCr, ","}
and even
    ' Dim Terminator() As Char = {vbCrLf}
The result is negative, the program does not work.

I deliberately violated the work of the program by this in order to understand - the data is correctly transmitted from oven to the program. Since, for example, the set temperature is displayed correctly. 

Аt this stage it seems to me that there is a problem in type conversion. The data from the oven comes in the form of char bytes, which must be converted to float.

I am very weak at VB. I will leave my research here, maybe it will help someone.
I was analyzing the data flow
[image]
 
every line that the oven sends with this code

if (!(mymode == REFLOW_STANDBY && standby_logging == 0)) {
		printf("\n%6.1f,  %5.1f, %5.1f, %5.1f, %5.1f,  %3u, %5.1f,  %3u, %3u,  %5.1f, %s",
		       ((float)numticks / TICKS_PER_SECOND),
		       Sensor_GetTemp(TC_LEFT),
		       Sensor_GetTemp(TC_RIGHT),
		       Sensor_GetTemp(TC_EXTRA1),
		       Sensor_GetTemp(TC_EXTRA2),
		       intsetpoint, avgtemp,
		       heat, fan,
		       Sensor_GetTemp(TC_COLD_JUNCTION),
		      modestr);

breaks down correctly into - tokens. There are 11 of them. (0-10)
        0       1           2            3            4                5             6      7         8        9       10
# Time,  Temp0, Temp1, Temp2, Temp3,       Set,     Actual, Heat, Fan,  ColdJ, Mode

tokens (5) is integer. And it displays correctly. The rest are floats. They are not displayed on the graph.

Maybe there will be time I will try to rewrite some of the code in the oven so that the oven sends integers.
if (!(mymode == REFLOW_STANDBY && standby_logging == 0)) {
		printf("\n%6u,  %5u, %5u, %5u, %5u,  %3u, %5u,  %3u, %3u,  %5u, %s",

zian31, 2021-07-02:
Good work ! I think that you are close to find how to upgrade this VB software, it's not my case : I don't know how to compile it.
In my case, it will be easier to code a new graph software in C++ to also do something new : possibility to upload new custom profile to the oven firmware from this C++ software (with in bonus also the possibility to save it in the oven EEPROM : in custom#1 or  custom#2 profiles).