UnifiedEngineering/T-962-improvements #203: Still applies?

smdjeff, 2021-03-29:
Given the majority of this code is 6 years old, and not yet owning a 962 I'm wondering if they still using the applicable hardware?

xnk, 2021-03-29:
It seems that even new stock of the T-962 and T-962A uses very much the same construction, and the same modifications are still required.

LokiMetaSmith, 2021-12-12:
Can confirm, mods are still necessary. Tried running without mods and the high temp lead free profiles wouldn't keep up with measurements from an external hand held TC probe.

xnk, 2021-12-12:
Correct, the only ones that are confirmed to be different are the T-962A+ and the T-937M, which uses a STM32 controller instead of the LPC2134/01. The thermocouple interface is still equally bad, and the T-962A+ makes things even worse when reversing the airflow through the oven (to support venting out the back instead of out the bottom)