UnifiedEngineering/T-962-improvements #204: Thermocouple temps 20 degrees above cold junction

Hengy, 2021-05-02:
I recently installed the firmware, and just installed a DS18B20+ cold junction sensor. The cold junction seems to work fine. However, my thermocouple temps are way off - reading 44 degrees when the cold junction says 21 degrees. 21 is close to my room temp. I have used an external thermocouple meter, and the inside of the oven reads 22 there, so it seems to be the thermocouples that are off. This happens right when I turn it on, so there is no way it's been heated.

Has anyone every seen this, and does anyone have an explanation/fix for it?

I have yet to run the oven to see if the temperature difference persists throughout the entire temperature range.

xnk, 2021-05-08:
This is not unheard of using the built-in thermocouple amplifiers, they are not very linear at all. Try heating the oven to two different temperatures (one close to the desired profile peak temperature and one somewhere above 100C) just to see if it’s more accurate there. That’s the region where you’ll want the most linear behaviour, room temperature will never be a good data point for that TC interface.

vjalle, 2022-01-25:
Replacing the thermocouple amplifier should be added to the list. Ditch the original TS27M2 or similar and replace it with a precision OPA that has maximum offset less than 100uV. LTC2051, NCS2333, AD8552 and the like. No other mod required, just remove the original SO-8 package and solder in the new one.
Any device with low offset, 3.3V operation and rail-to-rail input/output (at least to the negative rail) would work.
Offset problem is gone forever!