UnifiedEngineering/T-962-improvements #206: Building in VScode

JulesGren, 2021-05-16:
I know there is a hex already, but as I may wish to customise the code, I would like to build the bin/hex myself.
Has anyone ported this into VScode or PlatformIO within VScode?

I prefer all my builds in the one environment.

I would be OK with building in windows, but not really looked at this - anyone done either method able to pass on their knowledge?

GitLang, 2021-05-16:
No, I didn't know about PlatformIO at the time I was playing with it. I used LPCXpresso which is NXP own free offering that you can sign up for and download. Works well.

JulesGren, 2021-05-17:
Thanks.

Discussing this on another forum MaxGerhardt has taken a fork off this stream, and translated it to load and build in VScode with PlatformIO - Fantastic work from Max!

I have forked a copy off his and it builds fine, not able to test the upload yet as I need a 3V3 UART (FTDI232) - all mine are hard wired to 5V.

GitLang, 2021-05-17:
That's good news, thanks for passing it on. It cloned fine into PlatformIO and it nearly built!. It got almost to the end and failed on "arm-none-eabi-objcopy" with the very helpful message "Error 1"

JulesGren, 2021-05-17:
Did you download the files as a zip, or did you load directly into vscode/platformIO via the github addon?

I had loads of failures when zipping, but once I cloned it then loaded the clone copy directly into vscode it was fine.

GitLang, 2021-05-17:
I've tried both ways. I just loaded it via Github again and I'm still getting the same error. I've made sure everything is updated, but it's persistent. In the console build output, at the top, it says more detail is available using the -v or --verbose option, but where?. As I say, it's building the .elf ok, but failing when trying to build the .hex.

JulesGren, 2021-05-18:
Sounds like an issue with your install of VScode and platformIO, as all the code and script works fine for me.

One thing, to build I use the tick at the bottom towards the left to build, sometimes you can select a build in a debug state and if you haven't an emulator for the target build it doesn't work.

It could be a missing dependency - you have to load some things separate to  VScode such as python (I am on 3.9.1 I think) then ensure these are in the %PATH% my path is:

C:\Python39\Scripts\;C:\Python39\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS\;C:\Program Files\Intel\Intel(R) Management Engine Components\iCLS\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files\xLights;C:\Program Files\Git\cmd;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin;C:\Users\"myName"AppData\Local\Microsoft\WindowsApps;C:\Users\"myName"\AppData\Local\Programs\Microsoft VS Code\bin;C:\MinGW\bin;C:\MIDI\MIDI_Library-5.0.2\src

Then you need to update VScode to take this in, Ctl-Shift-P and select platformIO: rebuild intellisense index.

Just to check - you have installed Platform IO on top of VScode.

I have the following extensions:
Autobuild Marlin
C/C++
C/C++ Compile Run
Code runner
Github pull request
Jupyter
MSYS2/Cygwin/MinGW support
Project Manager
PlatformIO
Pylance
Python for VScode
Python

Obviously many of these are not needed in this case, but it shows it's working with these ones

GitLang, 2021-05-18:
Many thanks for taking the time for a comprehensive answer. I have been using Arduino on top of PlatformIO on top of VSCode for some time with no problems. I have the latest Python installed with it on the path as usual. I have added a couple of (non-critical I believe) extensions on your list that I did not use. I have compiler flag -DNDEBUG defined.

Python must be OK as it is getting as far building a full .elf. It is the elf to hex part that is failing. If I edit fix_linkflags.py to comment out the elf to hex conversion thus :+1: 

Import("env")
env.Append(LINKFLAGS=["-nostdlib"])

# Custom HEX from ELF
#env.AddPostAction(
#    "$BUILD_DIR/${PROGNAME}.elf",
#    env.VerboseAction(" ".join([
#        "$OBJCOPY", "-O", "ihex", "-R", ".eeprom",
#        "$BUILD_DIR/${PROGNAME}.elf", "$BUILD_DIR/${PROGNAME}.hex"
#    ]), "Building $BUILD_DIR/${PROGNAME}.hex")
#)

Then the whole process works OK compiling with the "tick" and ending with an .elf and 
=== [SUCCESS] Took 2.96 seconds ===

If I remove the commenting, clean, and build again then it fails on

Building E:\DEV\VS Code\projects\T-962\T-962-improvements\.pio\build\lpc2134_01/firmware.hex
Usage: arm-none-eabi-objcopy [option(s)] in-file [out-file]
...
...
usage info for arm-non-eabi-obcopy
...
...
arm-none-eabi-objcopy: supported targets: elf32-littlearm elf32-littlearm-fdpic elf32-bigarm elf32-bigarm-fdpic elf32-little elf32-big plugin srec symbolsrec verilog tekhex binary ihex
*** [.pio\build\lpc2134_01\firmware.elf] Error 1
=== [FAILED] Took 2.47 seconds ===

I think all the points you covered are OK here, somehow it is failing to do the elf->hex copy command. I need to see more detail than "Error 1". It says to add the -v or --verbose flag but I have no idea where or to what to add it. Tried it as a compiler flag and it complained. Actually, I think I just answered my own question - it probably belongs with the copy command. I'll try it.
Thanks again.

maxgerhardt, 2021-05-18:
>Building E:\DEV\VS Code\projects\T-962\T-962-improvements.pio\build\lpc2134_01/firmware.hex

Ah my bad, your space in the path is probably breaking the passing of the filename arguments. Let me fix that quickly.

maxgerhardt, 2021-05-18:
Can you `git pull` my repo (https://github.com/maxgerhardt/T-962-improvements.git) again and retry?

GitLang, 2021-05-18:
It built first time, no warnings and produced a .elf, .bin and .hex. Thank you very much. I had no idea I has a space in a directory name. Must have been one of Microsoft's progs that doesn't allow you to change it. I am almost paranoid about spaces in directory or filenames!

Again, thank you. Very nice contribution

maxgerhardt, 2021-05-18:
>I had no idea I had a space in a directory name. 

It shouldn't have been a problem if I did my homework and properly escaped paths :D. Glad that it's working now for you.

I've added some additional sanity checks (e.g., building the project from a zip file and not from a `git clone` doesn't result in a fatal error anymore) and made changes to that Makefile system still works. These changes are in the opened pull request https://github.com/UnifiedEngineering/T-962-improvements/pull/207.