UnifiedEngineering/T-962-improvements #208: Screwed Up Cold Junction Compensation

RocketCat20, 2021-06-18:
Hi all. I connected the cold junction compensation stuff the way I was supposed to, but I think the sensor was the wrong version so I disconnected it. Unfortunately while disconnecting it, I accidentally completely ripped off the middle pad that one side of the resistor was soldered to. Is there some way for me to salvage this or am I doomed to remain without CJC?

RocketCat20, 2021-06-18:
I can't really see where else that trace goes and I've been unsuccessful in soldering to that tiny trace. I see that it goes into the corner of the controller, but I'm wondering if it connects anywhere else I can use.

xnk, 2021-06-20:
The middle pad is the signal used for 1-wire communication, only used for cj compensation or improved tc interfaces. It goes straight to the MCU and can be accessed there. Did the firmware detect the temperature sensor at all?

RocketCat20, 2021-06-21:
It said Cold Junction Not Present. I think I managed to scrape off some of the trace and find the pin on the controller it goes to, but that seems really hard to solder to. I will certainly try if I must, but does that pin go anywhere else or is there maybe some way I can mess with the firmware and use another pin for it?

mikeanton, 2021-06-21:
You should be able to solder to it fairly easily if you use solder paste.  Put some paste onto a section of the trace that has the solder mask cleaned off, and set the wire into the paste.  Heat with a soldering iron.

RocketCat20, 2021-06-21:
Do you know if a hot air gun might work even better possibly? I do have one. Also, do you know if it will matter if I use a standard resistor instead of an SMD one?

mikeanton, 2021-06-21:
It would probably be more difficult with a hot air gun as it would take too long to heat up.  Just a quick touch with a soldering iron should do it, providing the trace is free of solder mask.  I suspect the reason you are having difficulty attaching the wire to the trace is due to a lack of flux.  Using solder paste, or additional liquid flux will help in that regard.  I've attached 30ga wires to very fine traces, so I know it is possible to do.

RocketCat20, 2021-06-21:
Thanks! I bought some paste and flux along with the oven so I'll give that
a shot. That's probably safer than trying to solder directly to the top of
the pin on the chip right?

mikeanton, 2021-06-21:
Probably a bit safer.  If it eventually fails that way, you can try the pin.  Just make sure you don't end up with solder bridges between the pins.  If you can get the trace tinned with solder, you shouldn't have much difficulty in attaching a wire to it.

RocketCat20, 2021-06-21:
So tin it with a tinned iron and then solder paste on top of that, then lay
the (tinned) wire on and hit it with the iron?

mikeanton, 2021-06-21:
Tinning with a tinned iron doesn't work well, unless you apply flux first.  You could just put solder paste on the trace to tin it, or solder paste and set the wire into the paste, and just solder it.  There are a number of ways to get the job done, but they all require flux, and solder.  Flux is your friend when trying to do any fine soldering.  Just the solder on the tip can be used to supply solder to the joint, but without flux it never works well, if at all.

GitLang, 2021-06-21:
Just as a reminder - there is a full pdf schematic here somewhere. It is in the master archive if you download it as a zip. Should answer your "what goes where" questions. To reiterate what Mike said, flux really is your friend it will cut through nearly all muck, kill the solder surface tension and allow it to flow. The one thing it will not touch is the solder resist mask. It's designed to resist solder as it's name suggests :). You need a scalpel with a curved blade, like a number 15 blade on a number 3 handle (Swann Morton) You can then scrape very precisely, and the rounded blade means you don't scrape neighbouring tracks so easily. Your other friend for these sort of repair job is the old wire-wrapping wire. It is 30 gauge silver plated solder core with very tough insulation made of Kynar or PTFE. Watch out for scammers describing PVC insulation as wire wrap wire. You can make precise track replacements with it and tack the wire down with cyanoacrylate when done. Good luck.

xnk, 2021-06-22:
100% agree with flux and wire-wrapping wire being great repair/patching tools. Magnet wire also work well in a pinch.

Nothing else except one-wire stuff is connected to the pin in question, and the original firmware does not make use of it. There aren’t really any other spare io pins if system PWM fan is being used (as can be seen in the schematics PDF in the Wiki).

RocketCat20, 2021-06-27:
Hey, Thanks for the help everyone!! The solder paste idea worked great and CJC works as intended now.

closed 2021-06-27 by RocketCat20