UnifiedEngineering/T-962-improvements #209: New T-962A+ Model - Anyone Tried This Unit Yet?

jwestmoreland, 2021-06-22:
Hello All,

Just wondering - anyone tried the T-962A+ yet?  

Thoughts?

Mods?

TIA,
johnw

xnk, 2021-06-22:
Was that the one that is STM32-based? I’m really curious to see how the boards look like (whether it’s a complete redesign or basically replacing the LPC with the STM32).

jwestmoreland, 2021-06-22:
@xnk Werner,

I think so - haven't seen a lot - I think it's 220V A/C only as well since it 2.3kW.  

20A 120V isn't that uncommon so I'm not sure why 120V units are not available.

Also - it appears you have to use a PC/Laptop via USB to drive the unit - I'm not sure but there's a standalone mode
too where no PC is required.

I've seen some comments where people have remarked it's difficult to set custom profiles - I'm not sure since I haven't 
used one of these units yet.

Regards,
johnw

xnk, 2021-06-23:
Seems like a 110V version is mentioned in the manual as estechnical seems to be able to order them somewhere but I also only found the 220V variant

xnk, 2021-07-08:
T-962A+ works in stand-alone mode (or Alone mode as it calls it) or tethered to a PC (Windows). Haven’t tried the PC software, need to replace masking tape with kapton tape before attempting to heat anything. Got sidetracked getting some controller firmware running as mentioned here: https://twitter.com/meckodon/status/1413260680787075072

GPIO pins are mapped out, I2C + Keypad + LCD works, unfortunately they removed F4 for some unexplained reason(!)

xnk, 2021-07-10:
I've gotten quite far on my quest to tame this control board. I haven't bothered with the ADC yet and 1-wire isn't implemented, I'm using the I2C-to-SPI adapter with SPI TC interfaces, as I had a spare one. The over will heat at around 1.0C/s even with quite a bit of airflow through the unit (min fan speed at 30)

This is my usual go-to reflow profile for lead-free from a cold oven. This result is way better than I can get out of the old T-962(A):
[image]

And here's the ramp test:
[image]

bgrigoriu, 2021-07-10:
Woauh ! 
The cooling curve is VERY interesting. It is really good
Do you have data on the uniformity of heating/cooling between the different TC ?
I am puzzled by the temp ramp of only 1°C/sec
On an old T962 well isolated i get 1°C between 35 to 45 then around 2°C from 60 140°C then over1.5 up to 180 °C and over 1°C up to 215°C and 0.9 °C up to 245°C

xnk, 2021-07-10:
I’ll check the uniformity as well, they’re front and back on this oven. The power is supposed to be 2300W, and the area is quite a bit bigger than the T962A so power/area is lower. I haven’t done anything to isolate the oven at this point, and the exhaust fan was running way faster than I have been able to run on the other units. It doesn’t really _have_ to heat faster that around 1C/ second, right?

bgrigoriu, 2021-07-10:
Hi,
I am not a specialist in reflow but just a hobbist( even a newbie) . 
But if 1 to 1.5 °C per sec is enough that means that the T962 modded with four 400W lamps (as DimaSM338 presented) could withstand some (or even heavy) air recirculation .  The volume of the T962 (180 x 240xheght ) seems to be between four to eight times lower than the T962A+ 370x450x ??) . Thus the 1600W    is much more than  2300W of the plus version.  The  area is theoretically more important when using IR only but volume come up when air circulation  increase.
I also build on the report of taliesin which introduced the air straightener which reduce greatly variability during cooling ( and expectedly during heating). 
I ordered all parts and I will try to include all these and report. 
Potentially this could be applied widely IF it works.
So globally an unified firmware and a similar modding for all oven type will be very interesting for covering all budgets and needs.
I will try to build a web interface over the ESP8266 that I mounted inside my oven. 
My codding skills are paltry but it is fun trying

jwestmoreland, 2021-07-13:
So, @xnk , this is the 220 V unit not the 120 V unit?

Looks promising.

Thanks,
johnw

perigoso, 2021-07-13:
what is the difference between the old the new plus variant? is the controller board the only difference? is the shell the same?

PricelessToolkit, 2022-03-19:
Hi, @xnk  Is there any news about firmware, how's it going?