UnifiedEngineering/T-962-improvements #210: Improving  heat distribution

bgrigoriu, 2021-06-23:
Hi everyone,
Just received my T962 and doing some initial testing. 
After reading a lot of post the thing that was still apparent for me is that the heat distribution is uneven. Using a fan improves it but power is insufficient to heat-up enough fast and high.  
Did anyone test  adding one/two lamps (apart from the issues regarding power dissipation over the triac) ? Does it improves anything ? 
I am not a fan of radical changes but mechanically it is do-able. 
I would like to spare time if anyone has already done it (Just ADDING  the same type of lamp).

Second question: it seems evident that there is left to right gradient of temp . Did somebody has an sound explanation for this ?

GitLang, 2021-06-23:
I'd need to look at the schematic, but on my earlier 220V model, the lamps are switched by a solid state relay which is well overrated.

bgrigoriu, 2021-06-23:
Hi,
The actuals lamps seems to have between 300 and 400W. So we would need one or two more that means 1200W to max 1800 W.
The actual triac is a 16A BT139- 600E. Did not find easily a replacement for it in the same size and the same (10mA) gate sensitivity. Bigger (20/25 or 40A) may work but I would need to test and thus to have another T962 electronic board which I lack.
However with a change of the heat sink can be done easily and improve max current.
I will test and post. 

In the meantime I have disassembled mine and I think I have an explanation for the huge right to left difference in temperature that some of the users report.

The lateral part of the oven are very poorly isolated with chucks of  Si/Mg plaques that leave a lot of free space and some metal exposed. 
I have replaced them with a continuous plaque and the difference between the right and the left part average now only 3.3°C during the heating phase and 6.3 °C during the cooling phase when the fan is on (which is expected). 
I will retest with the new firmware but this is a thermodynamic issue and no depreciation/improvement can be expected. 
If anyone is interested I have the photos and a video with instructions but I do not ( yet?) know how to post them on GitHub (newbie).

GitLang, 2021-06-24:
So they have done away with the 25A solid state relay, and replaced  it with a lower rating triac?. How many lamps of what power are fitted as standard?. Is this the small A version or the larger B version (or don't they have separate versions now?). 3C to 6C difference is much better than a lot of us get. What is a silicon manganese plaque? (sorry for so many questions)

bgrigoriu, 2021-06-24:
On my machine T962 (small version  produced 04/ 2021 if I read correctly)  there is a 16A triac on the PCB.  
If I read ok the images on various web pages only the larger T962A machine has a SSR fitted on the back. I do not think that there is enough space in the back to place a SSR on the smaller T962 machine.
 
I think it is possible to use the actual triac (16A) at 8A (i.e. 1600W). As for now the  temp of the  triac seems to not be very high but I still need to make precise measures.. 
At 8A it will dissipate around 12W ( compared to around 6W now).   A lager heat sink can compensate this. I will try to see how to fit one.
For the difference of temp yes I got a left to right difference 3 °C  during heating phase and a bit over 6 when cooling ( which is unimportant in my opinion). The silica/Mg plaque is a "stupid" thermal shield your plumber is using  for gas welding. I am buying it from my local DIY store (https://www.bricodepot.fr/leers/ecran-thermique-souple/prod5967/# - do not take as a advertisement) but you can find similar things on ebay 
(https://www.amazon.fr/WELDTEAM-Ecran-pareflamme/dp/B00BBU60Z2/ref=sr_1_9?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=1MXURE4SRLAN1&dchild=1&keywords=ecran+thermique+soudure&qid=1624536152&sprefix=ecran+thermique+%2Caps%2C150&sr=8-9 for ex).   

Thank you for asking, I am also asking a lot of questions as I am an (old) newbie and starting to reflow . 

By the way I think that a clear procedure on how to calibrate the TC  ( using the provided OA) would be useful.  Some direction are given but they are not quite clear and refer to the use of mathlab (which I do not have  being alone in my kitchen).

Currently I have a difference of about 8 °C between the two TC (tested at lower temps which is probably not relevant) BUT questions remaining are:
1. Do you need (is it of any utility) to calibrate first using the pot's on the PCB and then by software or all can be done by the software
2. What is the best way to obtain the two temp steps for calibration (a separate reflow program included in the firmware would be great).

xnk, 2021-06-24:
I have used the bake mode to set the two temperatures needed to calculate the gain and offset adjustments. I’m trying to remember whether the pots affected the gain or offset, but you shouldn’t have to touch them and instead compensate in firmware (as long as you won’t hit the ADC limit).

The offset adjustment simply adjust the readout up or down, the gain adjustment will adjust the gain +/-10% so by calculating the “real” temperature difference and the setpoint difference between the two temperatures you can calculate what the compensation needs to be.

bgrigoriu, 2021-06-24:
Thanks ! 
I was going to do this today and then test
I will try find some time next week to solder a 31850/31855 PCB and then compare.

bgrigoriu, 2021-06-24:
> I have used the bake mode to set the two temperatures needed to calculate the gain and offset adjustments. I’m trying to remember whether the pots affected the gain or offset, but you shouldn’t have [...]

Can you tell how to isolate left and right TC channels ? if one is absent how is temp is calculated

bgrigoriu, 2021-06-24:
Sorry, I hit enter to fast. 
How is temp calculated with only one channel ?

xnk, 2021-06-24:
You can’t really isolate the two channels, calibration will either have to be run twice or using two external thermocouples (each in close proximity with the corresponding oven tc). The regulation will use the average of the two channels, but the two channels can be individually monitored in bake mode.

bgrigoriu, 2021-06-24:
Thank you to all of you for the help. 
I just finished to calibrate the TC. Not a very fast task. 
I confirm that during the heating phase, with the change of the lateral insulation. the maximum difference between the right and the left side is no more than 3 degrees C in average. I have only <10 seconds a 4°C difference (highest one). 
I looked with a thermal camera and the part that a losing most heat are in order : the front part of the drawer and the bottom lateral part both exceeding 40°C during heating phase. 
However, during the cooling phase, the left side is cooling hyper fast and the differences are HUGE; I even had more than 30 °C during almost one minute. I think that air circulation is a mess and I do not have any modality to know where the issue is. I am not sure that makes a great difference anyway (and I do not know who to ask).  
Thank you again for the great work done by all the contributors to the project. You deserve a medal. 
I have just another  short question. is there a documentation on what the firmware transmits over the serial and how to use it ?

bgrigoriu, 2021-06-25:
Hi again,
I also looked at the taliesin post about introducing an air straightener with the claim that it reduced strongly the temp differences during cooling phase. Did anyone also tested it ? 
And if yes do you know if slimmer 120mm fans exists?

xnk, 2021-06-29:
The commands the firmware understands are handled here: https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/main.c#L187-L286 <https://github.com/UnifiedEngineering/T-962-improvements/blob/master/src/main.c#L187-L286>

DimaSM338, 2021-07-02:
> Did anyone test adding one/two lamps (apart from the issues regarding power dissipation over the triac) ? Does it improves anything ?

I installed two additional lamps of the same since my board is 200x140 mm. The result is positive. I was able to complete the task.

Without additional lamps, the edges of the board did not warm up. With additional lamps everything is OK

bgrigoriu, 2021-07-02:
> I installed two additional lamps of the same since my board is 200x140 mm. The result is positive. I was able to complete the task.
> 
> Without additional lamps, the edges of the board did not warm up. With additional lamps everything is OK

Hi,
Did you do any changes ?  which lamp did you use ? 110V or 220V ? which power
The triac was "happy" ? did you measure a surface temp on the triac ?
On my machine with the original configuration the triac heats up to 50°C during a ramp test to 245°C.
Based on specs it should dissipate 8W in this situation. with two more lamps in wil go up to 16W.
Can you share some photos/ details ?

bgrigoriu, 2021-07-03:
Hi does anyone has the exact dimmension of the PCB ( and the position of the fixing screws ?) I would like to build a thermocouple board which will include an Wifi interface and a USB and I will need for designing it.
I tried to measure on the board but it it not enough precise

GitLang, 2021-07-04:
Buy some digital calipers, they are cheap now. Close calipers over a stud/fixing hole, and zero the caliper. Now measure over a pair of studs/holes and the reading is the centre distance between them, Do this for the two holes in 'X' and two holes in 'Y' You should get better than 0.1mm accuracy total. Leave 0.5mm clearance on your PCB holes and you should be fine. More accurate than measure the fixing points on the machine, would be to remove the board and measure the holes centers on the board using the same method. Measuring the board itself is trivial with calipers. Don't forget tor rest the zero on the calipers :).

bgrigoriu, 2021-07-04:
Hi
I did it but wanted an "independent" opinion.
Thank you

DimaSM338, 2021-07-04:
> Hi,
> Did you do any changes ? which lamp did you use ? 110V or 220V ? which power
> The triac was "happy" ? did you measure a surface temp on the triac ?

Hi,
In the form in which the oven is sold, it can hardly be called a soldering oven, therefore, yes, there were many changes. Below I will try to tell you more.
1. First, eliminated all uncontrolled heat loss. I bought a rubber oven door seal and stuck it around the edge of the PCB tray. In the photo, all 4 re lamps are just visible.

[image]

Glued with special high-temperature silicone.
[image]

The profile of the rubber seal is like this. The bottom of the card tray is also papered.
[image]

Also inside the oven, I taped two of the four lower air inlet grooves towards the fan.
[image]

In addition, I added a lot of mineral insulation to the case (when working with it, use a respirator and gloves). All slots were closed with Kapton high-temperature tape.
This improved the situation, but still prevented me from melting large boards - which I am working with around 185 x 130mm. The center was burnt, the edges of the board were not melted.

2. After that I decided to install two additional lamps. I developed a modernization plan and blueprint and bought lamps.
[image]

I bought lamps in China on Alliexpress ([links inside](https://aliexpress.ru/item/32950686557.html?spm=a2g0s.9042311.0.0.264d33edRVyY3O&_ga=2.184354662.1982739725.1625395306-1037648726.1577442302&_gac=1.229321198.1624224815.CjwKCAjwzruGBhBAEiwAUqMR8He-9FNdQLHY4xUxPzjTfV--HS2aw97nEtqsVw5t43FPEeton__nRhoCjlIQAvD_BwE&item_id=32950686557&sku_id=66224023779)) - exactly the same as they are in the oven.
[image]

[image]

They cost about $ 50. The lamps themselves are designed for 110 volts, but since I have a voltage of 220 volts, naturally they are connected in series with me. Series connection gives less current through the triac. If you have 110 volts in the network, then you will not be able to do this trick. The lamps will have to be switched on in parallel. With triac, you have to do something. I did not photograph the process of installing the lamps - I did not think that I would have to talk about it.

I can only give a drawing and say that I cut the holes for the new places of the lamps with such a tool with a diameter of 14mm.
[image]
    
In the lid where the sensors are located, I had to work like this below
[image]
 
The horizontal grooves were cut with a conventional grinding machine with a circle of 125 mm. The lower table did not have to be dismantled from the body. We managed to do everything on the assembled case. Angles at 45 degrees in the grooves were cut with ordinary metal scissors.
I sealed the old places where the lamps were installed with aluminum self-adhesive tape and Kapton tape.

Here you can see the places of the old lamps and how I organized the sensors.
[image]

Note - When installing the lamps, use only special silicone wires, since the temperature there is about 350 degrees.
With 4 re-lamps on board - the ramp test results turned out like this.
[image]

Cooling in my opinion is not enough - this is with half-taped slots to the fan. I'll try to open them later.

I will answer your question on triac.
triac was replaced by another - instead of BT139-600E.127 (16A) I installed BTA140-800 (26A). They have the same package and pinout. So I soldered it to the same place. I have not changed any other details in the triac harness. Additionally, I added the largest possible heatsink and used quality Arctic thermal grease. ... The radiator profile is like this:
[image]

Radiator size 37x30x15 mm. The hole for the triac attachment is drilled slightly offset.
[image]

And since I am not a supporter of noise, I immediately turned off the small fan that was in the oven. I made an additional window in the case on top - in the oven lid, with a grill for a thin 120x120 mm fan and planned to install it.
[image]
I haven’t set it yet, as everything works well without it.

All these activities allowed to leave the triac in the same place. During test switching on twice the number of lamps, the triac did not heat up above 60 degrees. (I could hold on to the radiator with my hand !!! (With the power off   )
I recently made a series of 10 boards, nonstop for 1 hour. BTA140-800 stayed alive. But I still made the two external lamps switchable, so as not to burn off electricity when they are not needed - when I solder a narrow board, I work on internal lamps.All these activities allowed to leave the triac in the same place. During test switching on with twice the number of lamps, the triac did not heat up above 60 degrees. (I could hold on to the heatsink with my hand !!! With the power off 
3. Also, all firmware updates have been made for the furnace - interfaces for K-type MAX31855 sensors have been added - 4 pieces and a USB connector. Added 2 additional sensors and arranged sensor locations according to the blueprint:
[image]

I do not use contact sensors. I use the same sensors that I used, but I lowered them as low as possible to the board. See drawing for details. Above there is a photo where you can see how I installed the sensors in sleeves with a diameter of 5 mm.
4. In the PCB tray, I made nuts and swivel sliding mounts and raised the PCB over the tray by 8 mm - and somewhere the same distance I lifted the lamps when installing new ones.
[image]

[... cut ...]
5. Now I'm trying to solve an issue with a program for Windows - Reflow Controller.exe in an adjacent branch - https://github.com/UnifiedEngineering/T-962-improvements/issues/201#issuecomment-872310235 - I will be grateful for any help in fixing code.

What else I am planning is to make an additional screen on the bottom of the oven - since 4 re-lamps will significantly heat the surface under the oven, especially when a small board is melted.

And yes .. I plan to replace the LEDs on the heating and cooling indicator, otherwise it's not very good with the white insulating tape :)

bgrigoriu, 2021-07-04:
Thanks  a lot for these detailled explanations !

GitLang, 2021-07-05:
The rubber seal round the PCB tray will greatly reduce the cooling airflow if it is anything like my next larger model, as the air is exhausted via that route. You can see on your temperature profile that the final temperature drops slowly and exponentially rather than the fast linear slope I get without a seal. Try removing or removing sections of the seal. That final cool down is important on the time-temperature product of a lot of components.

permanent standoffs is a good idea as long as you can position your board to not get soldered to them :). I use MF 6mm M3 brass standoffs fitted one in each board mounting hole with a nut. If you use a dome nut you may have clearance enough to turn the board over to do the other side.

bgrigoriu, 2021-07-05:
Hi
Thanks for the comments
These mechanical improvements are really very nicely enginered
I am planning to implement them Asap

However in the hacking spirit i found that the heating tubes are too expenssive
I found some (same lenght) for 5€/piece shipping included
I will test the quality and report
They are 400w 220v but with a minimal change it should work

zian31, 2021-07-14:
Wow DimaSM338, I'm impressed with your upgrades !
I would prefer buy a new T962A with 4 IR lamps instead of my T962 than add 2 IR lamps.
I my case, I don't use MAX31855 sensors, but MAX31850KATB+ (I've added the PNOXI board).
I disconnect the 2 original TC of the T962 and put the 4 TC in the 4 angles on the PCB board to solder (with Kapton), but the firmware is only using 2 TC values for it's algo : I will change it in the C code to use the 4 TC values next.
Now I'm always working on a new windows software with curves dialoging with UART Port Com, it's almost working.

I'm wondering why you don't prefer to add a SSR-25A-DA (as existing in T962A) for the heating part instead of this new TRIAC+dissipator, it seems more simple to do : just a little place to find in the T962. And also possible to add another one for the cooling fan.

I think it's very important to put the TC sensors on the PCB because it's the source of the PID control.

Anyway, my problem is not the same as yours with your 200x140mm PCB : in my case with this kind of boards and the custom#2 profile, the time of cooling is too long, the heating time is more acceptable. And I have the T962Z new model (hard to say what are the exact differences with the original T962).

bgrigoriu, 2021-07-14:
> Now I'm always working on a new windows software with curves dialoging with UART Port Com, it's almost working.
> 

 And I have the T962Z new model (hard to say what are the exact differences with the original T962).

Hi, 
I did not know that a Z version exists. I looked on the net and the images as identical with the “classical” T962 with the 2020 upgrade on the exhausting tube connection ( which on my occasional use I am not sure is very useful).

Do you have some Photos of the Interior design and the PCB ? 

It is wonderful that you could develop a windows software for communicating /controlling the Owen. Can you give mode details? 

I am very interested by this. I put on the ISP connector an ESP-01 with ESP-link inside and it work well in terms of connection but it is not very practical ( apart getting rid of a wire) because the oven firmware does dot expects any other character after each command and is not easy to use a terminal connected with it. Your software could be a solution.

I wonder also if your code could be easily re-writed as a web page as the ESPlink can host a web page and thus the oven could be controlled in a platform-independent manner by a web browser.

zian31, 2021-07-14:
I ever used two ESP-01 to speak together with UART on each side, but I didn't know this ESP-link with Wifi, it's interesting.
Unfortunatly, my windows soft will not be compatible with a web browser, but it will work also on Linux.
I use the PNOXI board with a FTDI chip (or equivalent) to create a virtual COM port with a USB cable.

The main interest of a software is the see curves of differents TC in live (without it's possible to do it on Excel after), and I want to be able to upgrade the oven EEPROM from this soft (with also a firmware modification needed) to change the custom#1 profile as needed.

Concerning the T962Z version, it's the name of this 2020 upgrade with the exhausting tube, here are the differences I've noticed with the T962 :
- Power Earth connexion already done with the rack
- Circular output fan on behind
- With the T962 air comes from behind and get out with the bottom ? With the T962Z, the air seems to come from the center of the circular fan and get out by the borders of the circular fan output, and there is almost no circulation with the bottom of the rack : it's strange

bgrigoriu, 2021-07-14:
Thanks for the response.
I tested the ESP 01 and virtual TCP  to serial port (either comtocom or HW VSP3). They work very well even with the existing reflow controler software or everything else. 
If you finish you software and are eager to share the code, one can try to copy the algorithm and implement it on the webserver of the ESP link. I think it will be widely useful. 
I also designed a PCB wich include en ESP plus 4 TC controllers ( versions with 31855 and 31850 available) but I was not yet able to test them as I am waiting for the PCB from China.

George-2005, 2021-08-05:
> In the meantime I have disassembled mine and I think I have an explanation for the huge right to left difference in temperature that some of the users report.
>
> The lateral part of the oven are very [...]

> If anyone is interested I have the photos and a video with instructions but I do not ( yet?) know how to post them on GitHub (newbie).

It sounds great, actually! I've just bought T-962 so I'm preparing now for oven improvements. Please let me ask: Have you replaced original Si/Mg insulation completely (even at the top where thermocouples are situated)? Or have you replaced it at the sides (left & right only)? And is the just one piece of "Écran thermique souple - CASTOLIN" sufficient or it has to be used more pieces?

I'm very interested in watching your photos and video of replacing the insulation. Could it be possible to provide them? Thank you

bgrigoriu, 2021-08-05:
Please let me ask: Have you replaced original Si/Mg insulation completely (even at the top where thermocouples are situated)? Or have you replaced it at the sides (left & right only)? And is the just one piece of "Écran thermique souple - CASTOLIN" sufficient or it has to be used more pieces?

Hi,
I kept the  original insulation " as is". I just extracted the lateral parts then inserted the new continous  insulation (the one I had in my garrage was 1 cm thick  and had to put two layers of it) and the reinserted the old fragmented insulation that  was on the owen. 
I wil try to retreive he photos and put  then on the next post. The Top insulation is uniform, continuos  and should not be touched I think.  The heat loss  on the sides are probably due to the uneven insulation.

George-2005, 2021-08-05:
> Hi,
> I kept the original insulation " as is". I just extracted the lateral parts then inserted the new continous insulation (the one I had in my garrage was 1 cm thick and had to put two layers of [...]

Hi, thank you very much for your quick reply and all valuable information! If you could be able to post the photos (or even e.g. link to video etc.) I would appreciate it very much, actually! Thanks again for your help.

George-2005, 2021-08-07:
Just another issue regarding the oven insulation - some people say that the insulation material inside the oven (including inside the drawer's door) could be carcinogenic asbestos (e.g. please see #215 below). Is that really true? If yes, what to do with it to avoid the cancer risk?

bgrigoriu, 2021-08-07:
> Just another issue regarding the oven insulation - some people say that the insulation material inside the oven (including inside the drawer's door) could be carcinogenic asbestos (e.g. please see #215 below). Is that really true? If yes, what to do with it to avoid the cancer risk?

I am not sure that is so oxic 
1. the material is not asbestos as it looks like. Asbestos has been globally forbidden in all possible uses in almost all industrialized countries China included. The last countries have done so in the late 2010 (2018 for Canada). 
2. The fiber inside seems to be a silicate of Ca/or Mg. 
3. Of course everything can be harmful depending how you use it. I would not recommend to eat it (even if for ex Calcium silicate is allowed as a food additive) nor to purposedly inhale it.  Similar plaques are used for years by plummers as soldering shields. So keep it inside and well fixed with aluminum foil. 
Of course China is not as controlled as Western Europe or US but still. 

HOWEVER such materials are rather irritant for the skin/eyes and the use of gloves and not using contact lenses ( which retains fibers locally) can be recommended. Also it can be advised to used protective clothing and change them after manipulation

George-2005, 2021-08-07:
Thank you very much for all valuable information and advices! I'm just planning to fix the insulation by adding the "Écran thermique souple - CASTOLIN" as you've mentioned before. So I'm very sorry to bother you again but if you post some photos of such improving I would be so grateful. Thank you!

GitLang, 2021-08-08:
Echoing bgrigoriu really, but melting point of asbestos is around 850 degC / 1600 degF so if your torch wouldn't touch it then it's unlikely to be asbestos. Secondly, it's very unlikely that asbestos has been used, in fact extremely unlikely. It's use is illegal worldwide and has been for some time. If you don't like what is used, then any good refractory insulating fibre will be just fine. Remember, we are only going up to, what?, 350 degC max and most of these insulating materials are rated around 800 degC or 1200 to 1400 degC for the silica ones. If you are still worried search "How to identify asbestos" on google. If you have any remaining doubts then don't touch it, call a disposal firm in. Disturbing it will put the fibres in the air and that is the most dangerous time with asbestos.

bgrigoriu, 2021-08-09:
Hi ,
With much delay I am putting some images of the original lateral isolation and how to fix it.  Sorry for the nail speed but we all have top work from time to time. 

A: Initial insulation: 
[image]

[image]

B: Available space
[image]

NOTE: Since the cables have to be There,  two layers of isolation material of ~2cm are needed.  For the first layer a continous plaque is mandatory ( as this one): 

[image]

For the second layer you can reuse material  that was inside de owen plus cuts from the plaque you bougth. (just thinking at the planet not stingy) 

The first layer is eay to put in place. but for the second use this: 
[image]

Othewise you will not be able to put them in place tightly and ensuring a good isoaltion.

George-2005, 2021-08-09:
Thank you very much for your photos !!! Now that's clear how the original insulation is just ineffective.
 

BUT I'm still thinking if doing it or even return the oven back. As I've tried to find more and more information it makes me fear more and more :-( Someone has said that he let the insulation material test in laboratory and it's _**Alumina-Silica Fiber Wool (ASW), which, in fact, it's the almost same carcinogen as asbestos, although ASW hasn't been banned so far**_. And the problem could be that main fan could blow the insulation's micro-fibers from the oven into air ...

And someone else has said the oven **_"is insulated using asbestos cardboard". Does he mean the red cardboard below the alumunium foil (see picture) ?_**

[image]

 
I'm just confused and disappointed. I was prepared to mod the controller, thermocouples, firmware etc, not to fear of cancer :-((( On the other hand I don't know ho to reflow SMD in other way (I don't want to try to convert regular toaster oven to reflow oven, actually ...)

zian31, 2021-08-09:
> I'm just confused and disappointed. I was prepared to mod the controller, thermocouples, firmware etc, not to fear of cancer :-((( On the other hand I don't know ho to reflow SMD in other way (I don't want to try to convert regular toaster oven to reflow oven, actually ...)

I you have these doubts, you can also change all the isolation and assume that yours will be better than the original : 2 in 1 action.
Or you can use the oven outside (in exterior) with an exhausting tube on the T962Z : I'm doing that personally to be protected from solder gases.
I've ever seen on youtube people connecting exhausting tube on the T962 on the bottom with creating an hole in some piece of wood under, and another one with a kind of translucid plastic chamber connected with exterior with a tube to protect the air room.

George-2005, 2021-08-10:
Thank you very much to all contributors for valuable information and opinions! So I must think very carefully if I will return the oven or not. But I'm not happy with it anyway :-(

zian31, 2021-08-10:
> Thank you very much to all contributors for valuable information and opinions! So I must think very carefully if I will return the oven or not. But I'm not happy with it anyway :-(

Nobody is really happy with this T-962, but there is nothing better on the market except with more than 1500$.
Concerning your cancer risks with isolation, I think there is more cancer risks with solder gases (even with no Pb solder paste), that's why this T-962Z has been created in 2020 with the possibility to connect a tube for gases extraction.

But you can solve all theses problem with using the T-962 outdoor as I do.
So the main issue of this T-962 will be as the title of this topic : "quite bad heat distribution" : there is lot's of try to solve this on youtube, but none easy to do with good results. In my case, I prefer just to measure the Temp on 4 points of the PCB and compare heating after heating.

zian31, 2021-08-13:
> ... and another one with a kind of translucid plastic chamber connected with exterior with a tube to protect the air room.
I found it ! Here : https://www.youtube.com/watch?v=-DMYJmB4naA&t=1239s
There is a car filter for the particules ! wow

affetus, 2022-02-06:
First of all, a big thank-you to xnk for creating the custom firmware for the Puhui and for making it open source.
Thanks also to everyone who are reporting their enhancements to the original design -- be it success or a failure.

I recently purchased a T-962 oven being well aware of its issues and received a recently manufactured one (05/2021). Paid around 97 euro including VAT and shipping from Czech Republic so I think it was a pretty good deal. I was actually positively surprised of the quality of the mechanical construction and the paintwork. It is just that some corners have been cut in the firmware (again resolved by xnk) and in the temperature management (which I'm hoping to resolve for the most part).

My oven has the exhaust duct mount on the rear and the main fan is pulling air through the bottom slits and then out from the duct. There are two self-closing baffles just before the fan and I actually like this design better because:

1) it will not heat up the working surface much
2) the baffles will help keeping the reflow compartment hotter
3) the exhaust fumes will come out from the rear -- easily picked up by a fume extractor arm. I will not attach any duct hose to the oven, as any forced suction would affect the baffle positions during the reflow process.

The main fan in my oven is Papst, which is surprising as I will not be able to buy this fan for less than 40 euros. My other passion is vintage lighting and this fan type is frequently used in some gas discharge lamp based lighting fixtures and I know they last very long even in hotter temperatures. There is a concern that the fan will heat up quite a bit as it will pull out hot air (up to 250C), but this will happen only for a short duration and I will plan to use the oven with low duty cycles, anyway. As the fan is made of aluminium, there is also some thermal mass which helps to reduce the overall temperature which the fan experiences during the cooldown phase.

The biggest issue I see is uneven heating, mostly due to lack of air circulation inside the chamber. 
There is also a problem with the main fan sealing, it is done with aluminium tape but it does not properly seal the air duct from the reflow chamber to the fan. You'll notice the issue if you remove the rear panel with the fan. Some air seems to be pulled from the sides of the chassis and this causes noise during cooldown when the temperature has dropped to about 100C, as one of the baffles is opening and closing due to variations in air pressure.

What I'm going to do is leave the fan configuration as it is, so the baffles will stay (even though the instructions from ES Technical tell you to remove the baffles and reconfigure the fan to suck air from rear). I will also set the idle fan speed to 0 as the baffles will prevent it from pulling air at low speeds anyway.

I will then add a small motor on top and attach a thin metal blade which is short enough to clear the two thermocouples and thin enough to clear the IR lamps when it rotates above the lamps. I am hoping that the rotating blade will mix the air so that the reflow temperature is more even on the PCB surface.

affetus, 2022-02-06:
One more comment regarding the insulation used by Puhui. There have been concerns about it being asbestos (both in the tape and the wool). However, I believe the wool is made of aluminum silicate (ceramic) and the tape is indeed regular masking (painters') tape which is surprising as they did use proper Kapton tape to insulate the IR lamp terminals. So no major worries there, although it is never good to inhale these dusts.

GitLang, 2022-02-06:
Are you absolutely sure it is blowing very hot aire out through the fan?. All the other oves draw cold air in through the fan and exhaust the hot air down through the ducts. That avoids having to use a very expensive specialised fan that can handle 200 degC air!.

zian31, 2022-02-06:
> Are you absolutely sure it is blowing very hot aire out through the fan?
Yes, now I can confirm it and  the fan is also one "ebmpapst" with my T-962 "Z".

affetus, 2022-02-06:
During cool-down the exhaust air gets cooler very rapidly so I don't think any metal fan will have problems. But in any case the air is definitely warmer than what it would be when sucked in thru the rear. 

What I tried to say is, my oven was already equipped with an all-metal Papst fan from factory which is quite surprising given the total price I paid.

xnk, 2022-02-08:
It was mentioned by ES Technical here https://www.estechnical.co.uk/images/pdfs/Addendum_-_T962A_A_Fan_and_air_path_modification.pdf <https://www.estechnical.co.uk/images/pdfs/Addendum_-_T962A_A_Fan_and_air_path_modification.pdf> that the direction of airflow is important, and I can only agree that I had _huge_ issues with the T-962A+ which also had the new “improved” airflow. It looked like the solder oxidised really bad despite reachine the correct temperature. Reversing the airflow using a similar modification as estechnical describes resolved that issue for me. Pulling in even small amounts of cool air directly below the  PCB seems to be messing things up. The original firmware really isn’t impacted as the fan isn’t running during reflow.

/wj

zian31, 2022-02-09:
> Pulling in even small amounts of cool air directly below the PCB seems to be messing things up. The original firmware really isn’t impacted as the fan isn’t running during reflow.
Thanks xnk for these important informations.
How do you reverse the airflow on a T-962+ or T-962A+ ? ES Technical pdf is showing how to replace self closing slats, but the fan has also to be 180° rotated (to inverse the flow direction) ? Or maybe you can choose flow direction with your firmware ?
In your firmware, the fan is used during reflow (I suppose it's to mix air inside) : what is the really diffrence with the original firmware in the results ?
I'm also wondering why not just blocking self closing slats to an angle openned of ~15° ? It seems more simple that changing it ?

xnk, 2022-02-09:
Correct, the fan has to be rotated 180 degrees (it’s also described in that pdf, at least I know I saw that mentioned somewhere), the direction cannot be changed in firmware (it’s an AC fan with nothing but on/off control). It’s basically just rotating the fan and replacing the slats with a louvre (I used ~half of a outdoor vent louvre).

The slats really didn’t open that much on mine but the direction of the airflow made a huge difference. As it’s mentioned by ES Technical I assume there’s a valid reason that they stress that this is a required modification for reflow to function properly, but I don’t know exactly why. It definitely has to do with the fact that the fan runs (slowly) during reflow, as that’s the main difference between the original firmware and the open-source alternative. That and timing.

zian31, 2022-02-09:
Sorry xnk, I've edited my question before your answer.
Your firmware has to other nice difference : cold junction correction & 1-wire reading (with MAX31850KATB+ in my case).

So I re-ask you my new question : "With your (excellent) firmware, the fan is always working during reflow (can you confim that it's only to mix air inside ?) : what is the really difference with the original firmware concerning the final soldering results (which as you said is not using the fan during reflow), is it really better ?
Maybe in your firmare it's interresting not to use the fan during the period which temp is over Liquidus solder paste ? (or for example already 30°C before this Liquidus temp) ? In order to avoid cold air to climb during this critical "Liquidus" period ?"

Or maybe another solution could be :
- Keep T962+ with air flow from back to rear
- Keep the slats
- Use of your firmware modified without rear fan working during reflow
- Add a slow Convection Fan inside (working during all reflow) close to lamp IR as Jerry Walker : https://www.youtube.com/watch?v=9026Odx_jOw&t=960s

affetus, 2022-02-09:
> Or maybe another solution could be :
>
>     * Keep T962+ with air flow from back to rear
>
>     * Keep the slats
>
>     * Use of your firmware modified without rear fan working during reflow
>
>     * Add a [...]

This is exactly what I am planning to do for my T-962, as soon as I get the parts I need. Plus, assemble new thermal insulation and seal the airflow so that it is not sucked from outside the chamber.

xnk, 2022-02-09:
I didn’t notice the edit as I was looking at the emails (and responded there). You can already turn the circulation off by adjusting the idle fan speed to 0 in the setup, the reason for having the fan running is simply one way to even out the heat distribution. If you have another fan inside there’s no need to reverse the airflow of the cooling fan.

zian31, 2022-02-10:
Thanks xnk ! (so I'll modify your firmware, hope you are ok with that ;) )

> This is exactly what I am planning to do for my T-962, as soon as I get the parts I need. Plus, assemble new thermal insulation and seal the airflow so that air is not sucked from outside the chamber during cooldown.

I don't understand what you mean with "seal the airflow so that air is not sucked from outside the chamber during cooldown" : the air will be sucked from where in your new design ?
One of my interrogation will be that during cooldown, air will come from down to up (and rear Fan is at 100% during cooldown), with old T-962 it was from up to down directly on PCB, here PCB could be a kind of "barrier" in air flow.
I forget to write another thing in my new design : 
- 2 Thermocouples used directly put on 2 angles of the PCB (orginal 2 Thermocouples close to IR lamp will be disconnected, because these one don't represent usable temperature)

affetus, 2022-02-10:
> I don't understand what you mean with "seal the airflow so that air is not sucked from outside the chamber during cooldown" : the air will be sucked from where in your new design ?

Maybe there are variations in the manufacturing process, but in my unit the rear fan was not completely sealed to the plate which holds the slats, so there was air leakage from the inside of the main chassis, i.e. the space which also contains the electronics.

xnk, 2022-02-10:
Yeah it seems to be a bit of hit-or-miss how well things are put together, some units have wide gaps, others not so much. The whole concept of placing the electronics on top of the oven is a bit sub-optimal, and their mitigation using that really annoying 40mm system fan to cool the electronics seems to be a bit of an afterthought (that somehow stuck).

xnk, 2022-02-10:
You go right ahead and modify :) Although you don’t even need to modify the firmware in order to turn the fan circulation off. Fan will still turn on if temperature overshoots the curve during reflow though so I’m going to experiment with only allowing the fan to increase above idle during the cooling phase.

zian31, 2022-02-10:
Now I understand what you mean : you want that during cooldown, air comes only from back of the PCB, and not from other place in the T-962 (moreover because slats are limiting the good air way) ? Another solution could be one or two servo-motor to open the slats just during cooldown, with kind of "wires" to pull slats ? I prefer to try that :)

Have you seen this slow convection Fan inside (working during all reflow) close to lamp IR (Jerry Walker) : https://www.youtube.com/watch?v=9026Odx_jOw&t=960s
Or maybe will you use another AC motor ? I think Jerry Walkeris true when saying slow rotation is important, if it's too fast it's not working well...

affetus, 2022-02-10:
Yes I have seen the video from his T-962A and I will try to apply that exact method for T-962. 

The heating power of the smaller model is a bit underrated (about 670W), so I am trying to prevent all cool air from entering during the reflow and the air mixer should help in that. The goal is to not pull air from under the PCB but just circulate the hot air around like in a household convection oven.

The slats will stay closed nicely unless the rear fan runs with a higher speed, so I don’t think any extra control is needed for them. This also means that using the rear fan for circulation does not work, as it requires a higher flow to open the slats (especially if there are air leaks which I mentioned earlier).

Problem is that the rear fan is so thick that it limits the slats from opening completely. It is possible to increase the slat opening angle a bit (about 1.5mm) by removing two fiber washers which are between the fan and the chassis rear plate, but it requires some machining of either the chassis or the fan frame because those washers were used to add a small gap to fit the fan better. I need to investigate this more, maybe there is a way to modify the slats a bit on a way that they open up more, still sealing the airway when in a closed position.

zian31, 2022-02-10:
Ok, I believed that slats were no fully openned during 100% PWM power with rear fan, so no need of extra servo-motors.
I've now read that slats are 400mils openned, it should be enough compared to ES technical special new piece (with other air way direction) : https://www.estechnical.co.uk/images/pdfs/Addendum_-_T962A_A_Fan_and_air_path_modification.pdf 

So as you, I will put some aluminium to seal the air flow.

And the main modification is this new inside little fan motor : do you already find one ? It should also be thermally isolated. I think I will make it work all the time (so not with a firmware command), but I hesitate to find one with DC 12V or AC 230V.
I also use a T-962+ I've added a "pnoxi" board.

zian31, 2022-02-10:
Perfect ! I'd be interested with your 2mm to 3mm shaft adapter reference and also your helix or blade reference :)
I think I will also use a 25mm metallic tube arround the shaft to protect from wool shaft catching.

affetus, 2022-02-10:
This is what I will use as the adapter, motor shaft is 2mm:

https://www.aliexpress.com/item/1005002911230673.html

Blade design will need some more thinking but I will use Jerry’s design (from the Youtube video) as the starting point, as he had good results with it in T-962A.

GitLang, 2022-02-10:
Werner, how are you going to control the temperature if you have no forced cooling element?. Without it, you have fast heat in from the lamps and very slow heat out from thermal insulation losses. I've tried to do this in the past, and it just cannot work unless you are on an extremely slow control bandwidth. For the job we did controlling a standards room with +/- 0.2F. the cooling apparatus was as much power as the heating.  If you put energy in you have got to take it out, and unless you want to get into very non-linear control and stability problems, keep the cooling!. In fact, some schemes will heat the target with more power than required then control temperature with heat extraction..

xnk, 2022-02-10:
What I meant was that the PID regulator wouldn’t be allowed to turn the fan on until the reflow profile has hit the peak temperature. Normally the fan shouldn’t turn on during the ramp-up but I’ve seen it happen on the T-962A+ where the PID regulator hasn’t been tuned properly. During the ramp-down the cooling fan has to run.

[8 more replies]