UnifiedEngineering/T-962-improvements #211: Wireless serial with esp8266

bgrigoriu, 2021-06-26:
Made this morning the final mod by connecting an ESP8266 with ESP link as per m-kozlowski  notes on the wiki page.
I choose to use an ESP-01 for which I changes the 2x4 connector with a 1x5 connector that fit perfectly on the ISP connector of the over. Thus you don’t need to fix it with Kapton, it is really simple
Works excellent for getting temps during reflow but is very difficult to “talk” over serial. Any clues why (more than the wiki page) and how to fix/workaround?

xnk, 2021-06-28:
One of the major issues communicating with the firmware over serial is that there can’t be any idle time between the bytes, they have to go into the serial port as an uninterrupted stream, I wonder if that’s what you’re experiencing?

bgrigoriu, 2021-06-28:
Exactly !
I wonder how to fix this.
Coding is  not by best skill now but i can try. I will try to read and see wxampkes in order to learn . I am open to ant suggestion .
I also tried the reflow controller software . It is nice but still laks some basic useful features as for example storing a number of reflow profiles and then downloading to the owen on a as neded based. I did not look at the firmware but it would be awsome id the refliw profiles would be defined at a fixed eeprom position so you could update without reflashing.I frequenly use this on atmega328 which i use for my home automation projects.
I tried to look inside the code but visual studio is new to me

xnk, 2021-06-29:
The EEPROM can hold two profiles which can be replaced without refreshing the main controller (those are the two custom ones you can tweak directly on the over UI itself in a fairly time consuming way). The reason for the lousy use of the EEPROM space was to stay compatible with the OEM firmware (in hindsight that compatibility should have never been kept).

bgrigoriu, 2021-06-29:
Thnak you,
We are all much more "intelligent" after  than before seeing the outcome of an action .