UnifiedEngineering/T-962-improvements #215: T-962 - Asbestos inside ???

George-2005, 2021-08-07:
I've just bought T-962 and now I'm worried very much about the rumours on the net (please see examples below) that the insulation material inside the oven (including inside the drawer's door) could be carcinogenic asbestos!

**Please does anyone exactly know what insulation material inside is made of?**

Thank you

---------------------
[https://www.youtube.com/watch?v=Et5BhSMlLqg](url) :

"Thermal insulation is asbestos cardboard. Use protection when opening."

"A friend of my discovered in his imported T962A SMD reflow oven asbestos and company produced it was even honest enough to confirm all heat parts are covered with asbestos but as they can confirm and try to tell they sold already many pcs around the world and nothing is wrong with this asbestos used nobody die and asbestos is ok only a little. So as we can see modern technology don't come with modern thinking, for some companies asbestos seems to be a neutral."

"The manual for this thing warns to take caution when opening because "insulations in the machine had been treated strictly." I have no idea what that means but the fact that the insulation is mentioned is slightly worrying.
Also there have been a couple of reports that I've found online of ebay sellers of this oven claiming that it does, in fact, contain asbestos. None of this is strictly confirmed I suppose, but it doesn't provide much confidence.
...
So I ordered a T962 and as a quick test put the insulation material under a mapp torch and it glows white hot and doesn't flinch (compared to a known sample of fiberglass which melts pretty quickly). Not necessarily proof that it's asbestos by any means but not at all comforting, again. Now to decide what to do with this thing..."

VictoriaBentell, 2022-06-04:
Thanks for making this post, since I wouldn't have guessed that opening it up would present a health hazard. Not sure which one is in the machine, but unfortunately [fiberglass is also carcinogenic when inhaled](https://haztrainer.com/fiberglass-and-asbestos-information/), likely thanks to the silica it contains.  I'm going to start wearing a respirator when I have this thing open.

jimd80, 2025-04-07:
Just for information: I've sent a sample of the white 'wool' below the aluminium sheets to a certified lab, and they confirmed it's fiberglass and NOT asbestos. This is a big relief because unbounded asbestos fibers (near the fan!) would be a serious health risk.

I cannot speak about the cardboard, but I'm less worried about that as these fibers are bound. As long as you don't damage the cardboard, it should be fine.