UnifiedEngineering/T-962-improvements #218: Display blank after update (2021 hardware)

norcal88, 2021-11-17:
My issue is similar to #115 and #196 but different.  After the firmware update the backlight never comes on (even after 30+ boots).  My build date is 21-7, so I'm wondering if anyone else has been successful with newer hardware?

One thing that gives me hope is that the OEM firmware posted by @JohnWestlake in 2017 works perfectly.  That would seem to indicate the flashing process is working and that the hardware is similar to (if not the same as) what was shipping in 2017.

Here's what I have tried:

- Manually resetting after programming
- Pull-up resistor on the reset line
- Multiple versions of the UnifiedEngineering firmware (from v0.1.1 to v0.5.2)
- Disconnecting the programming interface after flashing
- Serial connection via terminal.  This works and can execute the "Synchronize" but the last message I see is "Initializing" and it seems locked up at that point.  

Any ideas?  I guess I'll just load the 2017 OEM firmware for now, but I'd be really curious if someone has had success with this project recently or if there is some incompatible hardware change.  Thanks!

maciekcz11, 2021-11-30:
Hi! 
I have many problems with my T-962 oven. Could you share with me this original HEX file? I didn't do the backup of my IC....
I tried all UnifiedEngineering firmwares and also the original one from other side. But my display still doesn't work. I see some fragments but upper sections are messed up. I tried to resolder the IC but something went wrong.

norcal88, 2021-12-01:
Hi @maciekcz11, here is the link to the firmware that worked for me:  [https://github.com/UnifiedEngineering/T-962-improvements/issues/115#issuecomment-336282412](url)  Unfortunately, that's just the OEM firmware, so it doesn't support any of the enhancements and crashes frequently.  At least it works...

I bought the T-962 specifically because of the new firmware and updates here.  Since those don't seem work anymore, I'm stuck with a buggy, barely working oven.  Hopefully someone will figure out the compatibility issue with newly manufactured units.  Good luck!

GitLang, 2021-12-01:
Of course they still work, and much better than the originally supplied firmware!. The majority of people here are using the modified firmware with good success. If it doesn't work for you, then there is a fault, and you have to fault find logically, one thing at a time.
1. Are you checking the MDA or SHA of the file on Github against the file you downloaded.
2. Are you following the firmware upload routine to the letter, and powering off, remove programming lead and any jumpers to put it into programming mode, and power up again to test it.?
3. If the above two points are OK, then you may have a hardware problem. A modification done incorrectly?. A component knocked from a socket?. A blob of solder dropped and shorting out two points?. Examine thoroughly, preferably after defluxing the board, and with a loupe or magnifying board. The most essential element to finding a fault is start with a completely open mind as to the cause of it, and proceed one check at a time, returning things to the starting state before trying the next.

John-Westlake, 2021-12-01:
maciekcz11,

Are you located in the CZ? if so I can help with your unit if your local to Olomouc area...

I dont believe the software will effect the top line of the LCD as you describe, it sounds more like a hardware issue to me..

sevstels, 2021-12-08:
I made a bug fix in the display driver. Now it works absolutely correctly.

maciekcz11, 2021-12-15:
Hello!

Thank's to all people which replied to my problem. Yes problem look like the problem with the PCB itself. I inspected it on the microscope. Due to poor quality of this PCB during resoldering i messed up some traces. But after checking this with multimeter everything seems to be correct. I will try the firmware flash one more time and if this wouldn't work i will try to wire to LCD directly from IC. 
The fact that display show some info on top suggest that it will be software issue. If there would be the hardware problem i would not probably see anything or just randomly send bits.

@sevstels where I can find the software which you talk about?

sevstels, 2021-12-16:
Try the firmware from here: 
https://en.gradient-sg.com/t962/

I have not tested it on a real device.
It was just theoretically compiled and various flaws were fixed. 
The tested version is made for MCP9600 sensors, the others are disabled for the time being so as not to interfere with the finalization of the firmware.

norcal88, 2021-12-20:
Hi @sevstels thanks a lot for helping with this issue.  I only saw one firmware link on that page that points to "update.zip" and says (fixed version 0.5.2).  However, that file is not there and just returns a 404 error.  Was that the correct file?  Is there another download location?  Thanks again!

sevstels, 2021-12-21:
Oh... yes, link was wrong.
Already fixed it.