UnifiedEngineering/T-962-improvements #220: The thermal profile execution periodically hangs

sevstels, 2021-12-12:
While carefully checking the operation of the latest firmware version, I found a critical error. 

The RTC timer periodically stops randomly. This leads to incorrect reading of the thermoprofile from the memory. The furnace is not soldering correctly. According to the CPU documentation, the power pin of the RTC ->VBAT node should always be connected to a power source.  But physically, the VBAT pin (49) is "hanging in the air" on the device board. This is a mistake.

> The VBAT pin must be connected to a voltage source supplying at least 1.8V all the time.
> This means that in case no external battery is used the VBAT can be connected to VDD.

But the author did not make the appropriate improvements to the hardware, so the behavior of the program is unpredictable.

xnk, 2021-12-12:
Oh that’s an interesting problem I haven’t seen on the hardware I’ve looked at. On the board I had available here there is a direct connection between pin 49 (VBAT) and 3.3V. I thought the original firmware made use of the RTC as well, but I can’t remember 100%. Even though the schematics available on GitHub doesn’t show VREF and VREF connected to 3.3V they both most certainly are on this board.

Which board is this failing on, and how new is it? Does RTC work reliably if you bridge VBAT to 3.3V? I wonder if this is a bad solder joint on this particular board or if the layout was changed.

The firmware may certainly be unpredictable, but the RTC definitely has been rock solid on the boards I have personally tested.

sevstels, 2021-12-13:
I made own debug board according yours schematic. Because on the original pcb there is no way to connect the jtag. For my taste, should fix the sch diagram for VBAT &  VREF to get it right.

xnk, 2021-12-13:
That would explain a thing or two. The schematics was created by a fellow user by reverse engineering a board more thorough than I did back in the day, those connections may still have easily been missed. If you want to provide an updated schematics and contribute that it would be great!