UnifiedEngineering/T-962-improvements #222: 2X  MAX31850K from adafruit

elmenno, 2022-01-07:
I have 2x the adafruit  MAX31850K https://www.adafruit.com/product/1727

i connecting the date to date en v3.3 to v3.3 en GND to GND

its te same situation as used by Werner Johansson https://github.com/UnifiedEngineering/T-962-improvements/wiki/Better-thermocouple-interface
[image]
[image]

Can somebody help me out please?
[image]

same temp readings 
See the pictures

xnk, 2022-01-07:
You hooked it up to Vin, not 3v3 on the Adafruit board according to the pictures, will that really work directly from a 3v3 line? In your case better hook Vin to the 5V rail on the controller, or bypass the regulator on board the Adafruit board. What does the serial console output, can you see the 1-wire chips being detected?

Also, note that the A0-A3 pins needs to be connected (A3, A2 and A1 grounded on both boards, A0 grounded on one of them and pulled to DQ or 3v3 on the other). This is to allow the firmware identify which TC interface is which and it needs to see two interfaces at either ID 0 and 1 (as described above) or at ID 2 and 3 (where A1 is also pulled high).

elmenno, 2022-01-07:
Realy thanks for the quick replay now i am happy
the cold juction temp works  the DS18B20

Do you mean this? see pict
[image]
ure red marking is for soldering

i go soldering now the correct voltage super thanks Werner

elmenno, 2022-01-07:
I hope it wil work now (-:

xnk, 2022-01-07:
Ah I didn’t spot the address select dog bones on the bottom side. Leave a1 a2 and a3 at “0” on both boards and cut a0 from “0” and connect it to “1”.

/wj

elmenno, 2022-01-07:
Werner super thanks for the fast replay

Now it working great for so far now i gong testing the oven 
see the picture
[image]

Greeting Menno

xnk, 2022-01-07:
Awesome!! Glad you got it working!!

borland1, 2022-08-27:
Menno,

Since the MAX31850 chip incorporates internal Cold-Junction temperature compensation, does the UnifiedEngineering TC-961 firmware take this into account?  Did you subsequently remove the DS18B20?  Seem unnecessary with the MAX31850's being used.

GitLang, 2022-08-28:
The DS18B20 is now used to monitor the electronics enclosure temperature and (I think) control the very small fan for that enclosure.

borland1, 2022-08-28:
GitLang,

The small 12V DC fan is connected directly to the power supply, so normally this small fan runs at one speed.  The wiki describes a circuit board mod, using a transistor switch, to add processor control  of this fan using a transistor.

The firmware of this repository adjusts the oven thermocouple temperature by adding, if detected, the DS18B20 reported temperature.  With the two MAX31850's, which have there own internal cold junction compensation, I'm not sure about how this UE firmware handles the compensation calculation. 

This code snippet from the sensor.c file.
```
coldjunction = OneWire_GetTempSensorReading();  <--
		if (coldjunction < 127.0f) {
			cjsensorpresent = 1;
		} else {
			coldjunction = 25.0f; // Assume 25C ambient if not found
		}
		temp[0] = ADC_Read(1);
		temp[1] = ADC_Read(2);

		// ADC oversamples to supply 4 additional bits of resolution
		temperature[0] = ((float)temp[0]) / 16.0f;
		temperature[1] = ((float)temp[1]) / 16.0f;

		// Gain adjust
		temperature[0] *= adcgainadj[0];
		temperature[1] *= adcgainadj[1];

		// Offset adjust
-->		temperature[0] += coldjunction + adcoffsetadj[0];
-->		temperature[1] += coldjunction + adcoffsetadj[1];

		tempvalid |= 0x03;

		avgtemp = (temperature[0] + temperature[1]) / 2.0f;
```

szenieh, 2024-08-15:
> Ah I didn’t spot the address select dog bones on the bottom side. Leave a1 a2 and a3 at “0” on both boards and cut a0 from “0” and connect it to “1”. /wj

May I clarify if your instruction to: "cut a0 from "0" and connect it to "1" " is for both boards or only for one of them? Thank you,

Salah Zenieh

szenieh, 2024-08-17:
I also have 2x Max38150K from Adafruit- the same exact model as in this post. I am unable to get them to work. I have configured the address on one them to be 0 and on the other one to 1. The firmware does not recognize them.  I have tested them with an Arduino and they work great. So, it must be a configuration issue so that the firmware can recognize them.

Should I configure them as 2 and 3? (making A0 (1, 0) and A1 (1,1)? Please help! Could elmenno describe how his worked?

Also, is how do I monitor the boot process via the programming header? Should I hook it up to a serial program like PuTTY? Will that work?