UnifiedEngineering/T-962-improvements #223: Temperature value mixing

djuseeq, 2022-02-03:
Hi and first of all thanks for your effort to rewrite the firmware and for sharing.
I got my oven and immediately i did the improvements what is recommended and updated the FW. 
First i used for CJ a DS1820 with one wire setup and i got strange temperature value for the right and CJ sensor.
[image]

 In the serial terminal i saw the MCU has detected the DS sensor as 18S20. So i tried with 3 wire setup (separate Vcc) then i got this.
[image]

After i found a 18B20, this sensor give back a correct value but the R sensor have exact the same value as the CJ sensor. 
[image]

This is the serial debug message
[image]

and this is my pcb
[image]

xnk, 2022-02-08:
Hmm the fact that the R channel would show the same as the cold junction would indicate that the thermocouple amplifier is completely shorted. If you have the possibility to flash the original firmware, which temperature is that one showing? (It will assume a 20C cold junction and otherwise just take the average value of the two, so it may be difficult to spot that something is off. You haven’t touched any of the cables going to the thermocouples what it looks like, and not the trimmer potentiometers either, so I’m surprised by this zero reading.

djuseeq, 2022-02-11:
Thanks for your reply. You were right. I have checked on the OPA left channel output i measured ~20mV while on the right channel was 0mV. I swapped the temperature probes, i got the same result so the problem is in the amplifier circuit. I checked all the resistors + capacitors in the amplifier circuit, they had the right values ( capacitors compared with caps in the left channel). I set 60 celsius and turned on the oven then the right channel has showed some value but it was too low. Then i set it to maximum feedback for the OPA right channel and i get this value
[image]

The value of CJ has jumped up maybe because is not glued to the screw terminal yet. Now i know why you guys use an a external circuitry for the temperature amplifier, i think i will do the same ;) . Thanks

closed 2022-02-11 by djuseeq