UnifiedEngineering/T-962-improvements #224: Add module options to documentation

Resonanz, 2022-02-15:
The Maxim 1-wire device is also available as a module, and cheaper than the device on its own...

https://www.dfrobot.com/product-164.html

... and it comes with wires and mounting holes for mounting away from the PCB is necessary.

southern-boy, 2022-02-16:
Where do you and/or the group mount your module or place your MAX18B20 
probe inside the T962?  Lay it on the bottom? Place it somewhere in 
particular?  Place it near the top? Lay it on the grate used to elevate 
the board above the bottom? Is there a better place to locate the probe 
or module?

Thank you

xnk, 2022-02-16:
Always interesting when modules are cheaper than the components. In this case the module will likely interfere as the cold-junction sensing sensor ideally should be glued to the connector body, but feel free to add a link to the module in the wiki!

Resonanz, 2022-02-16:
I bought the module today at about 60% of the cost of the device alone. It should arrive within a week. Btw, the + on the end of the IC part number differentiates the lead-free (Pb-free) version.

xnk, 2022-02-16:
The 18B20 needs to be in thermal contact with the thermocouple terminals on the controller PCB (for cold-junction compensation), it is not meant to measure the actual oven temperature.

Resonanz, 2022-02-16:
Mmm, I assumed it was measuring ambient (room) temperature... do I have that wrong?

Resonanz, 2022-02-16:
Ok, I found this very good article titled "Cold Junction Compensation in Thermocouple | Maxim Integrate".

The website link is https://www.maximintegrated.com/en/design/technical-documents/app-notes/4/4026.html.

IIRC, Maxim bought Dallas Semi about 15 years ago, so the article connects Maxim to Dallas who make the silicon device.

Resonanz, 2022-02-16:
A nice summary sentence or two from Example #1 in the above article is the following:

"A temperature-sensing IC, located in close proximity to the thermocouple connector, measures the temperature in the vicinity of the cold junction. This method is based on the assumption that the temperature at the IC is approximately the same as the temperature at the cold junction."

The "cold junction" referred to here is the junction where the thermocouple wires meet the connector contacts.

Thanks for the advice above guys. Feel free to add this stuff to the wiki.