UnifiedEngineering/T-962-improvements #227: Flashed New Firmware Blank Screen

Legohead259, 2022-03-20:
Hello!

I recently purchased a T962A with the 2021 hardware version. I began the upgrades suggested by this repo but encountered an issue. I managed to flash the newest firmware, but on reboot, the screen is blank, the fan runs, and there is no communication from the serial port. I have verified that there are no shorts and the 5V and 3V3 power rails look fine on the board.

I would appreciate any assistance!

Thank you,

Legohead259, 2022-03-21:
Unfortunately, I am a complete idiot and did not save a copy of the firmware. Would not help me anyways at the moment since I cannot seem to get the machine into a boot loader state. When I attempt the ISP boot loader sequence, there is no response to my synchronization query (i.e. the LPC2134 does not respond with "Synchronized" on the UART port when I send it "?"). 

I have not heard anything to contrary for 2021 hardware. Supposedly one guy flashed it just fine, but I can not confirm or deny that. Do I need any of the suggested mods (better themocouples, cold junction compensation, etc.) to make the updated firmware work?

GitLang, 2022-03-21:
That's possible, but you would expect a warning message on a programmed screen, rather than failing to boot.

As for improvements, there's a lot of info here, especially in the earlier days. There are some essential mods, first being sure all parts of the case have a protective earth, scraping paint off each joining area. You should also look at replacing insulation and the heat reflective golden coloured polyimide tape.

Of all the functional improvements, I think the best is the small board which adds interface for four thermocouples, with accurate linearisation, interfaced through spi/i2c. It also has a one wire temperature sensor which gets used to speed control the tiny fan in the side of the case to cool the electroncis. Don't know if the A model has that or not.

Legohead259, 2022-03-21:
> Thanks for the information.
>
> It's being finicky. It's not consistently booting at all. Sometimes it'll
> work. Then sometimes it just won't.

Could it be the front panel buttons malfunctioning? Specifically the F1
button being weird and forcing it to the bootloader intermittently? Does
the F1 pin need to be pulled up?

I've checked all power rails and everything seems to be working fine

xnk, 2022-03-21:
Hmm, that’s odd indeed! Do you have any images of this board?As far as we’ve seen no major changes has been made to the layout for years (up until the T962A+ at least). The fact that it won’t boot the firmware and won’t enter ISP mode seems very strange.

Legohead259, 2022-03-21:
Indeed. I didn't notice any hardware revisions when looking through the
guide/previous posts. I will update this when I get back to my lab with
pictures.

Interestingly, I know how to get the oven to boot now. I can just toggle
the power switch off/on rapidly and sometimes the machine will turn on
normally. It operates as intended as far as I can tell. Is there an
inductive/capacitive load somewhere that the controller needs in order to
boot properly?

Legohead259, 2022-03-22:
[image]
[image]
[image]

rattboi, 2022-07-04:
I have this same hardware revision, and am experiencing the same issues. It was working quite fine for a while, but on occasion, it would not start up and get a blank LCD. I would leave it and try again in a bit, and it would work again.

I've now made some other mods and upon reassembly, it no longer wants to boot to normal mode. However, I can always hold F1 and get it into bootloader mode, so I know some code is running from the new firmware.

When it tries to boot into normal mode, the LCD flickers on for a second, then the backlight goes off. The fan and lamp lights also stay on for what I would say is 0.25 seconds before turning off. 

It's quite strange, because on occasion it does turn on as expected, but for the most part I cannot get it to boot normally.

danielzinner, 2022-08-10:
I had a similar issue. Removing the Temperature sensor fixed my issue. I then added it back but supplying it 3.3V via a 100ohm resistor and a 10uF capacitor to the VDD pin directly instead of the parasitic power. The 100 ohm resistor and 10uF capacitor act as a RC- Filter on start up. It seems the MCU doesn't like GPIO.07 being pulled High when it is booting for some reason. This fixed my issue...

danielzinner, 2022-08-10:
on another note, it seems these boards are really badly soldered. It might be worthwhile inspecting all the solder joints and re soldering any suspicious joints.

rattboi, 2022-08-10:
I figured out the root cause of what was happening in my case, by bisecting the firmware and adding back initialization pieces one at a time. It turned out that in my attempt to add a pin for Vcc for programming, I broke one of the i2c resistors and so there was no communication to the EEPROM. No EEPROM -> blank screen. It gets stuck in an initialization step in a `while` loop.

Fixing the pull-up resistor got everything working again, and I flashed back the newest firmware and things are working as intended again.

danielzinner, 2022-09-14:
> Hi Daniel Zinner!
> The pull-up resistor that you have used which value has it?
> 5Kohm is the right value.

Yeah 5k

Legohead259, 2022-10-11:
Thanks for the information.

It's being finicky. It's not consistently booting at all. Sometimes it'll
work. Then sometimes it just won't.

zacharytomlinson, 2024-08-10:
@Legohead259 Did you ever find a resolution to this? I am just now starting to pick through the code to make sure there aren't any potential issues with a newer model and did note a small potential issue with timing and screen population

Legohead259, 2024-08-10:
@zacharytomlinson No. I have also left the lab that had this machine, so I won't be able to test any fixes. Can reach out to the team though.

We just kept power cycling the machine as described above whenever we needed to use it.

zacharytomlinson, 2024-09-28:
@Legohead259 I was able to get it running stably on the newer model without significant modification. Thanks!