UnifiedEngineering/T-962-improvements #228: Anyone Purchased T-962A+ Lately - Maybe T-962A + Mods Here Better Choice?

jwestmoreland, 2022-04-13:
Hello Everyone,

Anyone tried to pick up a T-962A+ lately?  Seems you have to pay some steep import duty fees to get one.

Maybe just getting an available T-962A and doing the mods is a better/equivalent approach?

Note this is related to post #209 .

Thanks,
John W.

PricelessToolkit, 2022-04-17:
Hi yes, I recently bught a T-962A+ it has the same problems as the rest of the younger models, can't be used without modification.
1. Paper tape inside
2. Uneven heating
3. PIDs are configured incorrectly "temperature overshoot then cooling down with fan"

since there is still no working firmware for the built-in controller, I will install another, 
I have such https://github.com/polihedron/DIY-Reflow-Oven
And inside I plan to install two low RPM fans

xnk, 2022-04-18:
I’d say the T-962A+ is even worse than the older models unmodified due to the airflow direction change (which seems to have propagated to the older models now as well). Reversing the fan helped a lot in my case but if you install low-RPM stirring fans inside that’s a moot point of course.

The T-962A+ I have here has a much more even heating compared to the older models (especially T-962), but that’s a hit or miss quality wise. I still haven’t had time to work on the ADC and 1-wire stuff for the STM32 (the ADC stuff is kind of pointless as one of the first modifications should be 1-wire/i2c tc interfaces anyway).

PricelessToolkit, 2022-04-18:
> 

I noticed a 15-22C overheat and then a 10-15C overcooling. The real temperature profile line is like a zigzag :)
I hope you will have time to finish the firmware, I will support your work with two cases of beer 👍  Tell me where to send it?
Thanks

PricelessToolkit, 2022-04-22:
[image]
[image]
[image]
Made a modification today, Do you think additional cooling is needed for DC motor?

PricelessToolkit, 2022-04-22:
> @PricelessToolkit,
> 
> Nice mods - but I'd check the specs on those motors - and maybe see if there's any MTBF data.
> 
> Try to seal all of the air gaps there without getting kapton on the rotor of course.
> 
> Do those motors have a temp rating that's reliable? I was wondering about some drone motors - and just run them as slow as possible since they're built to take a beating - not sure if you can run them that slow though - but will be a fun problem.
> 
> Do you have a P/N for those motors?
> 
> Regards, John W.

Hi, @jwestmoreland  Motor model is "Mabuchi rf-500tb-12560", I have already air-sealed it, but the problem is HEX stands, there can transfer heat. I will check it tomorrow on how  match it heats up.  PS: With 3v PSU speed is ok!
Datasheet https://datasheetspdf.com/pdf/1415685/Mabuchi/RF-500TB-12560/1
Ali: https://fr.aliexpress.com/item/4000369591592.html?gatewayAdapt=glo2fra&spm=a2g0o.order_list.0.0.28075e5br0rLqq

jwestmoreland, 2022-04-23:
@PricelessToolkit ,

Doubtful they will survive long > 150 degrees Fahrenheit.  Too bad those motors can't be on the bottom vs. the top.

Good Luck,
John

mikeanton, 2022-04-23:
@PricelessToolkit 

As it happens I'm in the process of repairing one of those cheap fireplace fans that are powered by a Peltier thermopile.  It happens to use the same motor style.  It didn't last long before the Peltier died, and the motor oil evaporated, but once the motor was oiled, it came back to life.  These seem to be a common motor used in this application, so maybe they can handle a bit of heat, or are just really inexpensive.

[https://www.aliexpress.com/item/1005003407344164.html](https://www.aliexpress.com/item/1005003407344164.html)

PricelessToolkit, 2022-04-23:
Checked, with vertical blades and the result is not good. The centerline "from front to back and sides" heat up more than under the fans.  Tomorrow I'll try to change the blade's angle so that they blow air upwards.

mikeanton, 2022-04-23:
@PricelessToolkit 

I've observed this on the T937M as well.  It comes stock with a convection fan built in, and it is definitely cooler right under the fan.  It has shaped blades the move the air either up or down, but I'm not sure which way the air moves.

How are you measuring the temperature?  I've been meaning to pick up a 16 channel thermocouple data logger for doing these sorts of measurements.  It should be fairly easy to determine how even the temperature is using one of those.

PricelessToolkit, 2022-04-23:
I measuring without spending 2k in logger :)   several small PCBs with PBFree solder paste on it

mikeanton, 2022-04-23:
You can get inexpensive Chinese ones for under $150, that have a Modbus interface.  Then some external logging software should be all that is required.

Using some temperature indicating wax would be better if you are not using thermocouples.

Solder melting alone isn't a good indicator of heating evenly.  I can dial in my oven to reach soldering temperature at any place in the oven, but there will be some location that gets far too hot as a result.  Generally, it's not a good idea to exceed 250C anywhere in the oven, as many parts are not rated for reflow temperatures higher than this.

One thing that does help to increase achievable temperatures, and rate of change, is to place the PCBs on some sort of riser to keep them off the bottom metal grate.  I have a bunch of tensioned wires running across the oven 3/8" above the grate, and the board sits on these.  This allows the board temperature to change quicker without heating up all of the thermal mass of the oven.  It probably doesn't help keep the temperature even though.

[image]

PricelessToolkit, 2022-04-23:
@mikeanton Thanks!!! for the next test, I will rise the PCBs

GitLang, 2022-04-24:
Agreed, raising the board up is pretty much essential, especially if there are components both sides, and you are reflowing the second side. I usually put an M3x6 screw + nut in the mounting holes, but I like your idea. The  possibility of the rear side board tinning bonding to the wire worries me but I'm sure you would have mentioned it. What material is the wire?. Iron?

mikeanton, 2022-04-24:
I've been running the oven like this for a couple of years, and thousands of double sided boards without any issues.  I'm pretty careful to make sure the wires are not near bottom side components.  The wire is just regular 22 ga tinned copper bus wire, which is very oxidized now.  Stainless steel wire would be much better, and guarantee that nothing would stick.  I loop the wires at the end, and use a ferrule to form a crimp.  Come to think of it, a regular non-insulated ring tongue terminal would work even better.

I used screws in holes initially as well, but that was awkward, as I had to move the screws frequently when building a different board.  This is much more convenient, and still very low mass.  I sometimes put standoffs under the middle of the wires if running heavier boards, as otherwise it can be a little bouncy.

GitLang, 2022-04-24:
Wow, I'm really surprised you didn't get wires soldering themselves to the pcb. Crimp ring would be nice and easy, good idea
Yes, it is a pain having to mess around with screws, but I only do around 4 to 6 boards maybe 3 times a year.

mikeanton, 2022-04-24:
Usually the double sided boards I run have rails on them, so that's where the wires usually sit.  Single sided boards don't usually have much in the way of pads on the bottom, and if there are some, they usually have and ENiG finish, so the wires won't solder to them either.  So, while copper wire isn't ideal, it's what I had on hand at the time, and it has worked so far.  Someday I'll get around to doing it properly.

PricelessToolkit, 2022-12-14:
Hi, I have decided to sell my T-962A+ Used 10-15 times, Can ship within Europe.