UnifiedEngineering/T-962-improvements #229: Can't re-flash upgrade after downgrade to original firmware

VictoriaBentell, 2022-06-04:
I was having display problems with the new firmware (but that's an issue for another day) so I switched back to what I believed was the original firmware, [provided here.](https://github.com/UnifiedEngineering/T-962-improvements/issues/103#issuecomment-574698755)

Now when I turn it on normally, the screen is just blank. Even worse, I'm unable to get back into bootloader mode using the [wiki instructions](https://github.com/UnifiedEngineering/T-962-improvements/wiki/Flashing-the-LPC21xx-controller#enter-bootmode-sequence-manually) for both manual and button-press methods. In Flash Magic I'm just getting that "Failed to autobaud" message. Crystal frequency is set to 11.059 MHz (and I don't understand if the fact that it automatically cuts it down from 11.0592 is causing problems, but that doesn't seem to affect anyone else).

When I turn on the machine, the screen is just blue and the board's LED is a bright solid red and the electronics fan spins continuously. Once n_RESET is grounded, the red LED becomes dim and the green front lights turn on, along with the oven fan and the heat. I believed that this was my "new original" firmware's bootloader mode, but this didn't work either. And grounding n_ISP doesn't seem to make a visible difference here. Releasing n_RESET brings the red LED bright again and the oven fan/heat/green lights all shut off. None of this seems to change the display.

I'm also trying various combinations of baudrate and erasing "entire device"/"sectors used by file" but the same message appears.

Any ideas on how I can get back into bootloader mode?

sevstels, 2022-06-06:
It is very likely that your COM adapter is not working correctly. 
To check this do the following sequence.

1. Connect the wires to the computer to the COM interface.
2. Turn off the oven power for a few seconds.
3. Open a terminal program such as COM Port Toolkit
4. Set the speed to 57600
5. Switch the oven processor to boot mode
6. Send to oven a sign several times: ?
5. See what the oven responds

Write the response here.
Then we will think about it depending on the answer.

borland1, 2022-08-14:
The bootloader is a built-in feature of the LPC2134 microcontroller, so regardless of what firmware is loaded, the bootloader should load with n_ISP pulled low while RESET is toggled low (or with Power Up).  For LCP2134 with a 11.0592 oscillator, the datasheet states that the only autobaud rates supported are 9600, 19,200, and 57,600 baud. 

The oscillator frequency value changing by the FlashMagic, flash utility, from the entered value of11.0592Mhz, to a value of 11.059Mhz, is bug, but does not affect serial communications functioning properly.

I had similar problem where FlashMagic could not connect with ISP,  T-962 oven.  Problem was due to an internal short in LPC2134 microcontroller, causing the microcontroller chip to be too hot to touch when powered up.  Solved this issue by replacing the LPC2134 microcontroller with a new one and then burning the firmware with FlashMagic.

VictoriaBentell, 2023-03-01:
> It is very likely that your COM adapter is not working correctly. To check this do the following sequence.
> 
> 1. Connect the wires to the computer to the COM interface.
> 2. Turn off the oven power for a few seconds.
> 3. Open a terminal program such as COM Port Toolkit http://old-dos.ru/dl.php?id=25319
> 4. Set the speed to 57600
> 5. Switch the oven processor to boot mode
> 6. Send to oven a string sign several times: ?
> 7. See what the oven responds
> 
> Write the response here. Then we will think about it depending on the answer.

I'm not getting any response when sending either ? or U. The oscilloscope is telling me that while Rx shows a signal, Tx is stuck at 3.3v. It's very much possible that last year I destroyed it by putting it into bootloader mode by grounding the ISP to the chassis ground, not the FTDI ground. I now realize there is a huge voltage difference between the two. I'll close the issue and take @borland1 's suggestion and replace the microcontroller. 
(Apologies for the extremely delayed reply. I put the project on the backburner and moved onto other things.)

closed 2023-03-01 by VictoriaBentell