UnifiedEngineering/T-962-improvements #231: "COLD JUNCTION NOT PRESENT"  DS18S20 not available from DigiKey; DS18S20+PAR substituted

CurtUSA, 2022-07-20:
###Is the DS18S20+PAR a direct replacement for DS18S20?
**BACKGROUND**
Others have noted and I too appreciate the excellent documentation available to upgrade this oven. I have a time-critical antenna project and need to get a volume of SMD's soldered to PCB's.  

**SYMPTOM**
I closely followed VoltLog's video (replaced painter's tape; grounding improvement; firmware upgrade) and am now installing a cold junction reference. I've purchased six each DS18S20+PAR for this purpose from DigiKey, carefully installed three -- and for all three installs the message "COLDJUNCTION NOT PRESENT" is present.  

**TROUBLESHOOTING NOTES**
(1) Pull up resistor measured as 4.7k ohms
(2) Voltage on both sides of the resistor measured at 3.3 volts (makes sense to me, as the device pulls very little current)
(3) I carefully held device leads with pliers prior to bending two of them, to ensure there are no continuity breaks inside the device
(4) I have a scope and fluent in using it, but not sure what signal I would be investigating -- happy to use it on someone's advice. 

**QUESTION**
**As the DS18S20 now appears unavailable from a reputable supplier (see attached screenshots from DigiKey), is the DS18S20+PAR a direct substitute, or is there a firmware modification needed?**  _(Apologies -- I compared the data sheets as best I can, but don't understand enough about firmware communications to determine if the two devices are direct substitutes.)_  

[Data Sheet DS18S20-PAR.pdf](https://github.com/UnifiedEngineering/T-962-improvements/files/9150453/Data.Sheet.DS18S20-PAR.pdf)
[Data Sheet DS18S20.pdf](https://github.com/UnifiedEngineering/T-962-improvements/files/9150457/Data.Sheet.DS18S20.pdf)

[image]
[image]
[image]
[image]
[image]

GitLang, 2022-07-21:
As far as I can see from the datasheet the are identical except the use of 1 wire only for the -PAR. I think your problem is your wiring to the actual device. It looks like you have wired to pins 1 & 3 (the two outside pins). In the -PAR flavour pin 3 is not connected. You should be connecting to pins 1&2.

CurtUSA, 2022-07-21:
Thank you for the feedback! I apologize, for I forgot to mention in my original report that in one of the previous configurations I had clipped Pin 3; the same error message "COLD JUNCTION NOT PRESENT" occurred. I repeated this configuration this morning (see attached photo), unfortunately with the same result. -Curt

[image]

borland1, 2022-08-14:
I purchased a few Dallas 18B20 from AliExpress.  I haven't installed one in my T-962 oven yet, as I'm still waiting for the 4.7K SMD resistor, but did test 5 of them with a 3.3V Arduino and a One-Wire Library.  Tested good with both pin 3 (VDD) grounded or unconnected.  Couldn't see any difference in performance between pin 3 grounded and unconnected.  Datasheets I looked at say to ground pin 3 for Parasitic Power Mode, but does not seem necessary with the devices I bought.

Seems like a more ideal place to mount the 18B20, is next to the Red LED, in the two unused plated thru-holes.  One hole is tied to ground, the other hole is not connected to any traces, so just a short wire from data line to the 4.7K resistor.

drmhtd, 2022-10-25:
Your resistor is marked 4R7.  That is a 4.7 ohm resistor.  You will always see 3.3V on the signal because the device will not be able to drive it to a logic 0.