UnifiedEngineering/T-962-improvements #234: I made a OpenOffice Calc sheet to help generate custom reflow profiles

dburr, 2022-09-27:
Recently I decided to make my own reflow profiles. Editing them on that little screen was absolute murder, and since I wanted my new profiles available permanently anyways, I decided to build a custom firmware with my profiles in it. I needed a way to visualize the temperature curve while adjusting the temperature points. So I came up with a little OpenOffice Calc spreadsheet that does just that. It shows a graph of what the temperature curve looks like as you enter the data, and when you're ready, it provides a line of code that you can directly copy and paste into `reflow_profiles.c`.

[image]

[Get it here](https://www.dropbox.com/s/zrnm1drltylivp6/ProfileMaker.ods?dl=0)

mikhailmihalkov, 2022-09-30:
There is a similar project... Can be sent immediately via uart  
[Reflow profile calculator.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/9683548/Reflow.profile.calculator.zip)

borland1, 2022-10-06:
The problem with this approach to developing Reflow Profiles, is that the profiles are just **guides** or **starting points** for making adjustments required that match the intended PCB and its associated SMD components with the solder paste being used.  The solder paste Manufacturers have their own reflow profile recommendations, while Component Manufacturers also have suggested reflow profiles.  So, finding a sweet spot usually requires some tweaking of the Reflow Profile.

I have developed a spreadsheet that generates a reflow profile based on separate input parameter for the various phases of the profile (e.g., Preheat, Soak, Reflow, Cooldown).  Here's a screen shot of the input parameters and resulting chart (Reflow Profile).

[image]

The spreadsheet also generates a Time/Temperature table in 10 second intervals up to 470 seconds.

[image]

I'm not going to provide copies of my spreadsheet, but do plan to code this implementation, and then submit a **Pull Request** that implements this approach in firmware.

This will modify the firmware so that it can generate all Reflow Profiles based on the same input parameters as my spreadsheet, for either firmware Stored Profiles, or Custom Profiles,  That way the user can select from those provided profiles and make parameter adjustments, as necessary.  Similar to current profile features, the user should be able to save Custom Profiles to the oven's EEPROM.

mikhailmihalkov, 2022-10-06:
Unfortunately, the problem of these stoves is not in profiles and firmware... Without a complete replacement of heaters and the addition of a convention, they do not work normally on the profile... There are two main problems are the low heating rate (not dependent on the graph) and not the uniformity of heating... In the first case, it is decided to replace the heaters in the second by adding a convention fan.

mikhailmihalkov, 2022-10-06:
https://budgetlightforum.com/node/66551

mikhailmihalkov, 2022-10-06:
[image]
[image]
[image]
[image]