UnifiedEngineering/T-962-improvements #236: The source codes are posted

sevstels, 2022-11-15:
If you want, you can try to figure it out and modify it to suit your needs. 
[description page](https://en.gradient-sg.com/t962/#downloads)

mikhailmihalkov, 2022-11-19:
Eh used quartz lamps at one time and it was a mistake. They have too much inertia.

WalterMurphy, 2022-11-19:
"Quartz lamp" is a term very generic, what differentiate IR lamps is their wavelength (Near Infrared, Medium IR, Far IR), so you might have had a bad experience with the wrong lamp for the application. Far infrared lamps (eg, transparent quartz tube with coiled carbon filament heater) are better fit for this application than the current tubes used on the T962A. Far infrared lamps are less color selective and have very short delay. Ceramic IR emitters are better used in conveyor ovens together with convection fans.

Check the book "Reflow Soldering Apparatus and Heat Transfer Processes, Elsevier 2020" for more information.

WalterMurphy, 2022-11-19:
sevstels,  Good job!!

mikhailmihalkov, 2022-11-20:
The fact is that in industrial furnaces there are many melting zones and we have only one... That's the big difference... Your board falls into a heated chamber and does not heat up from scratch in one chamber...

WalterMurphy, 2022-11-20:
I agree, if we could afford (both financially and physical space) a multi zone conveyor reflow oven we would probably not be trying to improve a T962 or derivatives. 

In the early 2000's I modified a toaster oven with two IR  top elements and two IR bottom elements. PID controlled using ATmega16pc @ 4 MHz. Serial communication (rs232) with PC and 4 lines LCD display. PWM output was synchronized to the zero crossing of the power line AC and consisted in a 512 power line semi-cycles time, the pwm power conmutation was equally distributed across the whole 512  (60hz semi-cycles)-cycle using a modified bresenham algorithm, trying to compensate against the IR lamp's inertia and delay. That oven is still in use today. It simply works but the limitation in pcb size made me investigate other desktop alternatives, it is not a multi-zone reflow oven but emulates one "in a good enough way" is is reliable and repeatable (we only use Sn/Pb soldering pastes), so we only change the profile when we buy soldering paste from a different brand. 

I think that sevstels' work is a step forward into another alternative for all of us who don't have the space available to own the bigger multi-zone below ovens.  I also appreciate the work involved into developing his advanced software. Don't misunderstand, I am very glad Unified Engineering developed the new firmware for the original board, but we should not be limited by the original board's limitations (eeprom size, display choice, analog thermocouple amplifiers, etc) only two custom profiles memory, profile customization mode, etc) Some of the limitations are very easily overcame, eg, we could replace the eeprom chip with a bigger memory size Fram chip, same footprint, pin by pin compatible. eeproms have limited amount of write cycles and not that fit for holding user profiles that might change often. Also, the display used is not commonly available nowadays. 

It also seems that nobody has noticed, or everyone has underestimated, the lack of proper thermocouple connectors in the original and on the subsequent "improvements", thermocouple connectors are not made only to inflate the bill of materials, they ensure that there are no other thermocouple-junctions in series besides the "cold junction" everybody now cares about. By using unknown metal contacts in the connection to the board we are introducing a big source of noise and error in the measurements by creating extra junctions.

WalterMurphy, 2022-11-20:
mikhailmihalkov, I saw photos of your mods, on another thread, using ruby IR lamps (near infrared), how was your experience?

mikhailmihalkov, 2022-11-20:
> mikhailmihalkov, я видел фотографии ваших модов, в другой теме, с рубиновыми ИК-лампами (ближний инфракрасный свет), как ваш опыт?

Of all types of lamps, these are the fastest.... They hold a thermal profile very accurately and have the smallest inertia they are perfectly friends with the convention.... Unlike those supplied from the plant.... I have already redone three furnaces 962,967....

WalterMurphy, 2022-11-20:
Thank you, Congratulations!!

mikhailmihalkov, 2022-11-21:
> Спасибо, поздравляю!!
https://budgetlightforum.com/node/66551

sevstels, 2022-11-21:
>> used quartz lamps it was a mistake

In my opinion, the speed of quartz lamps is enough. It is important to understand the basic principle - there should be a lot of lamps... So that the array of lamps works without significant overheating relative to the board temperature and PID regulator does not enter into saturation. It is important that the coil heating is as low as possible and the lamps emission is as long wavelength as possible and does not go into the visible red or orange range of the spectrum. If you have only one or two tubes in your furnace - to heat up the contents of the soldering chamber, you definitely have to overheat them considerably! In this case, the performance of the system will be the worst. Almost all single chamber furnace modifications that I have seen contain this very error. The lamps are few and glow in the visible part of the spectrum. In a "proper" soldering furnace, you shouldn't be able to see the operation of the lamps with your eye.

And so, if you have installed a lot of additional lamps, the dependence on their speed will no longer be critical. The power will be enough to heat up the board with long wavelength radiation only. The main delay now will be the thermal inertia of a board with massive components. Such a board heats up slowly and cools down slowly.  If you try to give more power to strictly follow the profile the lamps will go from the long wavelength range to visible red and the shorter wavelength light will melt and ruin the plastic of the component housings. This should not be allowed to happen. Therefore, when heating, you should move in small increments of temperature increase, e.g. +2 degrees. Take another step and wait until the temperature is established. The temperature step should be dynamically recalculated, automatically dividing the interval remaining to the point of extremum by the desired amount of temperature increment. It is possible to install an additional sensor for the intensity of visible radiation and limit the heating by it. In this case there will be no overheating of the components by the short-wave light radiation. Because the lamps will be in a small incremental current mode relative to the previous step and the board will warm up qualitatively and evenly. The temperature profile will strictly follow the temperature, but the time of its movement will dynamically change.

Translated with www.DeepL.com/Translator (free version)

mikhailmihalkov, 2022-11-21:
sevstels
1. Those lamps that come from the factory cannot provide heating speed as on the factory schedule. Even if you put them 6 and not 2. (you will have to stretch the schedule in time)
2 If you add a convention (fan), the picture becomes even worse
3 Inertia when they cool down is much more than that of halogen lamps, cooling according to the schedule becomes impossible (with their native) fan
4 Everything is learned in comparison (by experience)
This applies to all furnaces of this manufacturer even the most powerful 937m

sevstels, 2022-11-21:
I threw out the original bulbs. They were of very poor quality. Their spirals were not wound evenly. There was some sand inside the bulbs, so the inertia was very high.

You should use normal bulbs, with a lightweight quartz tube and a thin coil. My new bulbs were 300 watts with a thin coil. That's why they are fast enough.

sevstels, 2022-11-21:
If you have the desire and skills to help us make the T962 app for Android. Please write me to coordinate your efforts.

WalterMurphy, 2022-11-21:
I  just received my T962A, opened the top cover and got rid of the visible masking tape. Since my oven is a little bigger than the T962, I can't find the proper lamps to try sevstels' IR tube arrangement yet (I also need to do the proper spacing calculations for my oven), but first I need to keep on disassembling and make some measurements including the lamp sizes, distance to the pcb placement location on the drawer, etc
. 
I agree that the main pcb heating medium for this oven is IR radiation and not the air temperature, so, I won't try any convection mods until proved wrong.
 
As I was commenting previously, I did a toaster oven modification, for reflow purposes, about 20 years ago. I still have it, never bothered to replace the right lateral covers but it works fine for my needs.
 
I remember that one of my concerns at the time was trying to heat the pcb form the bottom too, so that the temperature gradient between top and bottom was limited to only about 30-40 degrees (worst case), so I also included the bottom heaters into independent PID control. Thermocouples on top and bottom of the pcb. Bottom  IR tubes were PID controlled up to just the final of the soaking stage, from there up the top tubes' PID (and top thermocouple) took control of the oven's heating until the final of the profile.
 
I can't find now literature to backup the need to limit the  temperature gradient between top and bottom of the pcb, but  having bottom heating also helps a lot when the top tubes are limited in power. The pcb to be reflowed rests on top of a wire grid instead of a tray, so it is easy to get IR heat evenly. PCB sizes are limited to about 10 x20 cm, the maximum (rated) power is 1500W. Not enough for unleaded  solder paste profiles because I also never bothered to insulate the small oven chamber. No convection fan involved, cooling by opening the door slightly. Profiles were saved in a monitoring pc, also each board's reflow logged data was saved in order to help investigate possible failures if there were problems when deployed.

sevstels, 2022-11-21:
>> The presence of bottom heating is also very helpful when the power of the upper tubes is limited.

I agree. Bottom heating is necessary.
But the mechanical design becomes very complicated.

WalterMurphy, 2022-11-22:
quote: "..I can't find now literature to backup the need to limit the temperature gradient between top and bottom of the pcb" ...
I was trying to remember the reason why I bothered so much about temperature gradient at the time. and finally recalled that there were some papers describing pcb warping due to temperature gradients and it mostly affected BGAs.

mikhailmihalkov, 2022-11-22:
> цитата: «... _Я не могу найти сейчас литературу, подтверждающую необходимость ограничения температурного градиента между верхом и низом печатной платы_ » ... Я пытался вспомнить причину, по которой я так беспокоился о температурном градиенте в то время . и, наконец, напомнил, что были некоторые статьи, описывающие коробление печатных плат из-за температурных градиентов, и это в основном затрагивало BGA.
In most cases, this is decided by the convention (in a closed chamber)

WalterMurphy, 2022-11-23:
Natural convection helps when there is a lower heating element to heat the pcb from the bottom, unfortunately using only top heating elements (without forced convection) colder air tends to accumulate below the pcb level so contributing to create the vertical temperature gradient that produces pcb warpage. The bigger and thicker the pcb the most probability of warpage in presence of temperature gradients, if the oven has horizontal hot spots at the pcb level those will also contribute to local warpage. The pcb substrate material also plays a role on the propensity to warpage.

WalterMurphy, 2022-11-24:
> _5 The wavelength of halogen lamps can be changed by the supply voltage_

I think we all are talking about the same subject in different words. Yes, the above statement is correct for all the IR lamps to produce far IR radiation depending on the filament temperature, the problem is efficiency, predominantly near and medium IR  lamps will require a power limit in order to produce more IR in the long wavelength range (by dimming the filament glow because of a lower filament temperature) but,  to get the same thermal effect, you will need to add more lamps, If you use the same power lamp with predominantly long wavelength  (ceramic radiators or carbon filament quartz tubes) you will have more radiation in the far IR zone with the same power without adding more lamps and without having to limit the power. 

I also agree with sevstels calculation to get an homogeneous radiation coverage by placing the heaters at the right distance between lamps, and in order to cover more area it will be necessary to add more heaters. He had to limit the power in order to get most of the radiation in the far infrared range using standard quartz tubes
So we are talking about the same subject, the best thermal effect (for IR reflowing purpose without forced convection) is done by the far infrared radiation which is not color selective.

wulfmans, 2022-11-24:
I do double sided boards and the oven after the mods works well.
if there was heating on the bottom doing double sided boards without 
epoxying the bottom components would not work

WalterMurphy, 2022-11-24:
I also do double side boards using a 2top/2bottom IR elements toaster oven (no fan), one side at the time, first the least populated side with the lighter components (normally the bottom side of my pcbs), on the second round I do the other side. The already reflowed components now on the bottom side of the pcb (resting on the support grid) won't fall down. They are normally capacitors and some resistors. If you had heavier components on the lower side I suggest to use epoxy. I solder the connectors, transformers, etc manually

mikhailmihalkov, 2022-11-24:
Don't come up with a bike when it was invented) Look at the designs of branded proven furnaces with one zone

WalterMurphy, 2022-11-24:
>Look at the designs of branded proven furnaces with one zone

I did, most of the so called "batch reflow ovens" are forced convection type, there are some made in France with very bad reviews in EEVblog due to safety issues and poor response from the manufacturers. Found some made in Japan, which have a motorized tray, they use far infrared heaters with no convection, also are ready for nitrogen atmosphere during reflow. The problem is that there are no other specs or disassembly diagrams in their website. They also use an enclosure around the oven itself, probably to contain fumes.

I was thinking that applying sevstels calculations instead of adding more tubes, with the right spacing, I can also leave the current spacing and increase the distance from the tubes to the tray, I know that, following the inverse square law, the W/cm2 at the pcb level will diminish and I will also require to change the four tubes for more powerful ones to compensate. I need to crunch some numbers to see which option makes more sense economically.

mikhailmihalkov, 2022-11-24:
In general, tested furnaces are those that work in real industries in laboratories, etc... If you do not have the opportunity to look yourself, contact the forums where the contract production works.... This applies not only to furnaces, but also to smd installers.... Such stoves as in this article are the bottom of the market (payment for a low price) If we talk about the most affordable option that actually works and is available to anyone, then this is an example https://www.instructables.com/DIY-REFLOW-OVEN/ the heaters of course change

WalterMurphy, 2022-11-24:
[[]](http://www.jpl.com/products/smt/RF-110N2MK2.html)(url)

WalterMurphy, 2022-11-24:
Examples:

https://www.fortex.co.uk/product/smt-reflow-oven-stainless-steel/
[[]](http://www.jpl.com/products/smt/RF-110N2MK2.html)(url)

mikhailmihalkov, 2022-11-25:
https://github.com/polihedron/DIY-Reflow-Oven

sevstels, 2022-11-25:
The heaters on this stove are also wrong.
The light is orange and the location of the heaters is not correct.

mikhailmihalkov, 2022-11-25:
https://github.com/UnifiedEngineering/T-962-improvements/issues/236#issuecomment-1327417929

mikhailmihalkov, 2022-11-25:
6 halogen lamps included in 3 sequentially solves this problem

sevstels, 2022-11-25:
I was able to connect a bluetooth dongle on a TI CC2540 to control the heater from a PC. There are plenty of them on aliexpress: https://www.aliexpress.com/item/1005001999985901.html

bgrigoriu, 2022-11-25:
> It was quite difficult to buy the GBU521 dongle. It has already been discontinued. I was able to connect a bluetooth dongle on a TI CC2540 to control the oven from a PC. There are plenty of them on aliexpress and cheap: https://www.aliexpress.com/item/1005001999985901.html

CAn you be more specific ? details, Code?

sevstels, 2022-11-28:
The code is not yet very clean.
When I'm done, I'll put it on Gihab.
If you are interested to see it as it is, you can download the project from here:  https://gradient-sg.com/files/t962/newver.zip

To connect to a Nordic UART, you need to manually set the int CC2540::CMD_Connect(void)
the device address. The address of the device can be viewed in the phone for example.