UnifiedEngineering/T-962-improvements #238: T-962 Screen blank after flashing new firmware

jjosse44, 2023-01-13:
Hi,

I'm really stuck here, we got a T-962 reflow oven at my place of work and it desperately needed the modifications to make it more user friendly for certain people. 
So, the first upgrade I went ahead with was the firmware, after so much trouble shooting I was finally able to get the following: 'FT232RL FTDI' adapter to flash the new firmware in flashmagic. It says erasing all data -> programming -> verifying and then ends with no errors. Then I turn off the oven, and turn it back on and the screen is always and still blank with both the 'Fan' and 'Lamp' lights on and no 'Unified Engineering' screen ever shows up, just blank. I really need help, thanks in advance!

borland1, 2023-01-15:
Sorry that you got stuck.  The Wiki here does a good job at showing how to set the oven in bootloader mode, and how to flash the firmware.  You can unplug the AC power to the lamps on the controller board so they don't come on during the firmware flashing operation; at least that's what I did.

Got a link to the HEX file that you used?  What 's the HEX file size and date?

The FlashMagic utility has been updated, so the Wiki is not up to date as far as what the menu's look like.  Here's what I have that was able to successfully upload the hex file dated Feb 4, 2018 ( ~180KB ).

[image]

[image]

[image]

[image]

[image]

tdarlic, 2023-01-16:
> Hi, thanks for replying.
> So I have followed the wiki and many videos the most current one being a
> guy who did it about 9 months ago. I have the same settings as you do with
> my flash magic and the weird thing is that it does erase all the blocks,
> program, and then verify and it LOOKS like everything worked, but then I
> shut it off, disconnect the serial adapter and everything from the ISP
> header and turn it back on and nothing. Blank screen, but it's still lit up.
>
> I have whatever the latest hex file off the wiki is Feb 2018 180 kb as
> well. I had read posts with other people having issues with the display
> with no solution because apparently in 'newer' T-962A units the LCD display
> has a different PCBA part number so the timing is different in the old hex
> file or something, I don't know. However, I've seen no fixes or anything to
> that issue if that is the case. That or the LPC2134 chip is dead and a new
> one needs to be soldered on, but I'm not sure how to exactly check if it's
> a hardware/chip issue before I start trying to desolder stuff etc. Thanks.
>
> Sorry that you got stuck. The Wiki here does a good job at showing how to
> set the oven in bootloader mode, and how to flash the firmware. You can
> unplug the AC power to the lamps on the controller board so they don't come
> on during the firmware flashing operation; at least that's what I did.
>
> Got a link to the HEX file that you used? What 's the HEX file size and
> date?
>
> [... 13 lines cut ...]
> [image: FlashMagic4]
> <https://user-images.githubusercontent.com/3272058/212522128-b56e5845-5173-4faa-9f56-3a1d127a702b.JPG>
>
> [image: FlashMagic5]
> <https://user-images.githubusercontent.com/3272058/212522145-eacc2851-7007-402a-8a6c-44f34c2c4c72.JPG>

Here's my experience from what I can recollect as it was long ago. As the
programming interface is just UART and I did not have an ftdi interface I
just used Digilent Max 32's ftdi chip to which I connected directly. After
some fiddling I got it to work. I remember that I needed to power all from
the same power supply as there was some problem with ground.
Here a google album with some images https://goo.gl/photos/RDBpVCiXKs32xWc76

jjosse44, 2023-01-16:
Thanks for the reply. I think this issue is a little different from what I'm experiencing unfortunately. It's more of a concern of the reflow ovens display not showing any options and being backlight but blank after flashing. As well since you did your mods a while ago it's hard to compare to now since others have had the same issue without success.

borland1, 2023-01-16:
> **Edit**: I noticed in a more recent post yours ended up being a hardware issue and you had to buy a new LPC2134 chip and resolder that. If that does turn out to be the case, any special tools you used to desolder this MCU. It has many leads/traces so I don't want to screw it up. Thanks!
> 

I bought my T962 oven on eBay.com, It was described by seller as for parts, not functional (blank display, heater lamps always on with power).  So, I don't know the oven's history, however I immediately noticed a loose screw which was floating around inside the case.  This unit is a newer T-962 model with the rear exhaust duct outlet.   This unit was probably an Amazon.com customer return that was resold to a liquidator.  The oven had already been opened up, most of the masking tape was already replaced with some Kapton tape.  After I re-flashed the firmware of this repository, I noticed that there was a user profile on Custom #2 which was burned into the board's I2C EEPROM.

When troubleshooting the unit, I noticed that the microcontroller was internally shorted, causing the it to heat up quickly and be too hot to touch.  That loose screw may be what caused the microcontroller to become defective and result in internal short.

I removed the PCB, used my hot air rework station to remove the LPC2134 microcontroller, and replaced it by hand soldering with a new part.  At the time of purchase, there was no inventory of these MCUs in the USA, so I ordered new part from China on eBay.com.  Here's the same listing and seller I purchased from.

[eBay.com:  1PCS LPC2134FBD64/01 Single-chip microcontrollers LQFP64](https://www.ebay.com/itm/221308412609)

To replace the MCU, you really need a digital microscope, solder flux paste syringe, flat/chisel blade soldering tip (like 900M-3C), and experience hand soldering small pitch SMDs. Solder bridges can be easily removed with the right techniques and experience.  I sold my digital microscope a while back, so it wasn't easy to solder the new chip as the pin spacing is quite small.  I masked some low melt temp plastic parts for hot air work. And temporarily removed a couple through-hole components to allow room for hand soldering.

jjosse44, 2023-01-16:
Interesting, I wonder if the previous owner did something similar and messed up flashing firmware or modifying in some way and ruined the chip. It's just weird everything acts like it flashes fine, but then just doesn't work anyway. Even when I try to flash the OEM firmware or any version of other user's original firmware, it programs and verifies the same way, but blank LED screen.

Thank you for explaining your process and giving tips to help with soldering a new chip in! I'll order a new part and reply back with results if it ends up working or I run into any issues.

Would still be nice to know if anyone else experienced this issue and could elaborate on why it seems like it flashes just fine, but the LED screen is completely blank, but lights up.

> I bought my T962 oven on eBay.com, It was described by seller as for parts, not functional (blank display, heater lamps always on with power). So, I don't know the oven's history, however I [...]

borland1, 2023-01-16:
Are you sure you were using a 3.3V FTDI serial adapter?  5V could kill the MCU.

What likely killed my oven's MCU was a shorted output pin to ground.

jjosse44, 2023-01-16:
Yeah the first one I had didn't have a jumper or switch, but I tested the outputs for Tx and Rx and both were ~3.3 V and I bought a new serial adapter that I saw most videos and tutorials using which had a jumper for 5 V or 3.3 V so I set the jumper to 3.3 V and tested the output to make sure. I felt like maybe I didn't set something to ground at some point or something, but not sure. 

I did just test on flash magic via the settings: ISP > read signature > and saw an output that it's reading the device and such. I'm going to try and perform a more thorough optical inspection now to make sure nothing on the PCBA is shorted or anything.

> Are you sure you were using a 3.3V FTDI serial adapter? 5V FTDI serial adapter could kill the MCU.
> 
> What likely killed my oven's MCU was a shorted output pin to ground.

borland1, 2023-01-16:
One MCU test you could try in order to rule out a bad MCU, is to see if you can use the keypad to place the machine in **bootloader mode**.  Pressing the **F1** button while turning **ON** the power switch.  That only works with the firmware of this repository already installed.

You can tell the bootloader mode is active using the **Flash Magic** utility.  If **Flash Magic** connects with the MCU, then the firmware was installed and you have another hardware problem.

jjosse44, 2023-01-16:
Good point, I believe I tried this before, but never knew how to confirm it. So I would hook up the serial adapter GND, TX, and RX to the ISP header and my computer and then where do I check in Flash Magic that it has connected with the MCU? Is it under **'ISP', or 'VIEW'?** I know there are external software's to check communications, but my understanding of this stuff is minimal.

> One MCU test you could try in order to rule out a bad MCU, is to see if you can use the keypad to place the machine in **bootloader mode**. Pressing the **F1** button while turning **ON** the power [...]

borland1, 2023-01-16:
I think you could try the drop down menu ISP-Read Signature.  Or just try to flash the same hex file.  If Flash Magic doesn't connect with the MCU, you get a warning message.

Maurizio66blu, 2023-02-15:
Sorry just one piece of information. But did you finally manage to solve the problem?

> Good point, I believe I tried this before, but never knew how to confirm it. So I would hook up the serial adapter GND, TX, and RX to the ISP header and my computer and then where do I check in [...]

jjosse44, 2023-02-15:
Unfortunately not. I did however, recently receive the new NXP MCUs so I will try soldering those on and re-flashing to see if it was a fried chip. I will update this thread and other's who offered help this week or early next hopefully.

> Sorry just one piece of information. But did you finally manage to solve the problem?

jjosse44, 2023-02-21:
Update for all: Had help soldering on the new LPC2134 NXP chip and flashed the 2018 hex file firmware using FlashMagic and everything works finally! 

So my advice is, if you've tried everything else, buy a new chip and solder that on and try flashing the new blank chip. Chips are easily shorted and that's what most likely happened to mine through ESD or 5 V RX/TX signals instead of 3.3 V etc. Thank you to everyone who offered advice/help.

closed 2023-02-21 by jjosse44