UnifiedEngineering/T-962-improvements #244: Timer does not start

arusch, 2023-09-08:
Hi,

I have a T-962A and installed a DS18B20 (with 4k7 pull-up) as described in the wiki. Also, I flashed the latest firmware (v0.5.2).
When I select a reflow profile and then execute it, the timer does not start (`RUN` always states `000`), and the profile is not executed.
Also, the temperature evolution graph does not show a trace, no matter how long I wait.
Pressing the S button again only brings back the main menu.
The timer works fine in the bake/manual mode. 
The oven heats up correctly in the bake/manual mode, and the cold-junction compensation works as intended.
It really is just the timer that does not start to run.

This T-962A unit is rather new and seems to have some hardware improvements (no more aluminum tape on the insulation, no more masking tape, still some cardboard, though...). The heating chamber fan also has some baffles now.

Any idea what's going on here?
Maybe the hardware has changed just enough as compared to older versions of the oven so that the v0.5.2 firmware (from 2018?) is not working correctly anymore?

Any help is greatly appreciated!

borland1, 2023-09-08:
Maybe your firmware flash was somehow corrupted and your flasher software didn't perform a verification that the upload was successful?

Can you post a photo of your boot screen?   The boot screen briefly displays the firmware version, but also the version of the MCU.  Most earlier ovens have the NXP LPC2134 microcontroller.

Here's a picture the boot screen on my model T962 oven which has the newer design chassis with the exhaust fan duct flange.
[image]

And here's the the repository firmware complied with PlatformIO for the LPC2134, but uploaded to a LPC2148 development board.  Notice that the boot screen displays the different microcontroller.  
[image]

The LPC2134 has 128K of flash memory, while the LPC2148 has 512K.

arusch, 2023-09-08:
That was my first thought, too. However, the Flashmagictool did, indeed, verify the firmware successfully after uploading. No problems there.
Also, I have re-flashed the controller several times but the outcome is always the same.

I did not take a picture of the boot screen, and for now, I have flashed the controller with the original firmware again so that at least I have it back in a running state, but I can confirm that the boot screen looks just like yours (in the first picture). I can also confirm that my unit has the NXP LPC2134 /01 (written on the chip).

borland1, 2023-09-08:
OK, suggest you try downloading another copy of the compiled hex file again.  Might have picked up errors there.

Make sure the MCU crystal oscillator is marked 11.0592.  Timer start up is all internal to the MCU.  So assuming the MCU is fully functional, any changes to the controller PCB  should not affect the timer start up.

arusch, 2023-09-11:
The oscillator has the correct frequency.
I will download new files, try again, and report back here.
In the meantime, if you are aware of a specific combination of FW version and version of Flashmagictool (for Windows) that reportedly works well, please let me know.

borland1, 2023-09-11:
Download and use the firmware.hex file from this repository.  It works well.  Use the latest firmware version downloaded from the wiki here under heading **Assets** ...  [Latest Firmware](https://github.com/UnifiedEngineering/T-962-improvements/releases)

I have only used Flash Magic v12.35.  I put screenshots of my Flash Magic configuration on the Wiki here.  Use that version or later version should work great.

T962A, 2023-09-14:
Same issue, timer doesn't start after following the steps on the wiki. 

https://github.com/UnifiedEngineering/T-962-improvements/assets/144983418/bd43a9aa-1380-4363-aa9e-6d528279fbec

https://github.com/UnifiedEngineering/T-962-improvements/assets/144983418/b290ec59-9237-4218-b1d9-05ac4f653718

[image]

Here is picture and video of the behavior and below I've attached photos of the boards on the inside (with TC compensation mod) for reference 
[image]
[image]

[image]

borland1, 2023-09-14:
Sorry about that.  I wouldn't waste any more time troubleshooting further and suggest you replace the MCU.  If you don't have experience hand soldering at that scale, there are plenty of YouTube.com videos showing how to do drag soldering with plenty of flux on LQFP64 package.

A new MCU will have the same bootloader, so you can flash the firmware, just as you can now.

borland1, 2023-09-14:
For removal, might want to use low temperature solder, like QuikChip, or use hot air rework station.   For soldering on new MCU, here's a good tutorial...

[QFP Drag Soldering](https://www.youtube.com/watch?v=YUryJOAiPa4&t=387s )

T962A, 2023-09-15:
Fixed by pulling VBAT to 3.3V. Soldered a wire between the pin and the 3.3V reg and timer starts as intended. Unsure if this is a change on the PCB they are using or just the LPC2134 IC that ended up in this oven is more sensitive to the floating nature of the pin.  Credit to [@](https://github.com/UnifiedEngineering/T-962-improvements/issues/220) for sending me down the right path

borland1, 2023-09-15:
Glad you got it running. 

Funny thing is, I did previously look at your PCB photo for the trace to MCU pin 49 (VBAT).  The 3.3V trace comes from the regulator via capacitor C7 and the square through hole pad, then connects to pin 49 from the side.  You can confirm that the trace is from C7 is in fact 3.3V if you want to look at this repository's schematic.   Also, your PCB looks the same as the PCB on my oven.

Image below is from the NXP LPC-213x User Manual.... 
[image]

And image from your PCB
[image]

And image of PCB on my Oven
[image]

GitLang, 2023-09-15:
borland1 - I'm sure you have realised by now, but low temperature will not help to melt the higher temp solder on the chip/pcb :)

nophead, 2023-10-09:
It does, I have used ChipQuick to remove chips soldered with normal solder. That is what it is made for. You blob lots on and it stays molten long enough to be able to pull the chip off. It must mix with the original solder and form a low melting point alloy.

GitLang, 2023-10-09:
I think you are saying "low temperature solder heated to a temperature above the high temperature solder's melting point will form a molten solution", is that correct?. If so, my statement "low temperature will not help to melt the higher temp solder " is correct. LOL, I do understand you, but I find ChipQuick terrifying!

nophead, 2023-10-09:
I have always used with normal soldering iron temperature. It isn't intended to allow you to do low temperature soldering. It is for removing devices with lots of legs. You run a bead around the chip joining all the legs together and it stays molten so long that you can just lift it off. I.e. all joints are molten at the same time.

It is possible for a liquid metal to act as a solvent for metals that would be solid at that temperature. So molten ChipQuick might melt high temperature solder at a low temperature but you don't use it like that because it would freeze too fast. The reason it works is because normal solder temperature is way above its melt point, so it stays molten for long enough to have all joints liquid at the same time.