UnifiedEngineering/T-962-improvements #246: quite original (Elektor-modified) T962 has significant temperature difference, solder not melting

MasteringTheMess, 2023-10-27:
I recently purchased a modified T-962 from [Elektor](https://www.elektor.de/infrared-reflow-oven-t-962-revised-elektor-version) and the first test-runs did not produce acceptable results

[image]
[image]

This was with a profile that should reach 250C°, I showed in a Type K probe with the board and it registered merely 210°C at max.
Around 40°C temperature difference all through the process...

Would the cold junction mod alone help remedy this?

GitLang, 2023-10-28:
I don't know what solder paste you are using, but I don't know of any paste that could still be that pristine after exposure to 210 degC. It doesn't look like the flux has even flowed, let alone any sign of pre-melt deformation. Try again with Tin/Lead solder as a test and my guess is that the *board* temperature is not getting hot enough to even melt that, Have you checked all your elements are heating up OK?

I've just remembered the old temperature witness markers you could get, which would turn on a new colour band for each 10 or 20 degC reached and stay like it. So after cool down you could see the maximum temperature reached. If you can find them, a few of them dotted around the board would show what is happening.

desie, 2023-10-31:
Hi, I bought the same T-962 from Elektor one week ago. I made almost the same observation. Solder paste partially melts.
I was also thinking of a problem of the cold junction.

borland1, 2023-11-01:
You didn't say if you had calibrated the oven.

The Cold Junction Compensation 1-wire mod only adds the measured cold junction temperature to the average measured thermocouple temperatures.  If no 1-wire cold junction mod is installed, then the firmware assumes the cold junction temperature is always 25C and adds that value to the thermocouple temperatures to arrive at the average oven temperature. 

So, depending on the ambient temperature and how long the oven is operated, the actual cold junction temperature could be much higher than 25C.

Also, the factory installed thermocouples are mounted much higher in elevation than the tray drawer. You might want to consider raising the PCB above the tray level.   You could measuring the temperature of the PCB at different levels above the tray.

MasteringTheMess, 2023-11-02:
I made no modification at all to the oven.
Good idea with the temperature strips, ordered some and cant wait for them to arrive.

@desie Did you contact elektor support? They sent me a replacement oven, that has the exact same problem (in fact the above pictures are from the second one). They say that they never had complaints, maybe you can contact them (service@elektor.de) so they get aware that there are more ovens with this problem...

desie, 2023-11-02:
> @desie Did you contact elektor support? They sent me a replacement oven, that has the exact same problem (in fact the above pictures are from the second one). They say that they never had complaints, maybe you can contact them ([service@elektor.de](mailto:service@elektor.de)) so they get aware that there are more ovens with this problem...

I don't think Elektor can bring a solution to this problem, I think it's a common problem of T962.
I'll try with another solder paster next week ( the one I used for the test is a bit old ) and raise up the pcb as borland1 said.

tdarlic, 2023-11-02:
> complaints, maybe you can contact them ***@***.***) so they get

I own T962 for years now and have soldered thousands of boards with it. At
best she can do leaded solder and that in the middle of the platter. If you
tried soldering lead free with temperatures of more than 210ish deg it is
useless. Part of the platter is burned and part did not solder. I have made
all mods that I can: better isolation, cold junction compensation,
firmware, a lot of calibration with 4 temperature sensors inside, etc...
The thing we do now for this oven is that we just use bismuth based solder
paste which has much lower melting point. This is the only way you can get
it to work reliably. It can do leaded paste but the yield and reliability
is not great. You need to carefully inspect each board and sometimes
resolder complete batch. Also, I've noticed that you need to let it run few
cycles before it can solder reliably. This beats the purpose of prototype
soldering as it needs at least 15 minutes to heat up. By that time the
conveyor belt owen is also ready for use so the only advantage is that T962
spends less energy.

MasteringTheMess, 2023-11-03:
Using "wave 3", this is the result:

[image]

not a very even distribution :/

I then placed my sample pcb with the unmolten paste on some waste pcb to raise it an reduce bottom contact in the area where the most heat was during the previous test

[image]

Seems that this already helps a bit!

[image]

[image]

turned out not bad after all
I will do some further testing with raised pcbs to see what areas of the oven I can use, since not all my pcbs are that small, I have a 190x74mm one coming along soon... that one will surely be tricky!

GitLang, 2023-11-04:
Mess, I can't remember if you posted your profile, but I wonder if the peak is a little low or maybe the approx 150C to 180C soak is not long enough. You will always see a difference of anything up to 25C between the two thermocouples momentarily. You've got air currents, radiation and convection all with their own idea of how o heat your board. Like another writer here, I've done a lot of board , both leaded and unleaded, with no problems at all. I always use an M3 caphead and a nut the other side of the board, through the mounting holes and then rest the screws on the oven table. I've marked a line on the centre of the X axis ditto the Y axis, and I always arrange boards symmetrically around the cross in the centre. I never try more than 4 boards of 100mmx100mm each, absolute max. The one trick that tipped the balance for me in the early days, was preheating the oven and boards. I use the bake mode and leave it at 50C to 60C for half an hour, so every thing gets brought up to nice and warm. Then when I run the profile, I'll find the first short period the fan goes on to cool the environment done to the start temperature (if it is below what you soaked at) but things soon catch up and I find the actual average temperature follows the profile withing 8C to 10C on fast moving parts of the profile and 3C to 5C on soak periods and the peak.

Like everyone, I went through all the new machine things, earth safety, insulation etc. Then, I made the most important change of all - I fitted one of the several mod boards that connect the thermocouples directly to a MAXxxxx chip, which very accurately linearizes the thermocouples and does all the cold junction compensation, then passes the information digitally to the CPU. My opinion is that, after safety, this is the most important modification to make.

Two other thing occurs to me. The flux on your last picture does not look right. It should be a smooth, flowed glassy looking whole, not the broken up crystalline look that your has. I'm no expert, but my guess would be paste that has been kept two long in drying conditions. The other is your AC supply voltage. I'm not sure if you are in the US or not, but the 110/120VAC machines have never had as much power as the 220/240VAC. Also, it's worth checking if your local AC is below norm. Eg, the European supply is 230VAC nominal, the UK is 240VAC nominal, so the UK gets (240/230)^2 = 9%, worth having. Some ares of the US are very variable, I've seen supplies 20% down on nominal. If this is the case, you could consider adding another element, buying a transformer that you can tap off 5% or 10% higher voltage, or if you're really serious, get a regulating diesel generator. First and foremost is that digital thermocouple interface though. Good Luck.

MasteringTheMess, 2023-11-06:
This was a board that did not solder in a previous attempt and lay around for days...

The mere objective was to see if the paste melts at all

I will go thermocouple mod then (this requires a new firmware if I recall correctly?)
And I saw a mod that lets the fan rotate at a fairly low rpm during soldering to help distribute the heat more evenly, might give that a try as well?

GitLang, 2023-11-06:
Yes, the firmware will need to be updated, but you won't need to touch the offset/gain trimmers anymore. Personally, I don't like the idea of leaving the fan running, as it is the cooling part of the cool/heat balance that maintains the temperature. If you do make that change then you should also look at the PID parameters. To be honest, they are not optimal anyway - it is not easy to control temperature this way when the demanded temperature changes relatively quickly. A better mod for distributing the hot air without cooling it, is the metal bladed fan in the 'ceiling' of the oven. A metal shaft through the ceiling connects to a normal fan situated in the electronics compartment.

MasteringTheMess, 2023-11-06:
Hmmm... that ceiling-fan sounds interesting! Got any links for this mod?

desie, 2023-11-08:
Hi,
I made modifications on my T-962. Results are correct. I used a Sn63/Pb37 paste.

[image]

I added the DS18B20 for cold junction compensation, flashed the lastest firmware from UnifiedEngineering , raise the pcb above the tray level and add more kapton to improve isolation of the oven.

MasteringTheMess, 2023-11-08:
Thannks for sharing!
Although someone above mentioned that leaded paste will melt much better... so that might have helped as well?

GitLang, 2023-11-09:
Well, yes, 60/40 tin/lead melts 20 or so degrees lower than common unleaded. But you knew that already :)

As I've said before, the thermocouple circuit as supplied direct from China is dreadful. I don know what, if anything, Elektor have done to it, but get that digital thermocouple. There are loads of people designed them and shared the info freely. Then you will have control over the temperature, or at least know what temperature the thermocouples are seeing. I always do what Desie above does, which is use a screw and nut in each mounting hole to raise the board off the bed. Please use stainless steel, and not brass which do a godd job of getting soldered to your mounting holes!.

GitLang, 2023-11-09:
The tin lead solder paste I use, Chip Quik SMD291AX10, has a liquidus of 183 degC. Whereas typical lead free are around 217 degC. That is quite a difference. My profile ramps up to 150 degC and soaks there for 2 minutes. It then rises quickly to 205, dwells for 10 or 20 seconds at most then falls at 2 to 3 degC/second.

If you try to raise the heat on your profile by 2 degC/sec you will lag behind. I find 1.5 degC/sec attainable and 1dewgC/sec comfortable. That approach gets me good board everytime, but I never use more than about 200mm x 200mm central to the bed.

For board with components both sides, I do the sides one at a time. Side A as normal. For side B, I remove the screws, clean the board, screen it with paste and fit Ni plated hex spacers taller than the tallest component on side A, then reflow it with side B uppermost. The pre-soldered components on side A seem to just hang there and not move during the side B soldering.

The other thing I mentioned about pre heating the oven I think is very important. I soak the empty oven for up to half an hour at 50 or 60 degC. I put the board(s) in and start my profile at 50 degC. That way, the cycle gets off to good start and temperature tracks the profile much better. If you start with a cold oven, you will lag way behind in temperature as all the power tries to heat the whole environment in the oven.

If I think of anything else I'll post it and encourage others to do the same. The more information we post, about what works and what does not, the better results we will all get.

Maurizio66blu, 2023-11-12:
> For board with components both sides, I do the sides one at a time. Side A as normal. For side B, I remove the screws, clean the board, screen it with paste and fit Ni plated hex spacers taller than [...]

Regarding the boards with the components on both sides, could you show the process with a video on YouTube?

sevstels, 2023-11-25:
I don't understand, why not do the upgrade as described here?

 http://en.gradient-sg.com/t962/

No problems with paste, melts and soldering good. Without using contact measurement of the board surface temperature in several places - no success.

MasteringTheMess, 2023-11-27:
Well, I now loaded the modified software from here and calibrated the two stock thermocouples and things seem to have improved a lot already.
The TC were off quite a lot, around 30°C, compared to a measurement 2-3mm above the tray bottom.

Still have to do a real soldering process, but I feel it is going to be a lot better!

Thanks a lot to everyone who contributed to this cool project!
It has been very useful to me so far already :)

MasteringTheMess, 2023-11-27:
> I don't understand, why not do the upgrade as described here?
> 
> http://en.gradient-sg.com/t962/
> 
> No problems with paste, melts and soldering good. Without using contact measurement of the board surface temperature in several places - no success.

Maybe, because that is a very massive modification and not everyone wants to disassemble their oven up to that point?
Quite usable state of the oven can be achieved with lot less effort...

sevstels, 2023-11-27:
It's been explained to you several times why you can't. It will not work for different boards. Because the mass is different and the heat needed is different. It will always be underheated or overheated.

Without contact measurement of the board in several - you will never be successful. Up to you...

MasteringTheMess, 2023-11-27:
> It's been explained to you several times why you can't. It will not work for different boards. Because the mass and reflection is different and the heat needed is different. It will always be underheated or overheated.
> 
> Without contact measurement of the board in several points and auto alignment temperature- you will never be successful. Up to you... No any smart profiles in this oven do not working properly. Soldering must be self-adaptive.

Well, If you REALLY would want to go this route, first step should be to control each tube individually ...

As I said, your modification is massive and beyond what I need as a below-average, time-to-time user for simple boards ...

sevstels, 2023-11-27:
I'm not going to talk you into it. But without a few sensors, you won't get it right. So your attempts at mosaicism will only end in a lot of wasted time and poor soldering quality. It dpe with 4 contact sensors is quite poor equalisation. Analyse the graph of real temperature distribution over the surface ttam in the description. The difference is under a dozen degrees and that's bad. It should be no worse than 3 degrees. You'll never get it right manually.

zian31, 2023-12-05:
@sevstels Can you upload examples pictures here of your temp sensors on PCB ? How many sensors do you use ? Thanks :)

sevstels, 2023-12-05:
I soldered small boards, about 4x8 centimetres. 2 sensors were enough for me. But if the board is much bigger and contains heavy parts, you need more sensors. They should be placed near large parts and near heat-sensitive chips. I don't have photos at the moment. The lab was closed and I was fired, no access to the furnace now.

sevstels, 2023-12-05:
I have T962+ and left the slats/blinds on, I didn't close them. They close themselves if the fan speed is reduced. The fan must be running at least a little, otherwise the exhaust from flux vapourisation will put you to the grave quite quickly. It is necessary to dump the exhaust outside the window. The temperature drop is easily compensated by heaters.

You can easily obtain the graph using 4 sensors. If you want, add a subroutine to the code to calculate the temperature gradient between the 4 sensors. It's simple enough. Then you will get surface levelling.

zian31, 2023-12-05:
You mean that in this curve (from your webpage), the PWM during heating phase is ~60/260*100%=23% ? And at 100% during the cooling phase ?

[image]

sevstels, 2023-12-06:
The PWM modulator has a range of 0-255. Therefore, 100% of the power will be at 255. When heating is replaced by cooling, the fan blows air through the chamber. If you close the shutters, it will not be able to cool the pcb quickly. Also, if the fan is not efficient enough the cooling rate will decrease. That's what you see in the graph. To make it perfect, you need to change to 4 times more powerful fan.

sevstels, 2023-12-06:
The upper TC1 and TC2 sensors are not used. I turn them off in the menu. 
I have already explained about the difference of 40 degrees. There are 2 ways to fix this.

1. Make a temperature step from the profile. Wait until the board warms up with an acceptable error, for example 10 degrees. Then do the next heating step. But this will no longer be the original profile... it will stretch out randomly over time. If you move too slowly, the flux will have time to boil away and soldering will not work. Therefore, you need to have a sufficiently large power reserve for the heaters. In order to quickly warm it up to the melting temperature at the activation stage.

2. Additionally, forcefully mix the air with a fan. That is, if you have very massive parts on the PCB, then a fan will probably be needed.

zian31, 2023-12-06:
Thank you :) So your solution with 6 IR Lamps is just  little bit better but not good enough : i'm sad :'(

The matter is still the same : how to uniformise the heat on PCB.

Note : the Jerry Walker T-962A Little Fan solution (with 4 originals IR Lamps) is really better than 40° difference :

[image]

Just for fun (don't laugh ;) ), my next test will be to add this "Rainbean adaptateur induction" between 2 original T962+ lamps and the PCB : 
[image]

Normally, it will not work because : heating will be to slow and cooling to slow too (but air from the bottom is nice here with T962+ air flow direction). If by miracle heating is fast enough on PCB, I can imagine a mecanism to shift only the plate out during cooling ?

sevstels, 2023-12-06:
>> Jerry Walker T-962A Little Fan solution is really better than 40° difference

That's a funny enough statement :)
He hasn't even measured the difference, he simply has no sensors to do this.
It seems to me that the difference would be much greater.

zian31, 2023-12-06:
> > > Jerry Walker T-962A Little Fan solution is really better than 40° difference
> 
> That's a funny enough statement :) He hasn't even measured the difference, he simply has no sensors to do this. It seems to me that the difference would be much greater.

Are you sure ? :) It's really less than 40° difference and he has lots of sensors ;)

[image]
[image]

zian31, 2023-12-06:
I understand. For your boards with "bigs" components as inductors or ferrites or capacitors, 6 IP lamps is needed, and your special firmware with special thermal profile, and so on.

I my case, boards are high density, but my highest SMD components are less than 5mm, so I can stay with my 2 IR lamps and work to integrate a slow little fan inside as Jerry W with maybe 0% of the cool fan during heating (to be tested), with my special firmware too (a UnifiedEngineering variant) which is in connection with 4 temp sensors on PCB (one to protect the expensive chip from high temp limit).

What do you think about my "Rainbean adaptateur induction" idea between lamps and PCB ? No chance or trying should be done ?

sevstels, 2023-12-06:
If you are interested, try it. Everyone implements modifications for their own soldering conditions, and it is difficult for me to advise you. Moreover, it makes sense to make your own stove. In China, the prices for prototype assembly have gone into space and look like madness.

GitLang, 2023-12-07:
I find it is interesting that you are trying to solve problems I do not have with my machine. I have only ever used the standard thermocouples as and where fitted. I see temperature differences up to 12 degC, but only when the temperature is changing quickly on a sharp part of the profile.  The pre-liquidus plateau is within 2 degC and the peak, albeit a few seconds late, is within 3 degC. Never had a trouble with a board. Only mods are the initial safety and thermal insulation mods, and fitting the MAX chip thermocouple interface. Biggest difference is maybe the preheat I use. I always bake everything at 50 degC for 20 to 30 minutes to warm up oven, boards, everything. Then I start my profile at 50 degC. It starts smack on the profile and more of less stays there.

zian31, 2023-12-07:
> I find it is interesting that you are trying to solve problems I do not have with my machine. I have only ever used the standard thermocouples as and where fitted. I see temperature differences up [...]

Preheat is another vry good idea.
You never ry to put TC on PCB linked to multimeters for example ? To see differences between TC temps ar real PCB temp during a profile ?
Because difference between the 2 original "flying TC" between IR lamps and PCB temps is really important.
Something like that :

[image]

sevstels, 2023-12-07:
Why are you gluing the circuit board directly to the metal?! It's a totally bad idea. I use holders and the board hangs on them in the air 1.5cm from the tray

zian31, 2023-12-07:
> Why are you gluing the circuit board directly to the metal?! It's a totally bad idea. I use holders and the board hangs on them in the air 1.5cm from the tray

Yes I do the same (but with 1cm). Here it was old tests to show TC and Multimeter. And it was not directly to the metal : 2 others PCB 1.6mm are under the main one ;)

GitLang, 2023-12-08:
No, I never used any temperature sensing except the two original thermocouples in their original places. Although some commercial ovens do use direct radiation with PCB spot temperature sensing, this oven is not one of them. This oven is designed to control the temperature of the air. This is a common form of temperature control in, for example, standards rooms, where the air is refrigerated, and then heat supplied as necessary in a closed loop to control the temperature. It can be done the other way round as well. This is why, except during extreme heating, our cooling fan runs a little all the time.  The weakest part of this oven, given the price, is the interface of the thermocouples. The digital interface with the Maxim chip is the first and most important performance modification. You will never control anything with a broken sensing chain.

As also mentioned here, spacing the board off the metal deck by several mm (I use 6mm stainless steel spacers) is important. Then our controlled temperature air can get all round the board.

zian31, 2023-12-08:
:) In my case, I prefer to know each time PCB temp (and make the firmware works with it and not original air TC) that just hope it will be ok each time (and never burn any chip or decrease drastically its life time), even with prototypes. I think that the chips soldering profiles are for PCB temp, not for approximate air temp ? It's easy to see on everybody curves the differences of sometimes more than 50° between original air temps and PCB temps.

But anyway it can works without it as you said, but not for trying selling boards with almost pro result.
I can show you PCB soldering microsections to have an idea of lots and lots possible problems with soldering (and in that case, boards don't work for years).

The question of "cooling fan runs a little all the time" is not solve because it works only with T962 and do horrible results with T962+ (due to inverse air flow direction), I remember that it has been explained here : https://github.com/UnifiedEngineering/T-962-improvements/issues/210